evcc-io/evcc #7001: Zielladen/Ladeplaner Verhalten

maf-soft, 2023-03-22:
### Describe the bug

Ich habe mal wieder mit dem Zielladen gespielt und denke, dass da noch Bugs sind. Es ist aber schwer, sie richtig zu greifen und zu reproduzieren und sicher zu sein, wie das Verhalten richtig wäre. Daher äußere ich hier nur ein paar Vermutungen und hoffe jemand wird aus meinen Logs schlau und sieht ob und was schief läuft.

Womöglich hat alles nur mit dem automatischen Phasenwechsel zu tun und betrifft somit nicht jeden.
Jedenfalls scheint mir schon mal, dass die Planung von der aktuell zufällig genutzten Phasigkeit abhängt, also viel zu lang wenn es auf 1p steht. Wenn dann die Zeit gekommen ist, könnte auch sofort auf 3p umgeschaltet werden, weil ja klar ist, dass jetzt maximal geladen werden soll, es wird aber erst 3m gewartet.

Habe große Teile das Logs angehängt, ab ca. 11:58 habe ich am Ladeplaner rumgefummelt.

Ist das ein Zeichen für ein Problem? Keine Ahnung was es bedeutet.
2023/03/22 11:58:27 plan too short- ignoring remaining 2h26m49s

Bei der Phasenumschaltung kam es dann dazu:
2023/03/22 12:00:56 charger out of sync: expected enabled, got disabled

Vielleicht seht ihr ja noch mehr...

### Steps to reproduce

sorry

### Configuration details

```yaml
network:
  schema: http
  host: ... # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: ...

interval: 10s # control cycle interval

sponsortoken: eyJ... # your token from sponsor.evcc.io

tariffs:
  grid:
    type: tibber
    token: "..." # access token
    homeid: "..." # optional if multiple homes associated to account
  feedin:
    type: fixed
    price: 0.2501 # [currency]/kWh

meters:
- type: template
  template: discovergy
  usage: grid
  user: ...
  password: ...
  meter: ...22
  name: grid
- type: template
  template: discovergy
  #usage: grid
  user: ...
  password: ...
  meter: ...23
  name: house

vehicles:
- type: template
  template: tesla
  title: Tesla Model 3
  accessToken: eyJ...
[... 815 lines cut ...]
[lp-1  ] DEBUG 2023/03/22 12:05:07 planned 3h44m10s until 2023-03-22 21:00:00 +0100 CET at 17250W: total plan duration: 3h44m10s, avg cost: 0.202
[site  ] DEBUG 2023/03/22 12:05:17 ----
[lp-1  ] DEBUG 2023/03/22 12:05:17 charge power: 10175W
[site  ] DEBUG 2023/03/22 12:05:17 grid power: 9428W
[site  ] DEBUG 2023/03/22 12:05:17 pv power: 747W
[site  ] DEBUG 2023/03/22 12:05:17 site power: 9428W
[lp-1  ] DEBUG 2023/03/22 12:05:17 charge currents: [15 15.2 15.1]A
[lp-1  ] DEBUG 2023/03/22 12:05:17 detected active phases: 3p
[lp-1  ] DEBUG 2023/03/22 12:05:17 charger status: C
[lp-1  ] DEBUG 2023/03/22 12:05:17 soc estimated: 83.01% (vehicle: 81.00%)
[lp-1  ] DEBUG 2023/03/22 12:05:17 vehicle soc: 83%
[lp-1  ] DEBUG 2023/03/22 12:05:17 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/03/22 12:05:17 vehicle range: 353km
[lp-1  ] DEBUG 2023/03/22 12:05:17 add additional charging time 2h54m0s for soc > 80%
[lp-1  ] DEBUG 2023/03/22 12:05:17 planned 3h44m10s until 2023-03-22 21:00:00 +0100 CET at 17250W: total plan duration: 3h44m10s, avg cost: 0.202
[site  ] DEBUG 2023/03/22 12:05:27 ----
```

### What type of operating system are you running?

Windows

### Version

0.114.1

maf-soft, 2023-03-22:
Nachtrag: es scheint, dass die "additional charging time for soc > 80%" immer länger wird, obwohl das Auto fast voll ist.

[site  ] DEBUG 2023/03/22 13:22:47 ----
[lp-1  ] DEBUG 2023/03/22 13:22:47 charge power: 7924W
[site  ] DEBUG 2023/03/22 13:22:47 grid power: 7341W
[site  ] DEBUG 2023/03/22 13:22:47 pv power: 583W
[site  ] DEBUG 2023/03/22 13:22:47 site power: 7341W
[lp-1  ] DEBUG 2023/03/22 13:22:47 charge currents: [11.6 11.8 11.7]A
[lp-1  ] DEBUG 2023/03/22 13:22:47 detected active phases: 3p
[lp-1  ] DEBUG 2023/03/22 13:22:47 charger status: C
[lp-1  ] DEBUG 2023/03/22 13:22:47 soc estimated: 97.56% (vehicle: 96.00%)
[lp-1  ] DEBUG 2023/03/22 13:22:47 vehicle soc: 98%
[lp-1  ] DEBUG 2023/03/22 13:22:47 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/03/22 13:22:47 vehicle range: 415km
[lp-1  ] DEBUG 2023/03/22 13:22:47 add additional charging time 3h52m0s for soc > 80%
[lp-1  ] DEBUG 2023/03/22 13:22:47 planned 4h24m22s until 2023-03-22 21:00:00 +0100 CET at 17250W: total plan duration: 4h24m22s, avg cost: 0.235

VolkerK62, 2023-03-22:
Die Berechnung der Ladedauer wurde neu überdacht https://github.com/evcc-io/evcc/discussions/6235 und befindet sich zur Zeit in Arbeit.

maf-soft, 2023-03-22:
> > Jedenfalls scheint mir schon mal, dass die Planung von der aktuell zufällig genutzten Phasigkeit abhängt, also viel zu lang wenn es auf 1p steht.

> Sollte es bei Automatik nicht- woran machst Du das im Log fest?

Ich fürchte da hatte ich das mit dem soc>80 noch nicht gesehen und das einfach aus der langen Planzeit geschlossen - er wollte viel zu früh und damit zu höheren Preisen anfangen zu laden und in der günstigen Stunde wäre er dann schon längst fertig gewesen. Also wenn du im Log keine Anomalien siehst, dann ist es ja gut.

> > Wenn dann die Zeit gekommen ist, könnte auch sofort auf 3p umgeschaltet werden, weil ja klar ist, dass jetzt maximal geladen werden soll, es wird aber erst 3m gewartet.

> Wo im Log?

Es geht wohl hier los: 2023/03/22 11:58:57 wake-up timer: start
Und wenig später sieht man dann dass er nur 1m wartet. Aber jedenfalls fängt er mit 1p an zu laden und unterbricht es dann wieder um auf 3p umzuschalten und dann ist das noch der sync-Fehler, wonach es aber offenbar korrekt weiter geht.

Da gleichzeitig aber noch die Sonne schien und ich in der UI rumklickte, ist das jetzt wohl schlecht nachzuvollziehen. Ich denke, wenn du sonst nichts siehst, was dich stört, können wir es dabei belassen und ich gucke erst wieder nach dem nächsten Release genauer hin. Aber vor dem Planer-Start auf 3p umzuschalten bleibt auf jeden Fall der Vorschlag, sofern das nicht schon so ist und hier nur aus anderen Gründen nicht geklappt hat.

> > plan too short- ignoring remaining 2h26m49s

> Das Log ist irreführend. Hier wird ein kurzer Restslot nicht gestartet, siehe #7007

:)

andig, 2023-03-22:
Check.

closed 2023-03-22 by andig