evcc-io/evcc #7003: Netz-Ladeplaner auch zulassen wenn Überschussladen ausgeschaltet ist.

maf-soft, 2023-03-22:
Ich bin tibber-Nutzer und möchte PV-Überschussladen ausschalten und trotzdem den Ladeplaner (Zielzeit/Zielpreis) nutzen können.

Hintergrund ist in meinem Fall, das tibber manchmal deutlich günstiger ist als meine Eigenverbrauchsvergütung von 2009. Heute hat Vormittags die Sonne geschienen, aber erst ab 12 Uhr war der Netzstrom richtig günstig. Da möchte ich natürlich nicht schon vor der günstigen Zeit das Auto laden. Aktuell lässt sich das über die UI nicht einstellen und ich musste mir einen Wecker stellen :)

Man könnte dafür einen 5. Modus einführen, z.B. "PV Aus".

Oder man ändert das Modi-Konzept ganz, so dass man mehrere Schalter hat, die man separat setzen und kombinieren kann, z.B. 
[Netz] [PV] [Min] [Max].
Die Abhängigkeiten dann über Ausgrauen visualisieren.

VolkerK62, 2023-03-22:
So abwegig finde ich deinen Luxus-Ansatz gar nicht. Die Einspeisevergütung ist ja bekannt. Die müsste "nur" als Tarif angesehen werden.

andig, 2023-03-22:
> So abwegig finde ich deinen Luxus-Ansatz gar nicht. Die Einspeisevergütung ist ja bekannt. Die müsste "nur" als Tarif angesehen werden.

Die Idee verstehe ich nicht. Es bräuchte ja zusätzlich noch einen Differenztarif *und* eine neue Tarifart die sich nur auf Einspeisung bezieht, richtig? Dann wäre das nicht "nur"...

VolkerK62, 2023-03-22:
Deshalb das -nur- in " ", weil ich keine Ahnung von dem Aufwand habe. 
Beim Zielladen werden doch die billigsten Zeiträume ausgewählt. Wenn der Einspeisetarif bei 25ct liegt und es bei Tibber Zeitäume mit z.B. 14ct gibt, dann werden diese Slots zum Zielladen genutzt. 
Aber, jetzt wo ich das schreibe, fällt mir ein, dass der Energieanteil der unter den Einspeistarif fällt, nicht berechnet werden kann, weil von der Sonne abhängig.
Ok. ich ziehe meinen Kommentar von oben zurück ;-)

maf-soft, 2023-03-22:
Wenn euch vor allem die UI-Änderung stört, natürlich ginge es auch mit einem Config-Parameter, der festlegt, dass "Aus" den Ladeplaner aktiv lässt, Standard bleibt wie es ist. Wäre für mich perfekt.

> Dein Anforderungsprofil- Tibbernutzer ein EV aus 2009 ist recht spezifisch. Dazu hättest Du mit aktuellem Verhalten bessere Netzdienlichkeit. Ich denke das ist wontfix.

Also dass das jetzt erstmal als niedrige Prio angesehen wird, weil es einfach wichtigeres gibt, könnte ich verstehen. Aber dass du das gleich so abtust und darin keinen Nutzen für andere siehst, verstehe ich nicht. Natürlich gibt es noch andere mit hoher Einspeisevergütung. Ich wollte den Vorschlag hier einfach schonmal untergebracht haben und nun könnte man gucken ob sich noch andere Befürworter finden.

Den Nutzer gegen seinen Willen zu marginal höherer Netzdienlichkeit (wenn überhaupt) und dafür höheren Kosten zu zwingen, finde ich nicht förderlich. Es reicht, die Standardeinstellungen dafür sinnvoll festzulegen und ich sehe keinen vernünftigen Grund, die Ladeplaner-Features davon abhängig zu machen ob man **zu anderen Zeiten** auch Überschussladen will. Ich denke, das hat sich jetzt einfach so ergeben, weil der Ladeplaner nachträglich rangeflanscht wurde und das einzige Problem ist, dass mein Wunsch etwas Aufwand wäre, den jetzt keiner machen will mangels Interesse. Aber dann könnte man es einfach offen lassen. Womöglich würde ich es mir sogar irgendwann ansehen ob ich das hinkriege.

StefanSchoof, 2023-03-23:
Wenn man gerade nicht per PV Mode laden will und trotzdem im PV Mode ist kann man den enable threshold per API auf einen Wert setzen der so hoch ist das PV nie genutzt wird. Das wäre ein Workaround ohne das jetzt irgendwas am Code geändert werden müsste