evcc-io/evcc #7010: Optimize SofarSolar G3 template

premultiply, 2023-03-22:
Hi @premultiply ,

Vielen Dank!

Ich habe jetzt zwei Dinge getan:

1. Schritt: Ich habe einen qualitativ besseren RS485 Adapter genommen: Waveshare Industrial USB to RS485 Converter with original FT232RL (https://www.amazon.de/dp/B082D5PXJ4) - das hat schon gleich mal eine spürbare Verbesserung gebracht. Fehlerrate etwa um 50% gesunken. Mit den LEDs des Adapters sieht man auch, dass nur alle x Sekunden Daten übertragen werden.
2. Schritt: Die delays wie Du sagtest eingebaut. 
    - Mit 1s wurde das schon deutlich besser. 
    - Runter auf 0.5s: wurde es wieder schlechter. 
    - Hoch auf 2s: Es wird besser
    - Hoch auf 3s: Es wird deutlich zuverlässiger. Zum ersten Mal bekomme ich komplette Auslesungen, aber immer noch hin und wieder Fehler. 
    - Hoch auf 5s: Die Auslesung wird zäh - aber es gibt immer noch Fehler. Keine spürbare Verbesserung gegenüber 3s.
 
Ich denke evcc ignoriert Fehler beim Einlesen und nimmt dann den nächsten Wert? Daher scheinen mir 3s delay der Sweetspot zu sein.

Meter config ist jetzt:

```
meters:
  - name: sofar-grid
    type: custom
    power:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0488 # Total PCC active power. Selling electricity is positive, buying electricity is negative
        type: holding
        decode: int16
      scale: -10
      delay: 3s
    energy:
      source: modbus
      id: 1
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x068E # Energy_Purchase_Total
        type: holding
        decode: uint32
      scale: 0.1
      delay: 3s
    currents:
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x0492 # R phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
        delay: 3s
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x049D # S phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
        delay: 3s
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x04A8 # T phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
        delay: 3s
  - name: sofar-pv
    type: custom
    power:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x05C4 # Total PV power
        type: holding
        decode: uint16
      scale: 100
      delay: 3s
    energy:
      source: modbus
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0686 # PV_Generation_Total
        type: holding
        decode: uint32
      scale: 0.1
      delay: 3s
  - name: sofar-battery-1
    type: custom
    power:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0606 # The charging and discharging power of the first battery pack. Charge is positive, discharge is negative
        type: holding
        decode: int16
      scale: -10
      delay: 3s
    energy:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x069A # Bat_Discharge_Total
        type: holding
        decode: uint32
      scale: 0.1
      delay: 3s
    soc:
      source: modbus
      id: 1
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0608 # No. 1 battery pack SOC
        type: holding
        decode: uint16
      delay: 3s
    capacity: 5
```

Baust Du den Delay in das Template ein? Dann würde ich morgen nochmal den nightly build testen.

_Originally posted by @cschlipf in https://github.com/evcc-io/evcc/issues/6953#issuecomment-1480071651_

andig, 2023-03-22:
Sollte man alle Anwender dafür bestrafen dass der Adapter schlecht ist oder liegt es am Gerät selbst?

premultiply, 2023-03-22:
Scheint eher am Gerät selbst zu liegen.

andig, 2023-03-23:
Bei einem Delay von 3s zwischen einzelnen Abfragen, noch dazu instabil, würde ich auf das Ding lieber komplett verzichten.

andig, 2023-03-23:
Ticket beim Hersteller könnte helfen?

cschlipf, 2023-03-23:
Kann ich mal versuchen.  Aber wäre es bis dahin nicht besser zumindest mit Delay einen Workaround zu haben? Wie gesagt: Ich teste heute abend mal ob ich mit der Zeit noch runtergehen kann und ob es mit Delay auch mit dem billigeren USB Adapter geht. Halte Euch in der Loop.

Firmware des Wechselrichters habe ich übrigens bereits auf die aktuellste Version aktualisiert.

cschlipf, 2023-03-23:
Hier die Ergebnisse meines Tests:

### Waveshare Industrial Adapter:

- EVCC läuft seit gestern mit einem Delay von 3s 
    - ✅ Keine einzige Fehlermeldung im web UI
- Delay auf 1s reduziert:
    - `evcc meter` funktioniert meistens - gibt aber selten einen time out bei einem Wert
    - EVCC Web UI zeigt bisher keine Fehler.
    - ✅  Könnte also brauchbar sein. Ich werde hier noch über Nacht einen längeren Test laufen lassen und morgen berichten, ob Fehler aufgetaucht sind.
- Delay auf 0s reduziert:  
    - `evcc meter` zeigt das vorherige eratische Verhalten.   
    - Es dauert nicht lange und die ersten Fehlermeldungen tauchen im Web UI auf.
    - ⛔ Nicht funktional
- Modus auf `id: 200` bei Delay 0s
    - Kein Unterschied zu `id: 1`  
    - ⛔ Nicht funktional

### QITA RS485 USB Adapter

- Delay auf 1s reduziert:
    - `evcc meter` funktioniert meistens - gibt aber selten einen time out bei einem Wert. Eigentlich kein spürbarer Unterschied zum Waveshare
    - EVCC Web UI zeigt selten einen Fehler. Aber längst nicht so viele wie ohne Delay. EVCC ist so benutzbar, aber es könnten Benutzerrequests kommen
    - ⚠️ Mit Einschränkungen brauchbar.

### Fazit
- Ohne Delay keine Chance
- Der Waveshare USB Adapter scheint eine leicht bessere Performance zu haben, jedoch nicht entscheidend.
- Ein Delay von 1s scheint akzeptabel zu sein. Wenn man ihn parametrisieren kann, könnte er bei Bedarf vom User erhöht werden.
- Im großen und ganzen gebe ich die Schuld dem Wechselrichter.

andig, 2023-03-23:
Also 1s konfigurierbares delay?

cschlipf, 2023-03-23:
Das wäre mein Vorschlag, wenn das konfigurierbar geht. Und wenn man den Parameter schon verfügbar macht, könntet ihr das eventuell ggf. Dokumentieren, wann man ihn erhöhen sollte um unnötige Support Calls zu vermeiden?

cschlipf, 2023-03-23:
EVCC läuft aktuell immer noch ohne einen einzigen Lesefehler bei Delay 1s mit Waveshare USB Adapter.

PR #7023 (2023-03-24): SofarSolar: Fix timeouts by delay
May fix #7010
closed 2023-03-24 by premultiply