evcc-io/evcc #7025: [Feature Request] Add 'error' as messaging event

cschlipf, 2023-03-23:
**Is your feature request related to a problem? Please describe.**
When EVCC runs into an error state like broken communication to the inverter, vehicle, charger,... or read errors of values, EVCC displays these on the top right in a drop down. However when you do not have the UI open, you do not get any notice of that error state.

**Describe the solution you'd like**
EVCC has messaging already built-in. However the events are just related to normal operations. Adding an event 'error', which sends a notification to the configured service with a variable containing the error message would be great for monitoring proper operation of EVCC.

andig, 2023-03-23:
Yeah, that was asked before but implementation turned out to be surprisingly difficult. Not at this time although we'd really like to!

andig, 2023-03-23:
Das war https://github.com/evcc-io/evcc/issues/6354. Hab keine Lösung gefunden :(