evcc-io/evcc #7027: Kostal Piko legacy: Regex matcht wo es NICHT soll

premultiply, 2023-03-23:
### Describe the bug

Ursprungs-Issue (mit HTML-Quellcode): https://github.com/evcc-io/evcc/discussions/3496

Erneuter, erfolgloser Fix-Versuch: https://github.com/evcc-io/evcc/commit/0680f16fed199a998792c3123335f05928181ddc

Tritt aktuell hier schon wieder auf: https://github.com/evcc-io/evcc/discussions/5340#discussioncomment-5410191

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

.

andig, 2023-03-23:
Kann ich nix tun, nach meinem Test gehts ja. Close?

premultiply, 2023-03-23:
Na das löst das Problem ja auch nicht unbedingt.

Hast du das nur im Regex-Tester geprüft oder in der Implementierung mit Webserver usw.?

andig, 2023-03-23:
RE Tester. Der claim war dass es da matcht. Ansonsten kann man das ohne Webserver sicher auch über mqtt testen.

premultiply, 2023-03-23:
Ja, so hab ich das natürlich auch alles vorher getestet und da funktioniert es wie es soll und erwartet wurde.

Nur in der Praxis, sprich durch den evcc Code durch leider nicht.
Dieses Rätsel gilt es hier zu ergründen und zu lösen. 😉

andig, 2023-03-24:
Wir bräuchten:

- [ ] die problematische Seite
- [ ] den Test das regex101 funktioniert
- [ ] einen Test und eine Config dass evcc nicht funktioniert (kann mit const Plugin gemaut werden)

Dann kann ich mir das anschauen.

djrepose, 2023-03-25:
bezüglich der problematischen seite reicht dir der quelltext?

andig, 2023-03-25:
Yep, plus Test auf regex101

djrepose, 2023-03-25:
ok ich schick die infos gegen den abend hin wenn ich zuhause bin.

djrepose, 2023-03-25:
Regex test Tag:
[image]

Regex Test Nacht:

Seitenquelltext:

<!DOCtype HTML PUBLIC "-//W3C//Dtd HTML 4.0 Transitional//EN">
<html>
<head>
<meta HTTP-EQUIV="Content-Type" CONTENT="text/html; charset=ISO-8859-1">
<meta name="Generator" CONTENT="ChrisB">
<title>PV Webserver</title>
</head>
<body nof="(MB=(DefaultMasterborder, 65, 60, 150, 10), L=(HomeLayout, 700, 600))" bgcolor="#EAF7F7" text="#000000" link="#0033CC" vlink="#990099" alink="#FF0000" topmargin=0 leftmargin=0 marginwidth=0 marginheight=0>
<form method="post" action="">
<table cellspacing="0" cellpadding="0" width="770" nof="ly">
<tr><td height="5"></td></tr>
<tr><td width="190" height="55"></td>
<td width="400">
  <font face="Arial,Helvetica,Geneva,Sans-serif,sans-serif" size="+3">
  PIKO 10.1
<br><font size="+1">                 
  PV  (255)
</font>
</font>
</td>
<td>[image]</td>
</tr>
</table>

<font face="Arial,Helvetica,Geneva,Sans-serif,sans-serif">
<table Border="0" width="100%"><tr>
<td width="150"></td>
<td> <hr> </td>
</tr></table>
<table cellspacing="0" cellpadding="0" width="770">
<tr><td></td></tr>
<tr>
<td width="190"></td>
<td colspan="2">
  <b>AC-Leistung</b></td>
<td>&nbsp</td>
<td>
  <b>Energie</b></td></tr>
<tr><td height="10"></td></tr>

<tr>
<td width="190"></td>
<td width="100">
  aktuell</td>
<td width="70" align="right" bgcolor="#FFFFFF">
  267</td>
<td width="140">&nbsp W</td>
<td width="100">
  Gesamtenergie</td>
<td width="70" align="right" bgcolor="#FFFFFF">
  80135</td>
<td width="50">&nbsp kWh</td>
<td>&nbsp</td></tr>
<tr height="2"><td></td></tr>
<tr>
<td width="190"></td>
<td width="100">
  &nbsp</td>
<td width="70" align="right">
  &nbsp</td>
<td width="140">&nbsp</td>
<td width="100">
  Tagesenergie</td>
[... 166 lines cut ...]
</table>
</td></tr></table></font>

<hr>
<table cellspacing="0" cellpadding="0" width="770">
<tr><td height="5"></td></tr>
<tr><td width="190"></td>
<td width="440">
<font face="Arial,Helvetica,Geneva,Sans-serif,sans-serif">
<a href="LogDaten.dat">Historie</a>
&nbsp &nbsp
<a href="Info.fhtml">Infoseite</a>
&nbsp &nbsp
<a href="BA.fhtml"></a>
&nbsp &nbsp
<a href="igak.fhtml"></a>
</font></td>
<td align="right">
<font face="Arial,Helvetica,Geneva,Sans-serif,sans-serif">
<a href="Solar2.fhtml">Einstellungen</a></font></td>
<td width="50"></td>
</tr></table></font>
</form>
</body>
</html>

djrepose, 2023-03-25:
[neu 3.txt](https://github.com/evcc-io/evcc/files/11069544/neu.3.txt)

andig, 2023-03-25:
Und diese Datei soll jetzt was tun oder nicht? Was ist Deine Erwartung und was ist das tatsächliche Ergebnis?

djrepose, 2023-03-25:
also aus meiner sicht funktionieren beide regex. Wenn er aber keinen int / zahlenwert findet bezüglich es steht x x x im feld 
springt das tamplate nicht auf den vorgesehnen default wert 0 sondern auf das nächste 
feld mit zahlen. da muss das template umgebaut werden dass wenn keine zahlen vorhanden sind das er raus springt und den wert 0 annimmt.

andig, 2023-03-25:
Und wo sehe ich das in Deiner Datei?

djrepose, 2023-03-25:
https://regex101.com/r/Q8SBUm/1

andig, 2023-03-25:
> Es braucht einen Testcase für das was nicht funktioniert, bitte erneut mit regec101 geprüft.

Das wäre jetzt der zweite? Wozu dient der erste Test? Bitte nicht falsch verstehen, aber bevor ich da Zeit rein stecke möchte ich gerne das Problem finden ;)

djrepose, 2023-03-25:
der erste testcase ist der fall am tag wenn alles in ordnung ist.
Der 2. Testcase ist der
problemfall wenn das template auf default 0 springen sollte. da springt er aber im evcc auf den nechsten zahlenwert welcher die 80kw sind. obwohl er auf regex 101 alles richtig macht.

djrepose, 2023-03-25:
[log trace evcc.txt](https://github.com/evcc-io/evcc/files/11069723/log.trace.evcc.txt) 

hier der log trace von evcc meter ---> Power: 80135W

andig, 2023-03-25:
Geht bei mir wunderbar:

```
gor main.go meter --log trace,db:error -c piko.yaml
[main  ] INFO 2023/03/25 19:13:19 evcc 0.0.1-alpha
[main  ] INFO 2023/03/25 19:13:19 using config file: piko.yaml
success
-------
Power: 95W

fail
----
Power: 0W

```

Mit

```yaml
meters:
  - type: custom
    name: success
    power:
      source: const
      regex: '(?s)aktuell</td>\s+<td[^>]+>\s+(\d+)</td>'
      default: 0
      value: |
        <!DOCtype HTML PUBLIC "-//W3C//Dtd HTML 4.0 Transitional//EN">
        <html>
        <head>
        <meta HTTP-EQUIV="Content-Type" CONTENT="text/html; charset=ISO-8859-1">
        <meta name="Generator" CONTENT="ChrisB">
        <title>PV Webserver</title>
        </head>
        <body nof="(MB=(DefaultMasterborder, 65, 60, 150, 10), L=(HomeLayout, 700, 600))" bgcolor="#EAF7F7" text="#000000" link="#0033CC" vlink="#990099" alink="#FF0000" topmargin=0 leftmargin=0 marginwidth=0 marginheight=0>
        <form method="post" action="">
        <table cellspacing="0" cellpadding="0" width="770" nof="ly">
        <tr><td height="5"></td></tr>
        <tr><td width="190" height="55"></td>
        <td width="400">
          <font face="Arial,Helvetica,Geneva,Sans-serif,sans-serif" size="+3">
          PIKO 10.1
        <br><font size="+1">
          PV  (255)
          PV  (255)
        </font>
        </font>
        </td>
        <td><img alt="Logo" height="42" width="130" src="KSE.gif"></td>
        </tr>
        </table>

        <font face="Arial,Helvetica,Geneva,Sans-serif,sans-serif">
        <table Border="0" width="100%"><tr>
        <td width="150"></td>
        <td> <hr> </td>
        </tr></table>
        <table cellspacing="0" cellpadding="0" width="770">
        <tr><td></td></tr>
        <tr>
        <td width="190"></td>
        <td colspan="2">
          <b>AC-Leistung</b></td>
        <td>&nbsp</td>
        <td>
          <b>Energie</b></td></tr>
        <tr><td height="10"></td></tr>

        <tr>
        <td width="190"></td>
[... 450 lines cut ...]
        </td></tr></table></font>

        <hr>
        <table cellspacing="0" cellpadding="0" width="770">
        <tr><td height="5"></td></tr>
        <tr><td width="190"></td>
        <td width="440">
        <font face="Arial,Helvetica,Geneva,Sans-serif,sans-serif">
        <a href="LogDaten.dat">Historie</a>
        &nbsp &nbsp
        <a href="Info.fhtml">Infoseite</a>
        &nbsp &nbsp
        <a href="BA.fhtml"></a>
        &nbsp &nbsp
        <a href="igak.fhtml"></a>
        </font></td>
        <td align="right">
        <font face="Arial,Helvetica,Geneva,Sans-serif,sans-serif">
        <a href="Solar2.fhtml">Einstellungen</a></font></td>
        <td width="50"></td>
        </tr></table></font>
        </form>
        </body>
        </html>
```

djrepose, 2023-03-25:
kann ich das runterladen zum testen?

andig, 2023-03-25:
Ist in 20min im nightly.

djrepose, 2023-03-25:
ok ich meld mich wenns drauf ist

andig, 2023-03-25:
Yep, probier mal bitte die Beispielconfig oben. Wenns dann heute nach immer noch schief läuft muss ich mal den Case aus Deinem Logfile auf einen Webserver legen. Ich sehe allerdings auch nicht, was da anders sein sollte; Dein Log ist wirklich *sehr* merkwürdig :/

premultiply, 2023-03-25:
Hab ich gerade schon gemacht:
```yaml
meters:
- name: piko_tag
  type: template
  template: kostal-piko-legacy
  usage: pv
  host: test.ok-rlp.de/piko_tag.html # IP-Adresse oder Hostname
- name: piko_nacht
  type: template
  template: kostal-piko-legacy
  usage: pv
  host: test.ok-rlp.de/piko_nacht.html # IP-Adresse oder Hostname
```

```
@premultiply ➜ /workspaces/evcc (master) $ ./evcc meter
[main  ] INFO 2023/03/25 18:35:27 evcc 0.114.1 (a163020b)
[main  ] INFO 2023/03/25 18:35:27 using config file: /workspaces/evcc/evcc.yaml
piko_nacht
----------
Power: 0W

piko_tag
--------
Power: 95W

@premultiply ➜ /workspaces/evcc (master) $
```

andig, 2023-03-25:
Geht also auch. Dann muss @djrepose mal seine Installation auf Deine Nachtadresse richten.

djrepose, 2023-03-25:
ok perfekt nun läufts bei mir auch richtig :)

andig, 2023-03-26:
Äh... was heisst das? Der Testfall geht oder evcc geht? Wäre schön, wenn auch sehr merkwürdig dass es vorher nicht ging. Der Commit aus Deinem Logfile schien mir richtig zu sein...

djrepose, 2023-03-26:
der testfall geht bei mir.

andig, 2023-03-26:
Bitte- welcher? Der mit const oder der gegen die http Seite?

djrepose, 2023-03-26:
der mit const

andig, 2023-03-26:
Fein. Und der andere? Bei Uli klappts ja.

closed 2023-04-01 by andig