evcc-io/evcc #7032: soc poll timer is not stopped

maf-soft, 2023-03-24:
### Describe the bug

Poll settings are at default, so mode should be "charging". But I stopped charging by setting to "Aus" in the UI and the timer continued in the log.

### Steps to reproduce

-PV
-Aus

### Configuration details

```yaml
loadpoints/soc/poll/mode not configured
```

### Log details

```text
[site  ] DEBUG 2023/03/24 09:24:50 ----
[lp-1  ] DEBUG 2023/03/24 09:24:50 charge power: 0W
[site  ] DEBUG 2023/03/24 09:24:50 grid power: -6285W
[site  ] DEBUG 2023/03/24 09:24:50 pv power: 6285W
[site  ] DEBUG 2023/03/24 09:24:50 site power: -6285W
[lp-1  ] DEBUG 2023/03/24 09:24:50 charge currents: [0.01 0.009 0.01]A
[lp-1  ] DEBUG 2023/03/24 09:24:50 charger status: B
[lp-1  ] DEBUG 2023/03/24 09:24:50 next soc poll remaining time: 49m50s
[lp-1  ] DEBUG 2023/03/24 09:24:50 available power 6285W > 4140W min 3p threshold
[lp-1  ] DEBUG 2023/03/24 09:24:50 start phase scale3p timer
[lp-1  ] DEBUG 2023/03/24 09:24:50 phase scale3p in 1m0s
[lp-1  ] DEBUG 2023/03/24 09:24:50 pv charge current: 27.3A = 0A + 27.3A (-6285W @ 1p)
[lp-1  ] DEBUG 2023/03/24 09:24:51 max charge current: 25A
[lp-1  ] DEBUG 2023/03/24 09:24:52 set charge mode: off
[lp-1  ] DEBUG 2023/03/24 09:24:52 phase timer inactive
[site  ] DEBUG 2023/03/24 09:24:52 ----
[lp-1  ] DEBUG 2023/03/24 09:24:52 charge power: 0W
[site  ] DEBUG 2023/03/24 09:24:52 grid power: -6258W
[site  ] DEBUG 2023/03/24 09:24:52 pv power: 6258W
[site  ] DEBUG 2023/03/24 09:24:52 site power: -6258W
[lp-1  ] DEBUG 2023/03/24 09:24:52 charge currents: [0.01 0.009 0.01]A
[lp-1  ] DEBUG 2023/03/24 09:24:52 charger status: B
[lp-1  ] DEBUG 2023/03/24 09:24:52 next soc poll remaining time: 49m48s
[lp-1  ] DEBUG 2023/03/24 09:24:53 charger disable
[lp-1  ] DEBUG 2023/03/24 09:24:53 wake-up timer: stop
[site  ] DEBUG 2023/03/24 09:25:00 ----
[lp-1  ] DEBUG 2023/03/24 09:25:00 charge power: 0W
[site  ] DEBUG 2023/03/24 09:25:00 grid power: -5814W
[site  ] DEBUG 2023/03/24 09:25:00 pv power: 5814W
[site  ] DEBUG 2023/03/24 09:25:00 site power: -5814W
[lp-1  ] DEBUG 2023/03/24 09:25:00 charge currents: [0.01 0.009 0.01]A
[lp-1  ] DEBUG 2023/03/24 09:25:00 charger status: B
[lp-1  ] DEBUG 2023/03/24 09:25:00 next soc poll remaining time: 49m40s
[site  ] DEBUG 2023/03/24 09:25:10 ----
[lp-1  ] DEBUG 2023/03/24 09:25:10 charge power: 0W
[site  ] DEBUG 2023/03/24 09:25:10 grid power: -4489W
[site  ] DEBUG 2023/03/24 09:25:10 pv power: 4489W
[site  ] DEBUG 2023/03/24 09:25:10 site power: -4489W
[lp-1  ] DEBUG 2023/03/24 09:25:10 charge currents: [0.01 0.009 0.01]A
[lp-1  ] DEBUG 2023/03/24 09:25:10 charger status: B
[lp-1  ] DEBUG 2023/03/24 09:25:10 next soc poll remaining time: 49m29s
[site  ] DEBUG 2023/03/24 09:25:20 ----
[lp-1  ] DEBUG 2023/03/24 09:25:20 charge power: 0W
[site  ] DEBUG 2023/03/24 09:25:20 grid power: -2859W
[site  ] DEBUG 2023/03/24 09:25:20 pv power: 2859W
[site  ] DEBUG 2023/03/24 09:25:20 site power: -2859W
[lp-1  ] DEBUG 2023/03/24 09:25:20 charge currents: [0.01 0.009 0.01]A
[lp-1  ] DEBUG 2023/03/24 09:25:20 charger status: B
[lp-1  ] DEBUG 2023/03/24 09:25:20 next soc poll remaining time: 49m20s
[site  ] DEBUG 2023/03/24 09:25:30 ----
```

### What type of operating system are you running?

Windows

### Version

0.114.1

maf-soft, 2023-03-24:
I wonder if I'm getting you wrong somehow, since my poll mode is mentioned twice above: it is the default, which should be "charging".

andig, 2023-03-24:
> loadpoints/soc/poll/mode not configured

I had no idea what that meant. The yaml would have been clear ;)

maf-soft, 2023-03-24:
I just saw no point in posting my complete yaml just to prove that it doesn't contain some setting :)

andig, 2023-03-24:
/cc @nename0 @Hofyyy ist das Fallout von https://github.com/evcc-io/evcc/pull/6267? 

Ich sehe keinen Grund im Code (ausser dem PR), warum bei `mode: charging` der Timer noch laufen sollte. Ich vermute das ist die letzte Abfrage. Ist es richtig, dass die erst nach Timerablauf zuschlägt oder hättet ihr erwartet, dass das Update sofort kommt?

maf-soft, 2023-03-24:
Kurz nach dem Stoppen noch ein soc Update zu machen würde sicherlich auch bei meinem Tesla funktionieren, weil er dann noch nicht schläft. Würde ich richtig finden, jedenfalls besser als das ausstehende Update ganz zu streichen.

maf-soft, 2023-03-24:
Darf ich, weil es hier vielleicht gerade passt, noch einen anderen Gedanken anbringen? Oder sagt mir wo sonst.

Wenn der Wagen voll wird, könnte man nicht auch dann die ganzen Aktivitäten einstellen, also soc poll, Leistung, Phasenwechsel? (Ich schätze dass auch dann mein Wagen einschläft und der nächste soc poll wieder zum Fehler führt)

nename0, 2023-03-24:
Das Grundproblem ist, dass dieses 'Wenn der Wagen voll wird' eigentlich zwei Szenarien sein können:

- der Estimator unterschätzt den Soc (und Fahrzeug Soc limit gleich EVCC limit): Also Evcc denkt Soc = 89%. Das Auto ist auf 90% limitiert und erreicht die 90% und stoppt. EVCC wundert dich warum es gestoppt hat. Hier wäre ein sofortiger Soc Poll sinvoll aber aktuell schwierig wegen dem Caching auf einer unteren Ebene (Vehicle Provider).
- der Estimator überterschätzt den Soc: Also Evcc denkt Soc = 90%. Das Auto ist aber auf 89%. EVCC stoppt die Ladung. Nutzer wundert sich warum Auto nicht voll. Auch hier wäre ein sofortiger Soc Poll sinnvoll, am Besten noch vor dem Ladestopp.

Wenn EVCC dann sicher ist, dass das Auto "voll" ist dann könnte man deinen Vorschlag umsetzen.

MarkusGH, 2023-04-04:
> Ja das ist "erstmal" richtig so. Hofyyy und ich waren uns auch einig, dass natürlich ein sofortiges Update besser wäre. 
> Das ist aber mit dem aktuellen Caching im Vehicle Provider schwierig, weil man halt nie weiß welchen Zustand der 
> Cache hat bzw. wann man einen neu oder alten Werten bekommen hat/bekommen wird.

Kann man den Cache nicht vom Loadpoint aus resetten um ein Update zu erzwingen?

nename0, 2023-04-05:
> Kann man den Cache nicht vom Loadpoint aus resetten um ein Update zu erzwingen?

Klar technisch ist das möglich, aber aktuell gibt das die Schnittstelle zu den Fahrzeug-APIs nicht her.
Wäre also ne größere Sache...

MarkusGH, 2023-04-05:
Ich versteh schon, dass es ein Akt ist, das in allen Fahrzeug APIs nachzupflegen. Allerdings könnte man doch als ersten Schritt ein Api definieren, das den Cache zurücksetzt. Das könnte man dann (falls es der betreffende Vehicle Provider unterstützt) vom Loadpoint aus aufrufen und dann sofort Updaten. Falls nicht vorhanden würde das bisherige Verhalten bleiben.
Ich bin leider nicht der größte GO crack, deswegen kann ich das nicht so gut ausdrücken wie ich es für z.B. C++ oder C# könnte...

andig, 2023-04-05:
> Kann man den Cache nicht vom Loadpoint aus resetten um ein Update zu erzwingen?

Kann man, was ist Frage dahinter?

MarkusGH, 2023-04-05:
Dann kann man im evChargeStopHandler den Cache resetten und sofort ein letztes Update vom SoC etc. machen

andig, 2023-04-05:
> der Estimator unterschätzt den Soc (und Fahrzeug Soc limit gleich EVCC limit): Also Evcc denkt Soc = 89%. Das Auto ist auf 90% limitiert und erreicht die 90% und stoppt. EVCC wundert sich warum es gestoppt hat. Hier wäre ein sofortiger Soc Poll sinvoll aber aktuell schwierig wegen dem Caching auf einer unteren Ebene (Vehicle Provider).

In dem Fall sieht es zwar komisch aus, ist aber alles ok- das Auto ist ja genau so voll wie es sein soll. Reset würde helfen diesen letzten Wert zu bekommen. PR folgt.

> der Estimator überschätzt den Soc: Also Evcc denkt Soc = 90%. Das Auto ist aber auf 89%. EVCC stoppt die Ladung. Nutzer wundert sich warum Auto nicht voll. Auch hier wäre ein sofortiger Soc Poll sinnvoll, am Besten noch vor dem Ladestopp damit einfach weiter geladen wird.

Schon schwieriger. Dafür müsste `targetSocReached` einmalig das Ergebnis ignorieren. PR dafür wäre super.

> Dann kann man im evChargeStopHandler den Cache resetten und sofort ein letztes Update vom SoC etc. machen

PR für Case 1 kommt.

andig, 2023-04-05:
Das isso, aber 2x Relais klackern will man eigentlich nicht für vllt. noch fehlende 10min. Da wär das sogar eher kontraproduktiv.

premultiply, 2023-04-05:
Irgendwann kommt das Update aber ohnehin und lädt dann auch weiter - zumindest wenn es auf "connected" steht.

MarkusGH, 2023-04-05:
Ich wüsste nicht warum es schlecht sein sollte, wenn evcc sofort reagiert und nicht erst nach Ablauf des SoC Update Intervalls.

andig, 2023-04-05:
Du kannst das drehen wir Du willst. Solange der soc nur bei "charging" aktualisiert würde gäbe es genau diese nächste Abfrage ja nicht... ymmv.

MarkusGH, 2023-04-05:
Das ist nicht das Verhalten, das ich sehe:

Apr  5 11:19:32 homebridge evcc[1374478]: [lp-1  ] INFO 2023/04/05 11:19:32 stop charging <-
Apr  5 11:19:32 homebridge evcc[1374478]: [lp-1  ] DEBUG 2023/04/05 11:19:32 wake-up timer: start
Apr  5 11:34:38 homebridge evcc[1374478]: [porsche] TRACE 2023/04/05 11:34:38 GET https://api.ppa.porsche.com/app/connect/v1/vehicles/*    **?mf=BATTERY_LEVEL&mf=BATTERY_CHARGING_STATE&mf=CLIMATIZER_STATE&mf=E_RANGE&mf=HEATING_STATE&mf=MILEAGE

Und bei mir steht poll mode auf "charging"

nename0, 2023-04-05:
> Das ist nicht das Verhalten, das ich sehe:

Genau, nach #6267 ist jetzt ein einmaliges Update erlaubt.

MarkusGH, 2023-04-05:
Exakt! Und mit #7238 erfolgt dieses Update früher.

andig, 2023-04-06:
PR ist drin, bitte schaut Euch das Verhalten morgen mal an.

closed 2023-04-21 by andig