evcc-io/evcc #7037: Wattpilot charging but showing "Not connected"

micw, 2023-03-24:
### Describe the bug

Hello,

my wattpilot is currently showing "not connected" while my car is properly charging:

[image]

How can I debug this to provide the required information?

Maybe of importance: my readings of the power meter (grid) are currently not working.

### Steps to reproduce

Unknown

### Configuration details

```yaml
log: debug

interval: 10s

site:
  - title: Falkenweg 5
    meters:
      grid: grid
      pv:
      - pv1
      - pv2

mqtt:
  broker: 192.168.1.3:1883
  topic: evcc
  user: sensor
  password: *redacted*

meters:
  - name: grid
    type: custom
    power:
      source: mqtt
      topic: grid/*redacted*/powerTotal
      timeout: 30s
  - name: pv1
    type: custom
    power:
      source: mqtt
      topic: solar/*redacted*/0/power
      timeout: 30s
  - name: pv2
    type: custom
    power:
      source: mqtt
      topic: solar/*redacted*/0/power
      timeout: 30s

vehicles:
  - name: Leaf
    type: template
    template: nissan
    user: *redacted*
    password: *redacted*
    vin: *redacted*
    capacity: 40
    phases: 1
  - name: Enyaq
    type: template
    template: enyaq
    user: *redacted*
[... 165 lines cut ...]
evcc    | [site  ] DEBUG 2023/03/24 13:46:42 ----
evcc    | [lp-1  ] DEBUG 2023/03/24 13:46:42 charge power: 0W
evcc    | [lp-2  ] DEBUG 2023/03/24 13:46:42 charge power: 9700W
evcc    | [site  ] DEBUG 2023/03/24 13:46:42 pv power: 122W
evcc    | [site  ] ERROR 2023/03/24 13:46:42 grid meter: grid/1EBZ0100597780/powerTotal outdated: 22h7m12s
evcc    | [site  ] DEBUG 2023/03/24 13:46:52 ----
evcc    | [lp-1  ] DEBUG 2023/03/24 13:46:52 charge power: 0W
evcc    | [lp-2  ] DEBUG 2023/03/24 13:46:52 charge power: 9690W
evcc    | [site  ] DEBUG 2023/03/24 13:46:52 pv power: 122W
evcc    | [site  ] ERROR 2023/03/24 13:46:52 grid meter: grid/1EBZ0100597780/powerTotal outdated: 22h7m22s
evcc    | [site  ] DEBUG 2023/03/24 13:47:02 ----
evcc    | [lp-1  ] DEBUG 2023/03/24 13:47:02 charge power: 0W
evcc    | [lp-2  ] DEBUG 2023/03/24 13:47:02 charge power: 9690W
evcc    | [site  ] DEBUG 2023/03/24 13:47:02 pv power: 122W
evcc    | [site  ] ERROR 2023/03/24 13:47:02 grid meter: grid/1EBZ0100597780/powerTotal outdated: 22h7m32s
```
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.114.1 (a216a867)

micw, 2023-03-24:
Hab das ganze mal auf meinem Kubernetes deployed (das stand noch aus). Nach dem Neustart (selbe Config, selbe Datenbank) das selbe Verhalten. Hier das Log ab start:

[mqtt ] DEBUG 2023/03/24 15:07:23 tcp://192.168.1.3:1883 connected
[lp-1 ] WARN 2023/03/24 15:07:38 locking phase config to 3p for switchable charger
[lp-2 ] WARN 2023/03/24 15:07:38 locking phase config to 3p for switchable charger
[site ] INFO 2023/03/24 15:07:39 site config:
[site ] INFO 2023/03/24 15:07:39 meters: grid ✓ pv ✓ battery ✗
[site ] INFO 2023/03/24 15:07:39 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2023/03/24 15:07:39 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2023/03/24 15:07:39 pv 2: power ✓ energy ✗ currents ✗
[site ] INFO 2023/03/24 15:07:39 vehicles:
[site ] INFO 2023/03/24 15:07:39 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[site ] INFO 2023/03/24 15:07:39 vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1 ] INFO 2023/03/24 15:07:39 loadpoint 1:
[lp-1 ] INFO 2023/03/24 15:07:39 mode: pv
[lp-1 ] INFO 2023/03/24 15:07:39 charger: power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2023/03/24 15:07:39 meters: charge ✓
[lp-1 ] INFO 2023/03/24 15:07:39 charge: power ✓ energy ✗ currents ✓
[lp-2 ] INFO 2023/03/24 15:07:39 loadpoint 2:
[lp-2 ] INFO 2023/03/24 15:07:39 mode: pv
[lp-2 ] INFO 2023/03/24 15:07:39 charger: power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-2 ] INFO 2023/03/24 15:07:39 meters: charge ✓
[lp-2 ] INFO 2023/03/24 15:07:39 charge: power ✓ energy ✗ currents ✓
[lp-1 ] DEBUG 2023/03/24 15:07:39 phase timer inactive
[lp-1 ] DEBUG 2023/03/24 15:07:39 pv timer inactive
[lp-1 ] INFO 2023/03/24 15:07:39 vehicle updated: unknown -> Leaf
[lp-2 ] DEBUG 2023/03/24 15:07:39 phase timer inactive
[lp-2 ] DEBUG 2023/03/24 15:07:39 pv timer inactive
[lp-2 ] INFO 2023/03/24 15:07:39 vehicle updated: unknown -> Enyaq
[lp-2 ] DEBUG 2023/03/24 15:07:39 max charge current: 6A
[site ] DEBUG 2023/03/24 15:07:39 ----
[lp-1 ] DEBUG 2023/03/24 15:07:39 charge power: 0W
[lp-2 ] DEBUG 2023/03/24 15:07:39 charge power: 4280W
[site ] DEBUG 2023/03/24 15:07:39 pv power: 233W
[mqtt ] DEBUG 2023/03/24 15:07:39 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:07:49 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:07:59 grid/1EBZ0100597780/powerTotal wait for initial value
[site ] ERROR 2023/03/24 15:08:09 grid meter: grid/1EBZ0100597780/powerTotal outdated: 10s
[site ] DEBUG 2023/03/24 15:08:09 ----
[lp-1 ] DEBUG 2023/03/24 15:08:09 charge power: 0W
[lp-2 ] DEBUG 2023/03/24 15:08:09 charge power: 4290W
[site ] DEBUG 2023/03/24 15:08:09 pv power: 238W
[mqtt ] DEBUG 2023/03/24 15:08:09 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:08:19 grid/1EBZ0100597780/powerTotal wait for initial value
[lp-2 ] DEBUG 2023/03/24 15:08:22 set charge mode: now
[mqtt ] DEBUG 2023/03/24 15:08:29 grid/1EBZ0100597780/powerTotal wait for initial value
[site ] ERROR 2023/03/24 15:08:39 grid meter: grid/1EBZ0100597780/powerTotal outdated: 10s
[site ] DEBUG 2023/03/24 15:08:39 ----
[lp-1 ] DEBUG 2023/03/24 15:08:39 charge power: 0W
[lp-2 ] DEBUG 2023/03/24 15:08:39 charge power: 9710W
[site ] DEBUG 2023/03/24 15:08:39 pv power: 252W
[mqtt ] DEBUG 2023/03/24 15:08:39 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:08:50 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:09:00 grid/1EBZ0100597780/powerTotal wait for initial value
[site ] ERROR 2023/03/24 15:09:10 grid meter: grid/1EBZ0100597780/powerTotal outdated: 10s
[site ] DEBUG 2023/03/24 15:09:10 ----
[lp-1 ] DEBUG 2023/03/24 15:09:10 charge power: 0W
[lp-2 ] DEBUG 2023/03/24 15:09:10 charge power: 9720W
[site ] DEBUG 2023/03/24 15:09:10 pv power: 269W
[mqtt ] DEBUG 2023/03/24 15:09:10 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:09:20 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:09:30 grid/1EBZ0100597780/powerTotal wait for initial value
[site ] ERROR 2023/03/24 15:09:40 grid meter: grid/1EBZ0100597780/powerTotal outdated: 10s
[site ] DEBUG 2023/03/24 15:09:40 ----
[lp-1 ] DEBUG 2023/03/24 15:09:40 charge power: 0W
[lp-2 ] DEBUG 2023/03/24 15:09:40 charge power: 9720W
[site ] ERROR 2023/03/24 15:09:41 pv 1 power: solar/116182009632/0/power outdated: 31s
[site ] DEBUG 2023/03/24 15:09:41 pv power: 87W
[mqtt ] DEBUG 2023/03/24 15:09:41 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:09:51 grid/1EBZ0100597780/powerTotal wait for initial value
[... 7 lines cut ...]
[mqtt ] DEBUG 2023/03/24 15:10:21 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:10:31 grid/1EBZ0100597780/powerTotal wait for initial value
[site ] ERROR 2023/03/24 15:10:41 grid meter: grid/1EBZ0100597780/powerTotal outdated: 10s
[site ] DEBUG 2023/03/24 15:10:41 ----
[lp-1 ] DEBUG 2023/03/24 15:10:41 charge power: 0W
[lp-2 ] DEBUG 2023/03/24 15:10:41 charge power: 9770W
[site ] DEBUG 2023/03/24 15:10:41 pv power: 346W
[mqtt ] DEBUG 2023/03/24 15:10:41 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:10:52 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:11:02 grid/1EBZ0100597780/powerTotal wait for initial value
[site ] ERROR 2023/03/24 15:11:12 grid meter: grid/1EBZ0100597780/powerTotal outdated: 10s
[... cut ...]
[mqtt ] DEBUG 2023/03/24 15:11:32 grid/1EBZ0100597780/powerTotal wait for initial value
[site ] ERROR 2023/03/24 15:11:42 grid meter: grid/1EBZ0100597780/powerTotal outdated: 10s
[site ] DEBUG 2023/03/24 15:11:42 ----
[lp-1 ] DEBUG 2023/03/24 15:11:42 charge power: 0W
[lp-2 ] DEBUG 2023/03/24 15:11:42 charge power: 9710W
[site ] DEBUG 2023/03/24 15:11:42 pv power: 388W
[mqtt ] DEBUG 2023/03/24 15:11:42 grid/1EBZ0100597780/powerTotal wait for initial value
[mqtt ] DEBUG 2023/03/24 15:11:52 grid/1EBZ0100597780/powerTotal wait for initial value

micw, 2023-03-24:
Ich weiß nicht, ob es am Wattpilot Plugin liegt. Beim 2. Fahrzeug hatte ich das selbe Problem. Sobald ich den "grid" Meter gefixt hatte, funktionierte auch die Wattpilot-Anzeige wieder.

closed 2023-03-24 by andig

micw, 2023-03-24:
@andig Wieso completed? Grid weg -> Wattpilotanzeige kaputt. Grid da -> Wattpilotanzeige geht. Ist aus meiner Sicht ein offener Bug.

andig, 2023-03-25:
Ohne Netzzähler gehts einfach nicht

mabunixda, 2023-03-25:
Bzw was versteht du unter "Grid weg"? Ohne Grid im Notstrommodus? 🤔

micw, 2023-03-25:
MQTT Datenstrom vom "grid"-Meter ausgefallen