evcc-io/evcc #7039: Custom meter - "usage" param not accepted

ThomasKalten, 2023-03-24:
### Describe the bug

My grid meter is already connected via an IR-interface to my iobroker instance.
I tried to get the meter values from there via mqtt and a custum configuration.
The parameter "usage" is not accepted for this custom meter:
**\* '' has invalid keys: usage**

Further info in the details below.

Would it be helpfull to create a template, where the param "usage" can be specified?

Thanks

### Steps to reproduce

1. Add the custom meter configuration to evcc.yaml

### Configuration details

```yaml
Here ist the mqtt and meter declaration:

mqtt:
  broker: 192.168.x.y:1885
  topic: evcc
  user: aUser
  password: aPassword
  clientid: evcc567

meters:
- usage: grid
  type: custom
  name: Hausanschluss
  power: # power (W)
    source: mqtt
    topic: smartmeter/0/1-0:16_7_0__255/value
  energy: # optional energy (kWh)
    source: mqtt
    topic: smartmeter/0/1-0:1_8_0__255/value
  currents: # optional currents (A)
    - source: mqtt
      topic: smartmeter/0/1-0:31_7_0__255/value
    - source: mqtt
      topic: smartmeter/0/1-0:51_7_0__255/value
    - source: mqtt
      topic: smartmeter/0/1-0:71_7_0__255/value
```

### Log details

```text
[mqtt  ] INFO 2023/03/24 16:49:21 connecting evcc567 at tcp://192.168.x.y:1885
[main  ] FATAL 2023/03/24 16:49:21 cannot create meter 'Hausanschluss': cannot create meter 'custom': 1 error(s) decoding:

* '' has invalid keys: usage
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.114.1

sschoof, 2023-03-24:
Der usage parameter steuert wie sich das template verhalten soll. Einige Geräte sind z.B. Netzzähler und PV Zähler in einem. Für einen Custom meter ist das nicht nötig und nicht vorgesehen: https://docs.evcc.io/docs/devices/meters#manuell

andig, 2023-03-24:
Hintergrund:  `usage` ist ausschließlich dazu da, in Templates die richtige Unterkonfiguration auszuwählen. Den "custom" Zähler baust Du ja schon so wie Du ihn brauchst.

closed 2023-03-24 by andig