evcc-io/evcc #7049: Wallbox Pulsar Plus via OCPP Fehlermeldungen

cnbe70, 2023-03-25:
### Describe the bug

Hallo zusammen, ich habe seit einigen Tagen evcc in Betrieb und bin wirklich sehr happy damit. Die Konfiguration ist:
- Docker auf Synology NAS
- E3DC Anlage mit Batteriespeicher
- Wallbox Pulsar Plus mittels OCPP

Das Problem ist folgendes: nach einem Ladevorgang bekommt evcc scheinbar nicht mit, dass der Ladevorgang beendet ist. Die Anzeige bleibt im Frontend bei 10kW stehen, was sich auch in den Logfiles widerspiegelt. 

Damit vermutlich zusammenhängend entsteht auch das Problem, dass irgendwand die Meldung "charge meter: not available" erscheint, die bis zum Neustart alle 10 sek. produziert wird.

Das Problem lässt sich nur mit einem Restart des Dockers beheben, tritt dann aber nach kurzer Zeit erneut wieder auf.

Ich weiß nicht, ob ich ein Konfigurationsproblem habe oder ob das Problem in der OCPP Schnittstelle liegt. Freue mich über jeden Tipp!

Vielen Dank
Christian

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: 192.168.xxx.xxx
  port: 7070

interval: 10s # control cycle interval

# log settings
log: info
levels:
  site: fatal #debug
  lp-1: fatal
  lp-2: fatal
  cache: fatal
  db: fatal

modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

meters:
  - name: grid
    type: template
    template: e3dc
    usage: grid
    host: 192.168.xxx.194
    port: 502
  - name: pv
    type: template
    template: e3dc
    usage: pv
    host: 192.168.xxx.194
    port: 502
  - name: battery
    type: template
    template: e3dc
    usage: battery
    host: 192.168.xxx.194
    port: 502

[... 208 lines cut ...]
2023-03-25T06:27:27.812537088Z,stdout,[site  ] DEBUG 2023/03/25 07:27:27 pv power: 227W
2023-03-25T06:27:27.805046348Z,stdout,[lp-1  ] DEBUG 2023/03/25 07:27:27 charge power: 10648W
2023-03-25T06:27:27.804694336Z,stdout,[site  ] DEBUG 2023/03/25 07:27:27 ----
2023-03-25T06:27:17.818115147Z,stdout,[lp-1  ] DEBUG 2023/03/25 07:27:17 vehicle range: 186km
2023-03-25T06:27:17.818092117Z,stdout,[lp-1  ] DEBUG 2023/03/25 07:27:17 vehicle soc: 50%
2023-03-25T06:27:17.818036027Z,stdout,[lp-1  ] DEBUG 2023/03/25 07:27:17 soc estimated: 49.98% (vehicle: 48.00%)
2023-03-25T06:27:17.817970206Z,stdout,[lp-1  ] DEBUG 2023/03/25 07:27:17 charger status: C
2023-03-25T06:27:17.817946516Z,stdout,[lp-1  ] DEBUG 2023/03/25 07:27:17 detected active phases: 3p
2023-03-25T06:27:17.817916026Z,stdout,[lp-1  ] DEBUG 2023/03/25 07:27:17 charge currents: [15.5 15.6 15.3]A
2023-03-25T06:27:17.817881526Z,stdout,[site  ] DEBUG 2023/03/25 07:27:17 site power: 11069W
2023-03-25T06:27:17.817813735Z,stdout,[site  ] DEBUG 2023/03/25 07:27:17 grid power: 10969W
2023-03-25T06:27:17.816239675Z,stdout,[site  ] DEBUG 2023/03/25 07:27:17 battery power: 0W
2023-03-25T06:27:17.816175624Z,stdout,[site  ] DEBUG 2023/03/25 07:27:17 battery soc: 0%
2023-03-25T06:27:17.811260142Z,stdout,[site  ] DEBUG 2023/03/25 07:27:17 pv power: 219W
2023-03-25T06:27:17.803072528Z,stdout,[lp-1  ] DEBUG 2023/03/25 07:27:17 charge power: 10626W
2023-03-25T06:27:17.802714206Z,stdout,[site  ] DEBUG 2023/03/25 07:27:17 ----
```

### What type of operating system are you running?

Docker container

### Version

0.114.1

andig, 2023-03-25:
> Das Problem ist folgendes: nach einem Ladevorgang bekommt evcc scheinbar nicht mit, dass der Ladevorgang beendet ist. Die Anzeige bleibt im Frontend bei 10kW stehen, was sich auch in den Logfiles widerspiegelt.

Vermutung: 
- die Box schickt Messwerte nur auf Anforderung (sollte im trace log sichtbar sein)
- wir fordern nur an solange geladen wird
- danach nicht mehr

...es fehlt dann also der erste Messwert nach dem Laden den wir wieder statisch anzeigen müssten

cnbe70, 2023-03-25:
Danke für die pfeilschnelle Antwort!
Kann ich da etwas z.B. über eine Konfigurationseinstellung machen? Oder sollte ich besser auf ein Update warten?

andig, 2023-03-25:
Erstmal bitte aussagekräftiges Logfile (ocpp: trace).

cnbe70, 2023-03-26:
Hier ist zunächst ein Logfile, das den Fehler "charge meter: not available" zeigt. Vielleicht übersehe ich da etwas, aber das Problem scheint quasi aus dem Nichts zu kommen.... (ocpp: trace ist eingeschaltet). Noch irgendeine Idee, was ich loggen könnte? Bei der Wallbox selbst, habe ich keine Logging Option gefunden.

[charge meter not available.csv](https://github.com/evcc-io/evcc/files/11070845/charge.meter.not.available.csv)

cnbe70, 2023-03-26:
Und hier ein Logfile, das das Ende des Ladens zeigt (inkl. ocpp: trace). Solltest du noch mehr Daten benötigen, melde dich gerne.

[ende laden.csv](https://github.com/evcc-io/evcc/files/11070855/ende.laden.csv)

andig, 2023-03-26:
> Hier ist zunächst ein Logfile, das den Fehler "charge meter: not available" zeigt. 

Das ist in der Nightly Version behoben- Du hast anscheinend noch das Release. 

> Und hier ein Logfile, das das Ende des Ladens zeigt 

Ich sehe da leider kein Ladeende sondern nur "nicht laden". Eine vollständiges Logfile würde helfen.

Könntest Du das bitte nochmal mit dem aktuellen Nightly testen? Zumindest der erste Fehler sollte dann schon weg sein!

cnbe70, 2023-03-26:
HIer ist schon mal das komplette Logfile (sorry, ist sehr groß). Spannend ist aus meiner Sicht der Timestamp: 2023-03-26T03:50:43 Danach geht die Grid Power und Site Power runter, während die Charge Power unverändert bleibt.
[evcc-evcc1 (1).csv](https://github.com/evcc-io/evcc/files/11071300/evcc-evcc1.1.csv)

andig, 2023-03-26:
Wie gesagt. Nimm bitte das Nightly.

cnbe70, 2023-03-26:
Sorry, war nicht schnell genug mit meiner Antwort. Ich habe das nightly installiert. Das Problem tritt in einer Variante leider immer noch auf.
[evcc-evcc-nightly.csv](https://github.com/evcc-io/evcc/files/11071321/evcc-evcc-nightly.csv)

andig, 2023-03-26:
Probier bitte mal neues nightly in 20min. Das gibt wenn nicht geladen wird und es keine neuen Meßwerte gibt entweder 0 zurück (für Strom und Leistung) oder den letzten Wert (für Energie). Das sollte ganz sinniges Verhalten sein für die Boxen die ihre Meßwerte nicht freiwillig raus rücken.

closed 2023-03-26 by andig

cnbe70, 2023-03-26:
Danke für die schnelle Umsetzung. Hier ist das Logfile. Der Meßwert für die Ladestation geht tatsächlich runter auf 0. Damit wäre dieses Problem aus meiner Sicht schon mal gelöst. (siehe Logfile).

[evcc-evcc-nightly2.csv](https://github.com/evcc-io/evcc/files/11071685/evcc-evcc-nightly2.csv)

Bei 2023-03-26T13:10:43.921762713Z ist im Logfile noch ein Fehler dokumentiert: charge meter total import: not available
Die Meldung tritt nach dem Start des Containers auf, nicht während des Ladens und auch nicht nach Abschluss des Ladevorgangs. (Vielleicht ist das aber auch nur Zufall.)

andig, 2023-03-26:
Also der Wert ist da:

```json
[2,"935685041","MeterValues",{"meterValue": [{"timestamp": "2023-03-26T13:10:45Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "263435.4"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Current.Offered","unit": "A","value": "16"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "228.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "241.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "236.0"} ]} ],"connectorId": 0}]
```

Warum er die nicht "sieht" kann ich jetzt nicht erkennen. Wenn wir das rausfinden wollen müsste ich Dir mal Zugang einrichten und Du dann Deine Box auf meine Debuginstanz zeigen lassen. Das wäre vielleicht eine Aufgabe fürs nächste Wochenende ;)

cnbe70, 2023-03-26:
Ich bin dabei! Nächste Woche habe ich allerdings Urlaub. Ich melde mich zuverlässig, wenn ich zurück bin. Danke für deine Unterstützung! Ich empfinde das nicht als selbstverständlich.

cnbe70, 2023-04-21:
Ich würde ja wahnsinning gerne den Fehler genauer dokumentieren. Allerdings tritt er nach einem Update der Wallbox Firmware und der Aktualisierung von EVCC nicht mehr auf.

andig, 2023-04-21:
Das ist doch wunderbar 👍🏻