evcc-io/evcc #7060: Opel car coordinates seem to be switched

pauxus, 2023-03-26:
### Describe the bug

We have an Opel e-Corsa and a VW T7 here. Both stand on parkingspots side by side.

evcc vehicle reports the coordinates for the VW as (48.xx, 12,yy southern Germany), which is correct, however Opel shows the coordinates as (12.yy, 48.xx Gulf of Aden), which definitely is not (ocean-ground estate is not affordable right now).

Since the coordinates are currently not used anywhere (?), this is no bug issue, but could be addressed at convenience.

Open question: is this specific to PSA, Opel or this specific model?

### Steps to reproduce

1. Create a config with an Opel car
2. call 'evcc vehicle'
3. validate coordinates
...

### Configuration details

```yaml
not relevant
```

### Log details

```text
not relevant
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.114.1 (a163020b)

andig, 2023-03-27:
Geopos for PSA is really just a slice:

    "coordinates":[13,48,340]

Since I don't have any Peugeot/Citroen etc for testing I could only switch them. Happy to take a PR.

pauxus, 2023-03-28:
I can do a PR, but beforehand I would like confirmation that it is a general problem.

andig, 2023-03-28:
Let's just change it- I guess position is not too visible as it's not in the ui.

closed 2023-04-01 by andig