evcc-io/evcc #7062: Safeguard for wrong climating notification

pauxus, 2023-03-26:
**Is your feature request related to a problem? Please describe.**
Currently Renault seems to report the climater as always active. This lead to draining the house battery quite a while until I noticed the actual problem. Even switching to "off" did not stop the charging, I had to manually unplug.

**Describe the solution you'd like**
There could be a couple of solutions.
1. it would be really useful if the climate-state would bei shown in the UI, this would have allowed me to find the problem a lot earlier.
2. #6588 would at least pose a quick workaround in the specific case
3. "off" should also turn of climater based charging
4. Some sort of sanity check: i.e. climater charge should only run for an hour or so.

**Additional context**
I noticed that "log level trace" does not show any requests for Renault (anymore). Has this changed?

andig, 2023-04-02:
Sollte jetzt per poll-mode mindestens als Workaround gelöst sein?

andig, 2023-04-02:
> it would be really useful if the climate-state would bei shown in the UI, this would have allowed me to find the problem a lot earlier.

/cc @naltatis haben wir das jetzt in den Notifications mit drin?

naltatis, 2023-04-02:
@andig der Climate Status ist schon vor einiger Zeit (neues Design) aus der UI rausgeflogen. Die Information ist in der API aber noch da. Würde vorschlagen, dass wir das als Statustext wieder mit aufnehmen.