evcc-io/evcc #7077: Helm-Chart

micw, 2023-03-27:
Hi,

ich hab evcc hier im Kubernetes laufen. Besteht Interesse, ein Helm-Chart ins Repo aufzunehmen? Dann mache ich einen pull request.

Viele Grüße,
Michael.

andig, 2023-03-27:
Gerne separat oder in "Lösungen". Das können/wollen wir nicht warten.

davek804, 2024-03-24:
@micw Es tut mir leid, ich spreche kein Deutsch. Ich verwende eine Übersetzungsanwendung. Haben Sie gesagt, dass diese Anwendung in Kubernetes ausgeführt wird? Wenn ja, haben Sie ein Repo, das ich klonen, forken oder zu dem ich beitragen kann? Danke schön!

---

@micw I apologize, I don't speak German. I am using a translation application. Did you say you have this application running in Kubernetes? If soz do you have a repo I can clone, fork, or contribute to? Thank you!

micw, 2024-03-24:
English is fine, no need to apologize.
Yes I run evcc in k8s. It's a single pod with nothing special, so quite a simple chart.
I can create a PR to add it to this repo if it is of interest.

andig, 2024-03-24:
@micw see https://github.com/evcc-io/evcc/issues/7077#issuecomment-1485441174

micw, 2024-03-24:
@andig would it make sense to have it here as a repo besides evcc? Or should it be completel separated?

andig, 2024-03-24:
Same answer for 3rd time: please add to https://github.com/evcc-io/evcc/discussions/categories/lösungen if you want.

KlavsKlavsen, 2025-05-30:
I would love to have helm chart / Kubernetes install method listed as an install method.. Can you share your chart?
I run everything on k3d on my OrangePI - makes life soo much simpler for me - running multiple stuff on the same box, having recovery etc. in place :)

KlavsKlavsen, 2025-06-05:
do the docker image need any specific volume mounts to work (keep config etc.) - or is everything "stored somewhere else".. in a db maybe?

micw, 2025-06-05:
hi. you need to persist store ~/.evcc and probably also the evcc.yaml config (or alternatively provide that as config map).

Here's the chart I use: https://github.com/micw/evcc-helm-chart
Not 100% battle proof but does the job for for since years