evcc-io/evcc #7093: Einspeisung trotz vollem PV-Akku und BufferSoC

Stefan0875, 2023-03-28:
Ich habe als BufferSoc 81% konfiguriert, als PrioritySoC 80%.

Wenn der Hausakku nun den PrioritySoC erreicht hat, die PV Leistung aber nicht die nötigen 4,1kW Überschuss liefert, startet das Laden des Autos nicht.

Solange der PV-Akku noch geladen werden kann und somit eine Einspeisung verhindert wird, ist das OK. 
Wenn aber der PV-Akku dann irgendwann voll ist, dann startet das Laden nicht und die erzeugte Energie wird eingespeist.

Erst wenn die PV Power einmal kurz die nötigen 4,1kW übersteigt startet das Laden und wird auch bei wenig PV Power fortgesetzt bis der BufferSoC erreicht wird.

Hier muss eine intelligentere Regelung gefunden werden, die ein Einspeisen verhindert.
Es muss eine gewisse Hysterese geben, nach erreichen bzw. überschreiten des BufferSoC soll auch nicht sofort das Laden des Autos begonnen werden.

Stefan0875, 2023-03-28:
Ist es geplant das zu ändern? 
Ich sehe das eher als Fehlerhaftes Verhalten (Bug) und nicht als fehlendes Feature, oder?

premultiply, 2023-03-28:
Das war zuvor so und es wurde dann gewünscht dass es so wie jetzt sein sollte 😁

VolkerK62, 2023-03-28:
Welche Art "Hysterese" schwebt dir vor?

VolkerK62, 2023-03-28:
Das ist das hauptsächlich ein 3p-Problem.
Dazu kommt, wenn mit Hilfe von `buffersoc` geladen wird und die Entladeleistung der Batterie nicht für die Mindestladeleistung des Fahrzeug ausreicht, wird mit Netzbezug weitergeladen.
Für eine vernünftige Regelung, die ohne Netzbezug auskommt, müsste die maximale Entladeleistung der Batterie bekannt sein.

GrimmiMeloni, 2023-03-28:
Die Entladeleistung könnte man ja in der Konfiguration hinterlegen.

Stefan0875, 2023-03-29:
Richtig, die maximale Entladeleistung sollte man konfigurieren können.
Wobei hier ja dann auch PVLeistung + maxEntladadeleistung < 4,1 sein müsste damit aus dem Netz bezogen wird.

@VolkerK62 , ja das ist bei 3p Laden ein größeres Problem. Aber 1p3p Phasenumschaltung haben bei weitem nicht alle (ich vermute sogar die wenigsten) und reines 1p Laden will man doch auch nicht. also trifft es viele.
Die Hysterese würde ich konfigurierbar machen, ich weiß nicht ob ich den oberen Wert auf 100% stellen würde, evtl. auf 95%

Aktuell ist es jedenfalls nicht gut gelöst.
Der PV-Akkus ist voll, ein "leeres" Auto ist angeschlossen und trotzdem startet das laden nicht und es wird stattdessen eingespeist.
Das Laden beginnt auch nicht beim Einstecken des Autos:

Frau kommt nachmittags nach Hause, die PV Anlage liefert zB 3kW, der PV-Akku ist bei 100%, und steckt das Auto ein.
Ich würde dann erwarten dass das Laden beginnt, mit 4,1kW, 3 vom Dach und 1,1 aus den PV-Akku

@premultiply Wenn das mal so war und geändert wurde scheint es ja Leute zu geben die es so richtig finden (aber warum?), dann sollte man sich aber zwischen beiden Varianten entscheiden können (Konfig).

Die Ideale Lösung verhindert ein Eispeisen ins Netz und nutzt den PV-Akku so wenig wie möglich und so viel wie nötig

VolkerK62, 2023-03-29:
(Meine Programmierkenntnisse beschränken sich auf Excel-Formeln)
Einen Schalter `bufferStart: true/false` stelle ich mir relativ einfach vor.

Wenn die maxEntladeleistung rein kommt, wird es wohl komplizierter.
In die Zukunft gedacht (Stichwort: Redesign PV-Mode) wäre die aber sicherlich nützlich.

premultiply, 2023-03-29:
Es gibt sogar eine nicht zu unterschätzende Anzahl von Nutzern die keinesfalls irgendein Watt aus dem Heimspeicherchen in den Fahrzeugakku laden wollen, sondern lieber diesen fürs Haus aufsparen und Fahrstrom aus dem Netz zukaufen wollen.

Wie man es macht ist es verkehrt ;)

andig, 2023-03-29:
…die nutzen aber auch kein `bufferSoc`!

Stefan0875, 2023-03-29:
Ganz genau
Und für die Meisten wird doch das Ziel sein den PV Strom möglichst geschickt selbst zu verbrauchen.
Da hat der PV-Akku erst einmal Priorität.
Jede kWh die ich aus dem PV-Akku ins Auto "umlade" ist natürlich teurer als wenn ich sie direkt ins Auto laden würde, somit sollte das auch gering gehalten werden.

aber wenn der Akku voll ist und noch Strom vom Dach kommt, aber eben <4,1kW, ist es noch teurer diesen einzuspeisen statt ins Auto zu laden. Das geht aber dann nur wenn der PV-Akku einen gewissen Beitrag dazu leistet.

Vielleicht wäre es schon ausreichend das Laden dann wieder zu starten wenn der PV-Akku "voll" ist und noch mehr als "xx"kW vom Dach kommen. Der SoC für "voll" sollte ebenfalls konfigurierbar sein. Der Unterschied zwischen "bufferSoc" und "voll" ist die gewünschte Hysterese

andig, 2023-03-29:
Es wäre sinnvoll, mal die verschiedenen Szenarien Auto/Akku voll/leer Erzeugung fluktuierend/abnehmend/Null und dem Wunschverhalten aufzustellen. Weitere ad-hoc Änderungen oder singuläre Parameter helfen uns nicht.

Stefan0875, 2023-03-29:
Interessant, dass war der Autor @binderth hier als Bug beschreibt, würde ich so erwarten.

Der Akkus ist voll, es steht aber nicht genutzte PV Energie zur Verfügung, die jetzt eingespeist  wird, was nicht gewünscht ist.
Dafür definiere ich ja einen "bufferSoC" der genutzt werden darf.
Wenn übrigens nur mal für einen kurzen Moment die Sonne raus kommt beginnt das Laden, und nutzt dann bei Bedarf den BufferSoC voll aus, auch wenn die PV Leistung sofort wieder sinkt.
Ich helfe mir auch meistens so, dass ich mal ganz kurz den Modus auf "min+PV" stelle und sofort wieder zurück auf "PV".

Man kann natürlich überlegen, so wie es @binderth schreibt, wenn der PV Output so gering ist, ob es dann noch Sinn macht.
Beispiel:

4,1kW nötig, vom Dach kommen 700W, Überschuss sind ~200W

Dann würden ja 3,9kW aus dem Akku entnommen und nur 200W PV. Bei so einem schiefen Verhältnis aus PV und Akku wäre es dann vielleicht besser die wenige Energie ins Netz zu speisen.
Hätten wir einen weiteren Parameter "minPercentageOfPVPower" oder so ähnlich. Ich persönlich könnte da aber drauf verzichten

Stefan0875, 2023-03-29:
> Es wäre sinnvoll, mal die verschiedenen Szenarien Auto/Akku voll/leer Erzeugung fluktuierend/abnehmend/Null und dem Wunschverhalten aufzustellen. Weitere ad-hoc Änderungen oder singuläre Parameter helfen uns nicht.

Das wäre ganz gut, dann könnte man ja evtl. verschiedene Modi konfigurierbar machen.

Ich versuche mal meine Wünsche kurz zu fassen:

- PV Akku wird bevorzugt geladen
- Ist der PV-Akku voll, will ich nicht einspeisen und die Ladung des Autos muss starten, auch wenn die 4,1kW nicht vom Dach kommen. Dazu habe ich unter anderem den bufferSoC

Wäre ein `bufferSocMin` und eine `bufferSoCMax` vielleicht eine Idee?

Der PVAkku wird solange geladen bis  `bufferSoCMax` erreicht ist. Wenn dann noch PV Leistung verfügbar ist startet das laden bis zu `bufferSocMin`. Sobald dieser erreicht ist wird wieder mit Priorität der PVAkku geladen bis  `bufferSoCMax`
Damit hätte man sich auch den Parameter `prioritySoC` gespart, wenn ich das richtig sehe

Ein weiteres Scenario, was ich die Tage beobachten konnte, was mit meinem Vorschlag auch behoben wäre:

1) `bufferSoc` ist 81%, bewölktes Wetter mit meistens unter 4,1kW Überschuss
2) SoC steigt auf 82%, kurzes Wolkenloch, für ein paar Sekunden mal mehr als 4,1kW Überschuss und das Laden des Autos beginnt
3) Aber nur kurz da der PV-Akku schnell wieder unter `bufferSoc` fällt, laden wird gestoppt und der PV-Akku wieder geladen
4) goto 1

Das hat ein recht häufiges ein- und ausschalten der Ladung de Autos zur Folge

Je mehr man nachdenkt umso komplizierter  wird es und umso mehr Szenarien findet man :-)

Wenn der PV Akku bevorzugt geladen werden soll (was unter anderem auch bei SoC<prioritySoc der Fall ist):
Ein PV-Akku hat auch eine max Ladeleistung. Was macht man wenn der PV Akku bevorzugt geladen werden soll (bei mir mit max. 6,5kW) aber vom Dach plötzlich 9kW zur Verfügung stehen? Der Fall ist zum Glück (bei mir) eher selten und meistens nur kurz

VolkerK62, 2023-03-29:
>Je mehr man nachdenkt umso komplizierter wird es und umso mehr Szenarien findet man :-)

Hat nicht direkt mit buffersoc zu tun, aber mit Auto vs Hausspeicher.
Ich habe einen kleinen Speicher 6kWh mit Pmax 1,2kW.
Ich behelfe mir momentan mit einer externen Regelung, wo ich das Verhalten von evcc per API beeinflusse.
Selbst da und nur für mich habe ich noch immer keine allgemeingültige Parametrierung gefunden. Das kommt immer darauf an: Wie voll ist das Auto? Was hat der Speicher noch aus der Nacht? Wieviel Sonne wird erwartet? usw.

Will sagen, wenn man es möglich vielen recht machen will, muss man sehr (zu?) viele Einstellungsmöglichkeiten zur Verfügung stellen.

andig, 2023-04-01:
> Es wäre sinnvoll, mal die verschiedenen Szenarien Auto/Akku voll/leer Erzeugung fluktuierend/abnehmend/Null und dem Wunschverhalten aufzustellen.

Ich hab noch keinen Überblick. Eine Tabelle wäre super!

JKWFamily, 2023-04-03:
Nicht ganz zum Thema aber:

Es wäre gut wenn die Priorität und der einsetzbare BufferSOC in der GUI einstellbar wäre.

Hintergrund: An einem Tag scheint die Sonne ganztägig. Der Hausspeicher lädt sich bis zum PrioritySOC voll welches bei uns meist bis kurz vor Mittag andauert und anschließend wird erst der Ladevorgang des EV gestartet (PV Modus). Meistens müssen wir aber nach dem Mittag mit dem EV los und haben dann zwar den Hausakku geladen aber nichts in den EV.

Mit der Einstellmöglichkeit könnten wir je nach Wetter und privater Nutzung selbst und bequem entscheiden wie die Priorität (an dem Tag) des Hausspeichers angewendet werden soll. Sobald wir das Haus verlassen und der EV abgesteckt ist, läd dann die Hausbatterie weiter. 

Ein Schalter welcher die Prio zwischen Ladepunkt oder Hausbatterie wechselt/ausschaltet wäre natürlich auch möglich. 

Unabhängig: BufferSOC sollte meiner Meinung nach so hinterlegt werden können, dass dieser Prozentual der verfügbaren geladenen Batteriedkapazität entspricht. 

Beispiel PrioritätSOC 30% (Hausbatteriegröße 10kw) & BufferSOC 10% von den geladenen 3KW ( min 30%PrioSOC) = 300W Buffer

Dann ist immer etwas für die zuziehenden Wolken verfügbar, ist aber begrenzt so das immer Hausspeicher verfügbar ist wenn es in die teueren Abendstunden geht.

andig, 2023-04-03:
> Hintergrund: An einem Tag scheint die Sonne ganztägig. Der Hausspeicher lädt sich bis zum PrioritySOC voll welches bei uns meist bis kurz vor Mittag andauert und anschließend wird erst der [...]

@JKWFamily das könntest Du auch ganz einfach, indem Du `prioritySoc` weg lässt- das Verhalten hast Du ja erst provoziert ;)

JKWFamily, 2023-04-03:
> @JKWFamily das könntest Du auch ganz einfach, indem Du `prioritySoc` weg lässt- das Verhalten hast Du ja erst provoziert ;)

Ja ok, wir haben uns aber angewöhnt gleich das Auto anzustecken wenn wir nach Hause kommen (steht immer auf PV). Ab und zu möchten wir aber doch erst das der Hausakku voll wird, weil es vieleicht sich zum Nachmittag zuziehen könnte und die Sonne verschwindet, Dann haben wir den Akku wenigstens schon volßl für Abends. Ist immer eine "Wettersituation" wann erst Haus und wann EV zuerst.  Daher ein Schalter :)

binderth, 2023-04-03:
> Ab und zu möchten wir aber doch erst das der Hausakku voll wird, weil es vieleicht sich zum Nachmittag zuziehen könnte und die Sonne verschwindet

aber da hast dus ja schon geschrieben. "Ab und zu" ist kein Schalter, der automatisch geht.
Dafür müsstest du z.B. per API die `prioritySoc` anzupassen - oder du schaltest in dem Fall in der GUI auf minPV!

andig, 2023-04-03:
Ich mache mal den Bezug zum Anfang des Threads:

> Hier muss eine intelligentere Regelung gefunden werden, die ein Einspeisen verhindert.

und

> Ich hab noch keinen Überblick. Eine Tabelle wäre super!

Da wir uns dem Ziel in keiner Weise nähern verschiebe ich mal nach Discussions.