evcc-io/evcc #7098: Fahrzeug API Abfragen einstellen solange kein Ladevorgang aktiv ist

MarkusGH, 2023-03-28:
Darf ich, weil es hier vielleicht gerade passt, noch einen anderen Gedanken anbringen? Oder sagt mir wo sonst.

Wenn der Wagen voll wird, könnte man nicht auch dann die ganzen Aktivitäten einstellen, also soc poll, Leistung, Phasenwechsel? (Ich schätze dass auch dann mein Wagen einschläft, soc mode charging aber ja aktiv bleibt, und der nächste soc poll wieder zum Fehler führt)

_Originally posted by @maf-soft in https://github.com/evcc-io/evcc/issues/7032#issuecomment-1483158291_

Ich möchte das stark unterstützen! 
Habe bei meinem Porsche PHEV massive Probleme mit dem 12V System (bis hin zu kaputten 12V Batterien). Eine mögliche Erklärung wäre (neben einem Versagen sämtlicher Sicherheitsmechanismen, die das 12V System vor Tiefentladung schützen), dass das Fahrzeug durch das ständige Pollen (bei mir alle 20 Minuten) nicht einschläft und so die 12V Batterie leert.

VolkerK62, 2023-03-28:
Wenn nicht geladen wird und dein Fahrzeug wird alle 20 Minuten gepollt, dann liegt der Fehler in deiner Konfiguration.
Per default wird nur beim Laden gepollt.

MarkusGH, 2023-03-28:
Würde mich freuen:

`
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: localhost
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 3s # control cycle interval

log: info
levels: 
  porsche: trace 
  lp-1: debug
  lp-2: debug
  openwb: trace

mqtt:
  broker: XXX:1883
  topic: evcc

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.4063 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.0816 # [currency]/kWh

meters:
- name: grid1
  type: movingaverage
  decay: 1 # 0.5
  meter:
    type: custom
    power:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Power 
      scale: -1  # Die Leistung wird in Watt erwartet. Falls z.b. kW geliefert wird, kann dies hier mit dem Wert 1000 angepasst werden
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    energy: 
      source: mqtt
      topic: mbmd/solaredge1-1-1/Import 
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    currents:
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L1 
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L2 
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L3 
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
- name: pv1
  type: custom
  power:
    source: mqtt
    topic: mbmd/solaredge1-1/Power 
    timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
  energy: 
    source: mqtt
    topic: mbmd/solaredge1-1/Export 
    timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
  currents:
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L1 
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L2 
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L3 
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist

chargers:
- name: openwb1
  type: openwb
  phases1p3p: true
  broker: XXX # openWB IP
  id: 1                      # loadpoint ID
- name: openwb2
  type: openwb
  phases1p3p: true
  broker: XXX # openWB IP
  id: 2                      # loadpoint ID

vehicles:
- name: ev1
  type: template
  template: porsche
  title: XXX  # Wird in der Benutzeroberfläche angezeigt
  user: XXX  # Benutzername um auf das Gerät zuzugreifen
  password: XXX  # Passwort des Benutzerkontos
  vin: XXX# Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind
  capacity: 17.9  # Akku-Kapazität in kWh
  phases: 2
  minCurrent: 6
  maxCurrent: 16
  cache: 20m

loadpoints:
- title: Carport
  charger: openwb2
  resetOnDisconnect: true
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  enable:
    threshold: -600
    delay: 1m
  disable:
    threshold: 800
    delay: 3m
  guardduration: 5m
- title: Straße
  charger: openwb1
  resetOnDisconnect: true
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  enable:
    threshold: -600
    delay: 1m
  disable:
    threshold: 800
    delay: 3m

site:
  title: H73
  meters:
    grid: grid1
    pvs:
    - pv1
  residualPower: -150 
#  maxCurrent: 63

influx:
  url: http://localhost:8086
  database: evcc
`

VolkerK62, 2023-03-28:
... und `interval: 3s` ist auch ziemlich heftig.

MarkusGH, 2023-03-28:
Um Diskussionen aus dem Weg zu gehen habe ich das alles auskommentiert und den Dienst neu gestartet..
Jetzt erfolgen die API Abfragen wie zu erwarten war alle 15 Minuten.
Und nun?

StefanSchoof, 2023-03-28:
Ich denke, die Anfragen werden noch gemacht, damit der Climater Status festgestellt werden kann und die Ladung bei aktiven Climater frei gegeben wird.

MarkusGH, 2023-03-28:
Das Problem liegt in climateActive - im PV und MinPV mode wird climateActive  permanent gepollt.
Und das führt dann zu den Abfragen.
1) Gibt es im Porsche API keine Funktion um abzufragen, ob das Fahrzeug gerade schläft?
2) Warum wird der Climater State nicht über die Kommunikation mit der Ladestation geprüft? Anstatt das Auto alle 15 Minuten zu wecken könnte man genauso gut kurz die Ladung freigeben und schauen, ob man im StatusC oder StatusD landet.

premultiply, 2023-03-28:
Das wird nicht signalisiert. Status D war die obligatorische Raumbelüftungsanforderung beim Laden von Fahrzeugen mit Bleiakkus zur Vermeidung von hochexplosiven Knallgasansammlungen.
Hat mit Vorklimatisierung nichts zu tun und würde zudem eine vorherige Ladefreigabe seitens des Ladepunkte erfordern.

MarkusGH, 2023-03-28:
OK - ich ziehe meine Frage 2 zurück.
Aber so wie das jetzt implementiert ist verursacht es potenziell massive Probleme.

premultiply, 2023-03-28:
/cc @andig

andig, 2023-03-29:
> Habe bei meinem Porsche PHEV massive Probleme mit dem 12V System (bis hin zu kaputten 12V Batterien). Eine mögliche Erklärung wäre (neben einem Versagen sämtlicher Sicherheitsmechanismen, die das [...]

Das Problem ist im Log ersichtlich: Abfrage der Vorklimatisierung. Ich hatte bereits vorgeschlagen, die Funktion auszubauen, ist aber auf massiven Widerstand gestossen. Weitere Probleme mit fehlerhaften APIs in https://github.com/evcc-io/evcc/issues/7062. Wir können die Diskussion gerne wieder auf machen. Die Funktion macht in Summe mehr Probleme als dass sie hilft.

Alternativ: Configoption um das Feature extra einzuschalten. Das würde ich aber nur ungern machen.

pauxus, 2023-03-30:
Oder eben die SoC Schwelle (#6588), dann geht beides, und vermutlich ist es auch nicht mehr Aufwand. Wenn SoC über Schwelle, dann braucht die API auch garnicht abzufragen. Ist die Schwelle bei 0 ist die Klimatisierungsregelung damit deaktiviert, bei 100 ist das aktuelle Verhalten. Und dazwischen für diejenigen, die mehr "tunen" wollen (Wenn das Auto bei 80% ist, keine Sonne da ist, ich morgen aber nur geschätzt 40% verbrauche und übermorgen ein "Stromtag" ist, kann ich auch locker aus der Batterie Klimatisieren, also Stelle ich die Klimatisierungsschwelle auf 60%.

premultiply, 2023-03-30:
Das funktioniert so eher nicht, da dann z. B. "rückwirkende" Änderungen via Fahrzeug-API und Fehlberechnungen des Estimators auf Grund von whatever nicht mehr berücksichtigt und korrigiert würden.

Ich denke die Kopplung an den Abfragemodus ist ein guter Kompromiss. Wer das auf "charging" stehen hat bekommt keine Abfragen (und Vorklima-Freischaltung) in anderen Zuständen. Wer mehr will muss halt auf "connected" oder "always" stellen und das Intervall ggf. noch indiviuell anpassen.

andig, 2023-04-01:
Habt ihr Euch das so wir im PR vorgestellt? Der wesentliche Punkt ist hier: https://github.com/evcc-io/evcc/pull/7151/files#diff-327717b3100908ca99e06df09e2e631ed2925f9a995d5e7ef3dcd0a036af6948R318. Im Kern: Klimaabfrage bei verbundenem Fahrzeug nur wenn `poll.mode: connected/always`.

Der restliche Diff ist egtl. nur so groß weil ein paar Funktionen ihren Platz geändert haben um den Loadpoint etwas auszudünnen.

MarkusGH, 2023-04-01:
Wie geschrieben fände ich es besser eine Einstellung am Fahrzeug hinzuzufügen, aber der PR hilft mir auch.

In Zeile 322 sind m.E. die Klammern falsch:

case (lp.Soc.Poll.Mode == pollConnected || lp.Soc.Poll.Mode == pollAlways) && lp.connected():

Richtig wäre:

case lp.Soc.Poll.Mode == pollAlways || (lp.Soc.Poll.Mode == pollConnected && lp.connected()):

andig, 2023-04-01:
Ne, das is so gemeint. Klimaabfrage bei "always" kann ich mir ja sparen wenn kein Auto eingesteckt ist.

> Wie geschrieben fände ich es besser eine Einstellung am Fahrzeug hinzuzufügen,

Isso, wäre aber eine größere Änderung. Erstmal nix Schnelles.

MarkusGH, 2023-04-01:
Stimmt. Dann passt das natürlich.

closed 2023-04-01 by andig