evcc-io/evcc #7117: Tesla API: 408 Request Timeout

rwenergy, 2023-03-30:
### Describe the bug

Tesla API wont answer on Request
I have two Tesla vehicles on my account.

Works:
GET https://owner-api.teslamotors.com/api/1/vehicles works

Does not work:
GET https://owner-api.teslamotors.com/api/1/vehicles/xxx25/data_request/x 

Maybe the vehicleid (xxx25) is wrong?!

### Steps to reproduce

evcc vehicle --name ev1 -l trace

### Configuration details

```yaml
vehicles:
- type: template
  template: tesla
  title: xxx1
  accessToken: xxx1
  refreshToken: xxx2
  capacity: 60
  phases: 3
  mode: pv
  minSoC: 5
  targetSoC: 95
  minCurrent: 6
  maxCurrent: 16
  cloud: false
  name: ev1
  vin: corret_VINxxx1
```

### Log details

```text
root@raspberrypi:~# evcc vehicle --name ev1 -l trace
[main  ] INFO 2023/03/30 12:59:29 evcc 0.114.1
[main  ] INFO 2023/03/30 12:59:29 using config file: /etc/evcc.yaml
[db    ] INFO 2023/03/30 12:59:29 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/03/30 12:59:29 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/03/30 12:59:29 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/03/30 12:59:29 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/03/30 12:59:29 SELECT * FROM `settings` 6 <nil>
[tesla ] TRACE 2023/03/30 12:59:29 POST https://auth.tesla.com/oauth2/v3/token
[tesla ] TRACE 2023/03/30 12:59:30 client_id=ownerapi&grant_type=refresh_token&refresh_token=***
--
{"access_token":"xxxxx","id_token":"xxxxx","expires_in":28800,"token_type":"Bearer"}
[tesla ] TRACE 2023/03/30 12:59:30 GET https://owner-api.teslamotors.com/api/1/vehicles
[tesla ] TRACE 2023/03/30 12:59:30 {"response":[{"id":929695938787201,"vehicle_id":xxx24,"vin":"xxx77","display_name":"xxx1","option_codes":null,"color":null,"access_type":"OWNER","tokens":["xxx3b","xxx73"],"state":"asleep","in_service":false,"id_s":"xxx01","calendar_enabled":true,"api_version":52,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false},{"id":1493045188765025,"vehicle_id":252472737376,"vin":"LRW3E7FA4MC344195","display_name":"xxx2","option_codes":null,"color":null,"access_type":"OWNER","tokens":["xxx89","xxxf5"],"state":"asleep","in_service":false,"id_s":"xxx25","calendar_enabled":true,"api_version":54,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}],"count":2}
[tesla ] TRACE 2023/03/30 12:59:30 GET https://owner-api.teslamotors.com/api/1/vehicles/xxx25/data_request/charge_state
[tesla ] TRACE 2023/03/30 12:59:30 {"response":null,"error":"vehicle unavailable: {:error=>\"vehicle unavailable:\"}","error_description":""}
[tesla ] TRACE 2023/03/30 12:59:33 GET https://owner-api.teslamotors.com/api/1/vehicles/xxx25/data_request/vehicle_state
[tesla ] TRACE 2023/03/30 12:59:33 {"response":null,"error":"vehicle unavailable: {:error=>\"vehicle unavailable:\"}","error_description":""}
[tesla ] TRACE 2023/03/30 12:59:33 GET https://owner-api.teslamotors.com/api/1/vehicles/xxx25/data_request/drive_state
[tesla ] TRACE 2023/03/30 12:59:33 {"response":null,"error":"vehicle unavailable: {:error=>\"vehicle unavailable:\"}","error_description":""}
.Soc:          408 Request Timeout
Capacity:      60.0kWh
Charge status: 408 Request Timeout
Charged:       408 Request Timeout
Range:         408 Request Timeout
Odometer:      408 Request Timeout
Finish time:   408 Request Timeout
Position:      408 Request Timeout
Target Soc:    408 Request Timeout
OnIdentified:  MaxCurrent:16, MinSoc:5, TargetSoc:95, Mode:pv, MinCurrent:6
Features:      []
```

### What type of operating system are you running?

Linux

### Version

0.114.1

Matze03, 2023-03-31:
Ich habe selbe / ähnliche Beobachtung.
Ich könnte vielleicht noch dieses Traceschnipsel, direkt nach dem Dienststart beisteuern...

Mär 31 09:07:11 evcc evcc[12225]: [lp-1  ] INFO 2023/03/31 09:07:11 vehicle updated: unknown -> Model 3 SR
Mär 31 09:07:11 evcc evcc[12225]: [lp-2  ] INFO 2023/03/31 09:07:11 vehicle updated: unknown -> Model Y LR
Mär 31 09:07:11 evcc evcc[12225]: [lp-3  ] INFO 2023/03/31 09:07:11 vehicle updated: unknown -> Seat Mii electric
Mär 31 09:07:12 evcc evcc[12225]: [lp-1  ] ERROR 2023/03/31 09:07:12 vehicle odometer: 408 Request Timeout
Mär 31 09:07:12 evcc evcc[12225]: [lp-1  ] INFO 2023/03/31 09:07:12 car disconnected
Mär 31 09:07:17 evcc evcc[12225]: [lp-2  ] ERROR 2023/03/31 09:07:17 vehicle odometer: 408 Request Timeout
Mär 31 09:07:17 evcc evcc[12225]: [lp-2  ] INFO 2023/03/31 09:07:17 car disconnected
Mär 31 09:07:22 evcc evcc[12225]: [lp-3  ] INFO 2023/03/31 09:07:22 car disconnected

In den Momenten, wo die beiden Tesla's "munter" sind, tritt das bei mir nicht auf...

premultiply, 2023-03-31:
408 -> Fahrzeug schläft, keine Daten.

Matze03, 2023-03-31:
Sollte es nicht so sein, dass EVCC nach dem Dienststart dann einfach die letzten zur Verfügung stehenden Daten annimmt? Bzw., das weiß och nicht, dass die Tesla API die letzten bekannten Daten liefert? Bei der Seat API habe ich noch nie 408er Meldungen gehabt...

premultiply, 2023-03-31:
408. Genau das ist die Antwort von Tesla im Schlafzustand. 😉

Matze03, 2023-03-31:
Gut okay, dann wäre mein beobachtetes Verhalten normal. Trotzdem meine Anregung und Frage, ob es sinnvoll wäre die zuletzt bekannten Daten zu nehmen. Da ich viel Update (Nightly) hab ich dann auch viel Error's "oben". Und ab und zu, fehlen in meinem Ladelog auch die Kilometerstände.

Doch nun zurück zur Anfrage des Themenerstellers ;-) 
Beobachtet @rwenergy das Gleiche wie ich?