evcc-io/evcc #7127: Charging via planner and EEBUS is not charging with full power after a pause

DerAndereAndi, 2023-03-31:
### Describe the bug

When using a plan in PV mode and a EEBUS compatible wallbox, charging will not resume with full power after a pause.

### Steps to reproduce

1. Use an EEBUS compatible wallbox
2. Set up a charging plan that will have a charging pause
3. Start charging and wait for the pause
4. Charging will resume with minimum power after the pause

### Configuration details

```yaml
Not relevant
```

### Log details

```text
[site  ] DEBUG 2023/03/27 05:59:44 ----
[lp-1  ] DEBUG 2023/03/27 05:59:44 charge power: 0W
[site  ] DEBUG 2023/03/27 05:59:44 pv power: 0W
[site  ] DEBUG 2023/03/27 05:59:44 battery soc: 11%
[site  ] DEBUG 2023/03/27 05:59:44 battery power: 0W
[site  ] DEBUG 2023/03/27 05:59:44 grid power: 318W
[site  ] DEBUG 2023/03/27 05:59:44 grid powers: [230 55 32]W
[site  ] DEBUG 2023/03/27 05:59:44 grid currents: [1.71 0.69 0.724]A
[site  ] DEBUG 2023/03/27 05:59:44 site power: 468W
[lp-1  ] DEBUG 2023/03/27 05:59:44 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/03/27 05:59:44 charger status: B
[lp-1  ] DEBUG 2023/03/27 05:59:44 soc estimated: 41.00% (vehicle: 41.00%)
[lp-1  ] DEBUG 2023/03/27 05:59:44 vehicle soc: 41%
[lp-1  ] DEBUG 2023/03/27 05:59:44 vehicle range: 137km
[lp-1  ] DEBUG 2023/03/27 05:59:44 planned 4h5m2s until 2023-03-27 10:00:00 +0200 CEST at 11040W: total plan duration: 4h0m15s, avg cost: 692.293
[lp-1  ] TRACE 2023/03/27 05:59:44   slot from: 2023-03-27 05:59:45 +0200 CEST to 2023-03-27 06:00:00 +0200 CEST cost 733.000
[lp-1  ] TRACE 2023/03/27 05:59:44   slot from: 2023-03-27 06:00:00 +0200 CEST to 2023-03-27 07:00:00 +0200 CEST cost 719.000
[lp-1  ] TRACE 2023/03/27 05:59:44   slot from: 2023-03-27 07:00:00 +0200 CEST to 2023-03-27 08:00:00 +0200 CEST cost 690.000
[lp-1  ] TRACE 2023/03/27 05:59:44   slot from: 2023-03-27 08:00:00 +0200 CEST to 2023-03-27 09:00:00 +0200 CEST cost 705.000
[lp-1  ] TRACE 2023/03/27 05:59:44   slot from: 2023-03-27 09:00:00 +0200 CEST to 2023-03-27 10:00:00 +0200 CEST cost 655.000
[lp-1  ] DEBUG 2023/03/27 05:59:44 plan slot too short- ignoring remaining 15s
[lp-1  ] DEBUG 2023/03/27 05:59:44 pv charge current: 0A = 0A + -0.678A (468W @ 3p)
[site  ] DEBUG 2023/03/27 05:59:54 ----
[lp-1  ] DEBUG 2023/03/27 05:59:54 charge power: 0W
[site  ] DEBUG 2023/03/27 05:59:54 pv power: 0W
[site  ] DEBUG 2023/03/27 05:59:54 battery soc: 11%
[site  ] DEBUG 2023/03/27 05:59:54 battery power: 0W
[site  ] DEBUG 2023/03/27 05:59:54 grid power: 318W
[site  ] DEBUG 2023/03/27 05:59:54 grid powers: [230 56 32]W
[site  ] DEBUG 2023/03/27 05:59:54 grid currents: [1.71 0.692 0.724]A
[site  ] DEBUG 2023/03/27 05:59:54 site power: 468W
[site  ] DEBUG 2023/03/27 06:00:04 ----
[lp-1  ] DEBUG 2023/03/27 06:00:04 charge power: 0W
[site  ] DEBUG 2023/03/27 06:00:04 pv power: 0W
[site  ] DEBUG 2023/03/27 06:00:04 battery soc: 11%
[site  ] DEBUG 2023/03/27 06:00:04 battery power: 0W
[site  ] DEBUG 2023/03/27 06:00:04 grid power: 319W
[site  ] DEBUG 2023/03/27 06:00:04 grid powers: [232 55 32]W
[site  ] DEBUG 2023/03/27 06:00:04 grid currents: [1.72 0.69 0.724]A
[site  ] DEBUG 2023/03/27 06:00:04 site power: 469W
[site  ] DEBUG 2023/03/27 06:00:14 ----
[lp-1  ] DEBUG 2023/03/27 06:00:14 charge power: 0W
[site  ] DEBUG 2023/03/27 06:00:14 pv power: 0W
[site  ] DEBUG 2023/03/27 06:00:14 battery soc: 11%
[site  ] DEBUG 2023/03/27 06:00:14 battery power: 0W
[site  ] DEBUG 2023/03/27 06:00:14 grid power: 317W
[site  ] DEBUG 2023/03/27 06:00:14 grid powers: [230 56 32]W
[site  ] DEBUG 2023/03/27 06:00:14 grid currents: [1.71 0.69 0.724]A
[site  ] DEBUG 2023/03/27 06:00:14 site power: 467W
[lp-1  ] DEBUG 2023/03/27 06:00:14 charge currents: [0 0 0]A
[... 126 lines cut ...]
[site  ] DEBUG 2023/03/27 06:01:44 grid powers: [766 560 517]W
[site  ] DEBUG 2023/03/27 06:01:44 grid currents: [3.84 2.72 2.62]A
[site  ] DEBUG 2023/03/27 06:01:44 site power: 1993W
[lp-1  ] DEBUG 2023/03/27 06:01:44 charge currents: [2.2 2.2 2.1]A
[lp-1  ] DEBUG 2023/03/27 06:01:44 detected active phases: 3p
[lp-1  ] DEBUG 2023/03/27 06:01:44 charger status: C
[lp-1  ] DEBUG 2023/03/27 06:01:44 soc estimated: 41.00% (vehicle: 41.00%)
[lp-1  ] DEBUG 2023/03/27 06:01:44 vehicle soc: 41%
[lp-1  ] DEBUG 2023/03/27 06:01:44 vehicle range: 137km
[lp-1  ] DEBUG 2023/03/27 06:01:44 planned 4h5m1s until 2023-03-27 10:00:00 +0200 CEST at 11040W: total plan duration: 3h58m15s, avg cost: 692.054
[lp-1  ] TRACE 2023/03/27 06:01:44   slot from: 2023-03-27 06:01:45 +0200 CEST to 2023-03-27 07:00:00 +0200 CEST cost 719.000
[lp-1  ] TRACE 2023/03/27 06:01:44   slot from: 2023-03-27 07:00:00 +0200 CEST to 2023-03-27 08:00:00 +0200 CEST cost 690.000
[lp-1  ] TRACE 2023/03/27 06:01:44   slot from: 2023-03-27 08:00:00 +0200 CEST to 2023-03-27 09:00:00 +0200 CEST cost 705.000
[lp-1  ] TRACE 2023/03/27 06:01:44   slot from: 2023-03-27 09:00:00 +0200 CEST to 2023-03-27 10:00:00 +0200 CEST cost 655.000
[site  ] DEBUG 2023/03/27 06:01:44 telemetry: charge: Î0/17Wh @ 1506W
[site  ] DEBUG 2023/03/27 06:01:54 ----
```

### What type of operating system are you running?

Linux

### Version

0.114.1

DerAndereAndi, 2023-03-31:
The problem is that EEBUS does not know about `Enable()` and simulates it by setting the allowed current to the minimum current. If the code detects that the charging mode is `Now` it then directly sets the current to the possible maximum values.

There is no public method available to detect if a plan is active and thus set the current to the maximum.

Another approach might be to no invoke `Enable()` only, but always directly set the allowed currents afterwards. That would help chargers not having such a switch no having to care about the charging modes and their respective consequences.

DerAndereAndi, 2023-03-31:
Ich versuch das mal analog umzusetzen.

andig, 2023-04-01:
Ich macht hier mal zu. Wenn nicht klappt können wir auf Slack diskutieren. Ursache scheint ja erstmal klar zu sein.

closed 2023-04-01 by andig