evcc-io/evcc #7129: Zu Abschaltzeiten hängt evcc und zählt in Overview weiter kWh

sash2kk, 2023-03-31:
### Describe the bug
Wenn über Sperrzeiten über Mittag die Wallbox vom Netz getrennt wird und im Vorfeld eine Ladung aktiv war, wird während der Sperrzeit Einspeisung/Verbrauch/Ladepunkt in EVCC nicht mehr aktualisiert. Der Status steht weiterhin auf "Ladung" mit dem zuletzt übermittelten Wert. Auch die Charge Energy Overview zählt munter weiter während dieser Zeit (und verfälscht damit die Statistik).
Ins Auto geladen wird in der Zeit natürlich nichts, und auch in den Charging Sessions sehen die Werte danach in Ordnung aus (da bin ich mir aber nicht 100% sicher).

### Steps to reproduce
1. Ladung starten (Überschuss, PV+Min, oder Schnell)
2. 11.30 Uhr schaltet die Wallbox netzseitig ab
3. ab dann bis 12.30 Uhr steht der letzte gesendete Wert weiterhin in der GUI und keinerlei Werte aus dem ausgelesenen Modbus aktualisieren sich in EVCC. Die Modbus-Daten selbst sind außerhalb von EVCC aber weiterhin korrekt.

### Hardware
Easee Wallbox
Tasla Model 3 (über Guest Vehicle)
Solar Edge Inline Meter über Modbus

### What type of operating system are you running?
Linux

### Version
0.114.1

closed 2023-03-31 by andig

premultiply, 2023-03-31:
Du kannst nicht einfach wesentliche Bestandteile des Systems abschalten und dann erwarten dass es dann noch weiterläuft.
Zum Ladestopp gibt es den Modus "off".

sash2kk, 2023-03-31:
Ich schalte nicht einfach etwas ab, sondern die Kommunikation zur Wallbox wird netzseitig geplant abgeschalten. Kann ich denn Abschaltzeiten hinterlegen, damit er vorher auf off springt? Server, Modbus, und alles andere läuft ja ungehindert weiter und müsste doch mindestens einen Timeout erkennen können, oder?

So verrückt sind netzseitige Abschaltzeiten ja auch wieder nicht...