evcc-io/evcc #7139: Hyundai Ioniq 5 als seperates Fahrzeug aufnehmen

VolkerK62, 2023-03-31:
Beim Hyundai Ioniq 5 funktioniert die automatische Fahrzeugerkennung beim connect nicht, weil `EvStatus.BatteryPlugin` weiterhin 0 ausgibt.
Meine Idee ist, in "vehicle/bluelink/provider.go" den Eintrag  `EvStatus.BatteryPlugin > 0` durch `EvStatus.chargePortDoorOpenStatus = 1` zu ersetzen.
Eine offene Ladeklappe ist zwar nicht ganz so eindeutig wir ein eingestecktes Kabel, aber immer noch besser als nichts.

andig, 2023-04-01:
Was liefert denn das Api im Rohzustand?

VolkerK62, 2023-04-01:
```
Mär 31 21:45:45 raspberrypi evcc[28642]: [hyundai] TRACE 2023/03/31 21:45:45 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status/latest
Mär 31 21:45:45 raspberrypi evcc[28642]: [hyundai] TRACE 2023/03/31 21:45:45 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":49.309944,"lon":8.539008,"alt":0,"type":0},"head":0,"speed":{"value":0,"unit":0},"accuracy":{"hdop":0,"pdop":0},"time":"20230331195220"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":82,"batteryPlugin":0,"remainTime2":{"etc1":{"value":1,"unit":1},"etc2":{"value":335,"unit":1},"etc3":{"value":85,"unit":1},"atc":{"value":85,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":331,"unit":1},"totalAvailableRange":{"value":331,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[3],"time":{"time":"1200","timeSection":1}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"14H","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"14H","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":0,"time":{"time":"0000","timeSection":0}},"end":{"day":0,"time":{"time":"0000","timeSection":0}}},"targetSOClist":[{"targetSOClevel":80,"dte":{"rangeByFuel":{"evModeRange":{"value":321,"unit":1},"totalAvailableRange":{"value":321,"unit":1}},"type":2},"plugType":0},{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":411,"unit":1},"totalAvailableRange":{"value":411,"unit":1}},"type":2},"plugType":1}]},"chargePortDoorOpenStatus":1,"batteryPreconditioning":false,"batterySoh":0},"ign3":true,"hoodOpen":false,"transCond":false,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0,"tirePressureLampFL":0,"tirePressureLampFR":0,"tirePressureLampRL":0,"tirePressureLampRR":0},"battery":{"batSoc":87,"batState":0,"sjbDeliveryMode":0,"batSignalReferenceValue":{"batWarning":65},"powerAutoCutMode":2},"lampWireStatus":{"stopLamp":{"stopLampStatus":false,"leftLamp":false,"rightLamp":false},"headLamp":{"headLampStatus":false,"leftLowLamp":false,"rightLowLamp":false,"leftHighLamp":false,"rightHighLamp":false,"leftBifuncLamp":false,"rightBifuncLamp":false},"turnSignalLamp":{"turnSignalLampStatus":false,"leftFrontLamp":false,"rightFrontLamp":false,"leftRearLamp":false,"rightRearLamp":false}},"smartKeyBatteryWarning":false,"washerFluidStatus":false,"breakOilStatus":false,"sleepModeCheck":false,"time":"20230331201912","remoteWaitingTimeAlert":{"remoteControlAvailable":1,"remoteControlWaitingTime":168,"elapsedTime":"05:47:43"},"systemCutOffAlert":0,"tailLampStatus":0,"hazardStatus":0},"odometer":{"value":13426.5,"unit":1}}},"msgId":"edd0888e-b6a3-4c86-b2bc-4ee0384759e4"}
Mär 31 21:45:45 raspberrypi evcc[28642]: [hyundai] TRACE 2023/03/31 21:45:45 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status
Mär 31 21:46:01 raspberrypi evcc[28642]: [hyundai] TRACE 2023/03/31 21:46:01 {"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":82,"batteryPlugin":0,"remainTime2":{"etc1":{"value":1,"unit":1},"etc2":{"value":335,"unit":1},"etc3":{"value":85,"unit":1},"atc":{"value":85,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":330,"unit":1},"totalAvailableRange":{"value":330,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[3],"time":{"time":"1200","timeSection":1}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"14H","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"14H","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":0,"time":{"time":"0000","timeSection":0}},"end":{"day":0,"time":{"time":"0000","timeSection":0}}},"targetSOClist":[{"targetSOClevel":80,"dte":{"rangeByFuel":{"evModeRange":{"value":321,"unit":1},"totalAvailableRange":{"value":321,"unit":1}},"type":2},"plugType":0},{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":411,"unit":1},"totalAvailableRange":{"value":411,"unit":1}},"type":2},"plugType":1}]},"chargePortDoorOpenStatus":1,"batteryPreconditioning":false,"batterySoh":0},"ign3":true,"hoodOpen":false,"transCond":false,"steerWheelHeat":0,"si [... cut ...] ncLamp":false},"turnSignalLamp":{"turnSignalLampStatus":false,"leftFrontLamp":false,"rightFrontLamp":false,"leftRearLamp":false,"rightRearLamp":false}},"smartKeyBatteryWarning":false,"washerFluidStatus":false,"breakOilStatus":false,"sleepModeCheck":false,"time":"20230331204601","remoteWaitingTimeAlert":{"remoteControlAvailable":1,"remoteControlWaitingTime":168,"elapsedTime":"06:14:32"},"systemCutOffAlert":0,"tailLampStatus":0,"hazardStatus":0},"msgId":"1668225e-e43f-4417-9770-bc315d2ad732"}

```

andig, 2023-04-01:
Meinst Du so?

VolkerK62, 2023-04-01:
das ` | | ` ist dann ein "oder" ?

andig, 2023-04-01:
Genau

VolkerK62, 2023-04-01:
top, danke

closed 2023-04-01 by andig