evcc-io/evcc #7148: mercedes oauth funktioniert nicht mehr

simowNgithub, 2023-04-01:
### Describe the bug

Seit dem Update auf die 0.115 klappt die unter 0.114.1 funktionierende Config meiner mercedes api nicht mehr. 

### Steps to reproduce

1. configure mercedes api like evcc documentation
2. test sign in button with evcc 0.114.1 - working
3. update to evcc 0.115
4. test sign in button with evcc 0.115 - error in redirect

### Configuration details

```yaml
vehicles:
    - name: car_mercedes_glc
      type: mercedes
      title: Mercedes GLC
      vin: W1N2ABCDAbDXD
      clientId: clientid
      clientSecret: clientsecret
```

### Log details

```text
[lp-1  ] ERROR 2023/04/01 11:53:18 vehicle soc: Get "https://api.mercedes-benz.com/vehicledata/v2/vehicles/W1N2539531V295940/containers/electricvehicle": oauth2: token expired and refresh token is not set
```

### What type of operating system are you running?

Docker container

### Version

0.115

andig, 2023-04-01:
Da brauchts bitte dem kompletten Flow auf `log: trace`.

closed 2023-04-01 by andig

simowNgithub, 2023-04-01:
> Da brauchts bitte dem kompletten Flow auf `log: trace`.

@andig done, aber ich glaube man kann sich hier nichts rausziehen in Bezug auf das oauth Problem.

andig, 2023-04-01:
Siehe https://github.com/evcc-io/evcc/issues/7156, denke ich weiss grob was los ist.