evcc-io/evcc #7156: Mercedes login error - 0.115

Macrisu, 2023-04-01:
### Describe the bug

I've just made the upgrade to 0.115
I try to login for my Mercedes EQC and I get on the page this:

```
Internal Server Error
Login error
```

And the returned URL is like this:

`https://ssoalpha.dvb.corpinter.net/v1/callback?error_description=The+requested+scope+is+invalid%2C+unknown%2C+malformed%2C+or+exceeds+that+which+the+client+is+permitted+to+request.&state=xxxx&error=invalid_scope#.`

No problem in 0.114.1

Nothing was change in the config file!

### Steps to reproduce

just upgrade

### Configuration details

```yaml
x
```

### Log details

```text
x
```

### What type of operating system are you running?

Linux

andig, 2023-04-01:
Seems the switch to ne mercedes API didn‘t work. Will consult with MB developer support.

andig, 2023-04-02:
Probably caused by https://github.com/evcc-io/evcc/pull/6894. According to MB, `mb:vehicle:mbdata:payasyoudrive` does not require additional contracting. We'll need to double-check.

andig, 2023-04-09:
Reverted 31201a9 for time being. Waiting for feedback from MB.

HiroP77, 2023-04-17:
BTW: MB just confirmed, that they have an issue with their API anyway (contacted them because of [https://github.com/evcc-io/evcc/issues/7465](https://github.com/evcc-io/evcc/issues/7465)

"
Sebastian Schempp commented:

Dear [NAME](https://mercedes-benz-data.atlassian.net/secure/ViewProfile.jspa?accountId=qm%3A9f6c2438-e39c-4147-991c-a8cc17973fba%3A088ea12c-61c4-47c8-a303-04202d3ae5e0) , thank you for reaching out to us.

Unfortunately we detected a general issue for the BYOCAR API. Our developer team is on it. We will come back to you as soon as this issue has been resolved .

 

We apologize for the inconvenience

BR,

Sebastian

Sebastian Schempp changed the status to Under investigation.

[View request](https://mercedes-benz-data.atlassian.net/servicedesk/customer/portal/1/MBDS-492?sda_source=notification-email) · 
"

HiroP77, 2023-04-18:
Seems to be solved:
:

Hello [NAME](0) , thank you for your patience.

We got informed that the incident has been resolved, the API should now work again.

Please check and provide your feedback if this is not the case.

Thank you in advance

BR,

Sebastian

markus76, 2023-04-20:
Login works again.
SoC is also displayed
Target soc is adjustable

it seems to work fine at the moment... Thx.

andig, 2023-04-20:
Ahh, thanks for letting us know!

closed 2023-04-20 by andig

Sweti934, 2023-04-20:
Hei,

still get the refresh token error .- where to put this ?

https://developer.mercedes-benz.com/content-page/oauth-documentation#

According to mercedes there might be something missing with "Please note that you will have to request an additional scope offline_access to receive a refresh token." - is this done in this integration? Because I still receive this error...

andig, 2023-04-21:
It is done:

	oc := &oauth2.Config{
		ClientID:     id,
		ClientSecret: secret,
		Endpoint:     provider.Endpoint(),
		Scopes: []string{
			oidc.ScopeOpenID,
			oidc.ScopeOfflineAccess, // <---
			"mb:vehicle:mbdata:evstatus",
		},
	}

If you need help troubleshooting please provide login details for info@evcc.io.

andig, 2023-04-21:
Seems MB had API issues that have been resolved. If we want to have odometer support (which is the reverted https://github.com/evcc-io/evcc/pull/6894) I would need help:

- test if https://github.com/evcc-io/evcc/pull/6894 works as-is
- if not add the "pay as you drive" api to the MB client configuration (in the developer portal)
- restest

Does anyone want to try that?

HiroP77, 2023-04-22:
* installed 0.115.0, unfortunaltetly getting this "Internal Server error" error-message
[image]

* will now try to fiddle around with the "pay as you drive" api

HiroP77, 2023-04-22:
ok, much quicker as I expected and **seems to work**. No more error-message when registering the vehicle in evcc.
Unfortunaletly, I did no screenshots when I registered the API, but it was pretty straightforward. 
only strange thing is, that MB lists it under "insurances"
[image]

andig, 2023-04-22:
@HiroP77 inwiefern bezieht sich das auf https://github.com/evcc-io/evcc/issues/7156#issuecomment-1518037542?

HiroP77, 2023-04-23:
Ich habe wie du oben geschrieben hast, 
"
* test if https://github.com/evcc-io/evcc/pull/6894 works as-is
* if not add the "pay as you drive" api to the MB client configuration (in the developer portal)
* restest
"
Ich habe aber nur 0.115.0 als release probiert (also nicht selber gebaut), da ist ja der Odometer-Support drin.

andig, 2023-04-23:
> Ich habe wie du oben geschrieben hast,

Ahh, jetzt check ichs- Du hast den PR getestet indem Du die 115 installiert hast? Super. Heisst also: wir könnten Odometer einbauen, es müssten aber alle Mercedesnutzer ihren Client im Developerportal aktualisieren. Bin mir nicht sicher, ob es das wert ist...

HiroP77, 2023-04-23:
genau.
den echten Mehrwert sehe ich auch nicht, zumal das "insurance pay-as-you-go"-anmelden für simple Odometer-werte bei MB echt kontraintuitiv ist.
Vielleicht eher was für spätere Versionen, mit einer optionalen Aktivierung?

Chair-for-Processor-Design-TU-Dresden, 2023-05-21:
I am happy to test this feature. It should be possible to integrate it, since it does seem to work in my Openhab binding (with payasyoudrive for free). Alternatively, is it an option to use http based odometer value? One could then use the openhab Restapi to get the value.