evcc-io/evcc #7175: Directly switch 1p3p without going though disable/enable

andig, 2023-04-02:


MarkusGH, 2023-04-03:
Sollte passen, das muss die Wallbox alleine hinbekommen

MarkusGH, 2023-04-05:
#7234

foto-andreas, 2023-04-11:
Ähm, habe die Easee Home und Lade meinen Kona (2022) mit automatischem Phasenswitch. Der Link im PR-Thread dazu macht mich gerade etwas nervös. Sollte ich das besser sein lassen? Und erst recht, wenn der PR hier gemerged ist? Ich lade halt munter PV/MinPV und wenn eilig auch mitt 11kW.

MarkusGH, 2023-04-11:
Sehe kein Problem - die Wallbox muss die Ladung ohnehin vor der Umschaltung selbständig beenden. Das Problem dass sonst bei Umschaltung von 1p auf 3p ein Kurzschluss auftreten kann (bzw. wird!) weil bei 1p Ladung fahrzeugseitig die Phasen gebrückt werden dürfte verbreitet sein. OpenWB löst das Problem z. B. rein elektrisch, wenn die Netzspannung auf das Fahrzeug aufgeschaltet ist kann eine Zuschaltung der zusätzlichen Phasen schlicht nicht erfolgen.

foto-andreas, 2023-04-11:
Ok, dann bin ich wieder beruhigt. Danke!

closed 2023-05-07 by github-actions[bot]