evcc-io/evcc #7192: Sungrow schwingende Ladeleistung wenn DC Power großer AC PowerMAX

mhibu, 2023-04-03:
### Describe the bug

Habe Sungrow SH10RT, 14kWP Module, 11kW Wallbox, 11kW Auto, 13kwh Speicher.

Der Wechselrichter kann nur 9,9kW. Wenn aber 13KW vom Dach kommen weil der Speicher geladen wird, wird die Wallbox auf 11kW geschaltet, dadurch entsteht Netzbezug. Darauf hin wird der Speicher und die Wallbox runter geregelt und dann beginnt alles von neuen, es schwingt.

Kann irgendwie die maximale AC Leistung bei PV und minPV laden berücksichtigt werden? Oder werden die falschen Register aus dem Wechselrichter ausgelesen?

### Steps to reproduce

1. PV Leistung großer AC Leistung 
2. Sungrow SH10RT (Sungrow Tamplate)
3. Laden mit 11kW möglich 
...

### Configuration details

```yaml
Habe Sungrow SH10RT, 14kWP Module, 11kW Wallbox, 11kW Auto, 13kwh Speicher.
```

### Log details

```text
Unrelewant
```

### What type of operating system are you running?

Linux

### Version

1.11

closed 2023-04-03 by andig

mhibu, 2023-04-03:
Würde das bereits behoben?--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

   
  Closed #7192 as completed.

premultiply, 2023-04-03:
`maxGridSupplyWhileBatteryCharging`
`residualPower`
Siehe Doku!