evcc-io/evcc #7227: Bluelink HTTP 400 Error

VolkerK62, 2023-04-05:
### Describe the bug

seit ca. 24h funktioniert die API nicht mehr
Auch über Node Red (Bluelinky) ist keine Abfrage mehr möglich.
Scheinbar funktioniert der Login nicht mehr.

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
pi@raspberrypi:~ $ evcc vehicle --name auto --log trace
[main  ] INFO 2023/04/05 11:27:37 evcc 0.115.0 (cdc7cf06)
[main  ] INFO 2023/04/05 11:27:37 using config file: /etc/evcc.yaml
[db    ] INFO 2023/04/05 11:27:37 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2023/04/05 11:27:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/04/05 11:27:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/04/05 11:27:37 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/04/05 11:27:37 SELECT * FROM `settings` 7 <nil>
[mqtt  ] INFO 2023/04/05 11:27:37 connecting evcc-1602132420 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/04/05 11:27:37 tcp://localhost:1883 connected
[hyundai] TRACE 2023/04/05 11:27:37 retry stamps download, last attempt: 0001-01-01 00:00:00 +0000 UTC
[http  ] TRACE 2023/04/05 11:27:37 GET https://raw.githubusercontent.com/neoPix/bluelinky-stamps/master/hyundai-014d2225-8495-4735-812d-2616334fd15d.v2.json
[http  ] TRACE 2023/04/05 11:27:37 {"stamps":["dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyXChtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyXDhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyXAgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyXBgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyXOjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMybGhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMybHhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMybEgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMybFgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMybCjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMybAhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMybBhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMybOgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMybPgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyfGjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyfEhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyfFhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyfCgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyfDgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyfAjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyfOhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyfPhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyjGgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyjHgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyjEjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyjChtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyjDhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyjAgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyjBgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMyjOjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMynGhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMynHhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMynEgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMynFgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMynCjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMynAhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMynBhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMynOgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMynPgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiDGjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiDEhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiDFhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiDCgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiDDgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiDAjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiDOhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiDPhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiHGgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiHHgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiHEjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiHChtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiHDhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiHAgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiHBgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiHOjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiLGhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiLHhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiLEgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiLFgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiLCjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiLAhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiLBhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiLOgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiLPgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiPGjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiPEhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiPFhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiPCgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiPDgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiPAjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiPOhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiPPhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiTGgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiTHgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiTEjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiTChtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiTDhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiTAgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiTBgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiTOjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiXGhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiXHhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiXEgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiXFgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiXCjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiXAhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiXBhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiXOgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMiXPgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMibGjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMibEhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMibFhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMibCgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMibDgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMibAjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMibOhtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMibPhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMifGgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMifHgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMifEjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMifChtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMifDhNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMifAgtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMifBgNY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4JAlLv76lGxrWLqMbn3rTaiqQMifOjtY=","dGpcIp3dP0dLfkNRUItk/sCdmouaEMW4 [... cut ...]
[main  ] ERROR 2023/04/05 11:27:39 creating vehicle auto failed: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
Capacity: 0.0kWh
Features: [Offline]
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.115.0 (cdc7cf06)

stefanschleich, 2023-04-05:
Gleiches Problem mit einem Ioniq 5 unter v0.115.0.

```
[hyundai] TRACE 2023/04/05 11:42:55 POST https://prd.eu-ccapi.hyundai.com:8080/api/v1/user/signin
[hyundai] TRACE 2023/04/05 11:42:55 {"email":"***","password":"***"}
--
{"errId":"e9824ae6-2afa-41ec-8b99-5717072b75bc","errCode":"4003","errMsg":"Invalid values"}
[main  ] ERROR 2023/04/05 11:42:55 creating vehicle ioniq5 failed: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
Capacity: 0.0kWh
Features: [Offline]
```

VolkerK62, 2023-04-05:
Im Auto ist nix zu bestätigen. Eine Webseite ist mir nicht bekannt. Die APP funktioniert, wurde allerdings gestern aktualisiert.

emcepe, 2023-04-05:
hier seit gestern gleiches Problem:

[main  ] ERROR 2023/04/05 12:22:54 creating vehicle ev failed: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)

Hyundai Kona EV aus 2022

VolkerK62, 2023-04-05:
>Kia geht bei mir

KIA hat wohl auch Probleme
https://github.com/evcc-io/evcc/discussions/7200#discussioncomment-5530969

andig, 2023-04-05:
Please open an issue on https://github.com/Hacksore/bluelinky and link here.

emcepe, 2023-04-05:
puuh, der letzte Commit dort ist 1 Jahr her :-(

tobiasbayer, 2023-04-05:
Anscheinend hat Hyundai/Kia das mit Absicht kaputt gemacht: https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/issues/309

andig, 2023-04-05:
Und https://github.com/Hyundai-Kia-Connect/kia_uvo/issues/618

andig, 2023-04-05:
Refs https://github.com/Hacksore/bluelinky/issues/240

emcepe, 2023-04-05:
Wenn ich sowas wie in dem Post lese, könnte ich 🤮 

Die schneiden sich mit sowas ins eigene Fleisch.

Das nächste E-Auto wird dann definitiv kein Hyundai mehr, wenn die ihre API für 3rd Parties blockieren.

Wie ich diese ständigen Katz- und Mausspiele hasse 😠

Die sollten sich lieber mal um eine anständige Anbindung/Server kümmern so dass API-Requests nicht immer ewig dauern oder oft in einen Timeout laufen.

*seufz*

mhibu, 2023-04-06:
Muss ich jetzt den Kia aus der Config entfernen und zukünftig als Gastfahrzeug laden? Mit dem Error vom Bluelink Login kann ich EVCC nicht neu starten.
Das Tibber Power UP scheint jedenfalls noch aufs den Kia zugreifen zu können.

mhibu, 2023-04-06:
Der Log sagt, dass alle 10sekunden EVCC stoppt und versucht neu zu starten, was wegen Bluelink Error nicht gelingt. Die Visualisierung läuft gleichzeitig noch mit einem alten Stand.--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

   
   

   
  was meinst du damit?

andig, 2023-04-06:
Bitte neues Ticket. Nicht initialisierbare Autoa sollten egtl. kein Problem sein, das wurde schon vor langer Zeit umgesetzt.

rediculum, 2023-04-06:
Also bluelinky auf Nodered gibt diese Infos raus (die client_id hab ich maskiert):
```
nodered  | [2023-04-06 06:27:24] warn: @EuropeanBrandAuthStrategy.login: Cannot find the argument userId, please use the following url and connect from a browser.
nodered  | [2023-04-06 06:27:24] warn: https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/authorize?response_type=code&state=test&client_id=XXXXXXXXX&redirect_uri=https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/redirect
nodered  | [2023-04-06 06:27:24] error: @EuropeController.login: sign in with EuropeanBrandAuthStrategy failed with error Error: @EuropeanBrandAuthStrategy.login: Cannot find the argument userId in https://prd.eu-ccapi.kia.com:8080/web/v1/user/integration/auth?&locale=de.
nodered  |      Please use the following url and connect from a browser, it may probably ask you to "Change your password" or "Accept the new conditions"
nodered  |      Once done, try again.
nodered  |      https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/authorize?response_type=code&state=test&client_id=XXXXXXXXXX&redirect_uri=https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/redirect
```
Bringt wahrscheinlich nicht viel, denn das besagte zu Tun ändert nix. Der Fehler bleibt.
Böse Zungen behaupten ja, dass Bluelink das extra gemacht hat, weil zu viele 12V Batterien gelitten haben und die Kulanz wohl langsam erschöpft ist, denke ich  mal.

andig, 2023-04-06:
Falls open source bitte Link

foto-andreas, 2023-04-06:
Soweit ich weiß ist das weder OpenSource noch gibts einen sichtbaren Ansprechpartner dazu. Und mit dem Tibber-Support wird das nix.

emcepe, 2023-04-06:
Hi Andi,

> Du verwechselst da was. Die blockieren nichts sondern es gibt einfach keine Api. Wenn Du eine willst musst Du Mercedes kaufen. Mehr Auswahl hast Du nicht. Volvo kann das auch, will sie dich dann aber nicht benutzen lassen…

ähm ...

Du willst mir also sagen, dass Mercedes die einzige Firma ist, die eine API für EVs haben, Volvo zwar auch, nur wollen die nicht, dass man die benutzt?

Und alle anderen Hersteller wie VW, BMW, Peugeot, $you-name-it haben da nichts? Und wenn man auf die offensichtlich ja existierenden "Schnittstellen" der ganzen Hersteller zugreift, ist das weder offiziell noch gewünscht, eher geduldet?

Ich würde doch fast wetten, dass Tesla da definitiv was hat, was a) offiziell nutzbar ist und b) den anderen weit voraus ist - oder nicht?

Oder reden wir aneinander vorbei? 😀

andig, 2023-04-06:
Genau das will ich sagen solange du unter api etwas dokumentiertes und supportetes verstehst. Alles (ALLES) Andere ist mühsam reverse-engineered.

emcepe, 2023-04-06:
Oha - das hätte ich so nicht gedacht 😤 - eigentlich ein Armutszeugnis für die Hersteller.

LE4711, 2023-04-07:
Vielleicht hilft das weiter?!

Falls jemand der Sprache mächtig ist:

https://developers.kia.com/web/v1/kia/account_api
https://developers.kia.com/web/v1/kia/specification/account/account_authorize
https://developers.kia.com/web/v1/kia/specification/account

VolkerK62, 2023-04-07:
Die automatische Übersetzung von Chrome ist echt klasse ;-)

Da ist eine offene API von KIA beschrieben. Allerding nur nutzbar in Korea.
Gleiches gibt es auch von developers.hyundai.com

VolkerK62, 2023-04-07:
>das Tibber-Hyundai-Power-Up funktioniert offenbar noch/wieder

das würde doch bedeuten, dass Tibber offiziellen Zugang zu Bluelink hätte, oder?

Bluppy, 2023-04-07:
Bissel liesst sich das so das es kaputt bleiben wird.Kann ich den irgendwie so eine Art offline Auto erstellen?

Also das ich es zum laden selbst auswähle und wenigstens die KWH auf das Auto erfasse. Weil den Akkuladezustand wäre zwar nice ist aber finde ich mega wichtig.

VolkerK62, 2023-04-07:
```
- name: offline
  type: template
  template: offline
  title: ohne Bluelink
  capacity: 72
  mode: pv
```

LE4711, 2023-04-07:
wie bekomme ich bei UVO / Bluelink die Client-ID. Ich lese das so, dass diese User-spezifisch ist?!

emcepe, 2023-04-07:
die openWB kann zumindest den SoC des EV über deren Modul weiterhin auslesen.

VolkerK62, 2023-04-07:
Vorteil von ISO 15118.

VolkerK62, 2023-04-08:
es tut sich etwas :-)    https://github.com/Hacksore/bluelinky/pull/241

emcepe, 2023-04-08:
> Vorteil von ISO 15118.

Ja, das greift doch aber nur, wenn das Auto angeschlossen ist, oder irre ich mich?
Mein Auto ist seit Tagen nicht angeschlossen, weil der Akku fast voll ist und die openWB bzw. das enthaltene Kia/Hyundai Modul kann den SoC dennoch abfragen.

panzerdev, 2023-04-08:
> es tut sich etwas :-) [Hacksore/bluelinky#241](https://github.com/Hacksore/bluelinky/pull/241)
> 
> Nach einem update funktioniert es bei Node Red wieder.

Wie bei Bluelinky gefixt 

https://github.com/evcc-io/evcc/pull/7298

Flock82, 2023-04-08:
Bluelinky/EVCC loggen sich bei jedem Abruf komplett neu ein, während das OpenWB-Modul permanent eingeloggt bleibt (so wie die App). Da sich am Abruf selbst nichts geändert hat, sondern nur am Login-Prozess lief es mit der OpenWB weiter, solange sie schon eingeloggt war und nicht neu gestartet wurde.

panzerdev, 2023-04-08:
> Bluelinky/EVCC loggen sich bei jedem Abruf komplett neu ein, während das OpenWB-Modul permanent eingeloggt bleibt (so wie die App). Da sich am Abruf selbst nichts geändert hat, sondern nur am [...]

Bin mir bei evcc nicht sicher ob das wirklich so ist. Muss ich mal prüfen und ggf einen PR machen um das zu ändern.

binderth, 2023-04-08:
> es tut sich etwas :-) [Hacksore/bluelinky#241](https://github.com/Hacksore/bluelinky/pull/241)
> 
> Nach einem update funktioniert es bei Node Red wieder.

Ich habe meinen Flow in Node-Red gelöscht, dann `node-red-contrib-bluelinky` gelöscht, neu instaliert und den Flow wieder importiert. 
Ich erhalte dann im Node-Red log dieses:

```
2023-04-08T17:03:46.307780556Z	stdout	https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/authorize?response_type=code&state=test&client_id=REDACTED&redirect_uri=https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/redirect
2023-04-08T17:03:46.307746532Z	stdout	Once done, try again.
2023-04-08T17:03:46.307692342Z	stdout	Please use the following url and connect from a browser, it may probably ask you to "Change your password" or "Accept the new conditions"
2023-04-08T17:03:46.307631203Z	stdout	[2023-04-08 19:03:46] [31merror[39m: @EuropeController.login: sign in with EuropeanBrandAuthStrategy failed with error Error: @EuropeanBrandAuthStrategy.login: Cannot find the argument userId in https://prd.eu-ccapi.kia.com:8080/web/v1/user/integration/auth?&locale=en.
2023-04-08T17:03:46.307492841Z	stdout	[2023-04-08 19:03:46] [33mwarn[39m: https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/authorize?response_type=code&state=test&client_id=REDACTED&redirect_uri=https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/redirect
2023-04-08T17:03:46.306898379Z	stdout	[2023-04-08 19:03:46] [33mwarn[39m: @EuropeanBrandAuthStrategy.login: Cannot find the argument userId, please use the following url and connect from a browser.
```

Ein Klick inkl. Login auf den angegebenen URL hat leider auch nichts gebracht, meine Nodes bleiben im "ERROR". 
Wie hast du das geschafft, @VolkerK62 ?

panzerdev, 2023-04-08:
> Ich habe meinen Flow in Node-Red gelöscht, dann `node-red-contrib-bluelinky` gelöscht, neu instaliert und den Flow wieder importiert. Ich erhalte dann im Node-Red log dieses:
>
> ``` [...]

Ich denke du musst das per npm direkt machen und in der package-lock.json sicher gehen die neue Version zu haben.

binderth, 2023-04-08:
> Ich denke du musst das per npm direkt machen und in der package-lock.json sicher gehen die neue Version zu haben.

Mein Node-Red läuft im Docker und bei den bisherigen bluelinky-Updates hab ich das so gehandhabt und es hat dann gepasst, steht so auch im Readme: https://github.com/SondreNjaastad/node-red-contrib-bluelinky#what-to-do-when-bluelinky-is-updated

panzerdev, 2023-04-08:
> Mein Node-Red läuft im Docker und bei den bisherigen bluelinky-Updates hab ich das so gehandhabt und es hat dann gepasst, steht so auch im Readme: [...]

Hab ich auch in docker laufen aber das ganze node nmp Zeug liegt im Data Ordner und ist eh persistent wenn raus gemounted. In container rein gehen per exec und schon kann man `npm u` machen

VolkerK62, 2023-04-08:
Ich hab kein Docker. Habe in Node Red bei "Palette verwalten" eine update Info gehabt. Angeklickt und Node Red neu gestartet. Fertig.

Flock82, 2023-04-08:
Node-Red funktioniert aktuell nicht wegen eines Versionskonflikten ( es gibt zwei Bluelinky-Versionen, nur eine geht mit Node-Red. Diese wurde leider als letztes aktualisiert und wird somit als "latest" installiert.)

binderth, 2023-04-08:
> Ich hab kein Docker. Habe in Node Red bei "Palette verwalten" eine update Info gehabt. Angeklickt und Node Red neu gestartet. Fertig.

ah! dann hattest du noch nicht das `nodered-contrib-bluelinky`-Update von letzter Woche drin. Dann ist klar, warum es bei dir geklappt hatte. die NodeRed-Palette beinhaltet ja nicht die bluelinky-Bibliothek von hackscore, sondern ist nur ein Wrapper, der sich eine neue Version direkt zieht. daher der Umweg wie von mir oben verlinkt.

> Node-Red funktioniert aktuell nicht wegen eines Versionskonflikten ( es gibt zwei Bluelinky-Versionen, nur eine geht mit Node-Red. Diese wurde leider als letztes aktualisiert und wird somit als "latest" installiert.)

hmm... sicher? Ich hab mal NodeRed spasseshalber auf nem Rasbpi installiert und dann hatte ich auch die aktuelle bluelinky bekommen. Muss jetzt nur schauen, wie ich das auf meine produktive NodeRed kriege...

RustyDust, 2023-04-08:
@VolkerK62
Für HomeAssistant [geht's wieder](https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/pull/317)

VolkerK62, 2023-04-08:
>ah! dann hattest du noch nicht das nodered-contrib-bluelinky-Update von letzter Woche drin. Dann ist klar, warum es bei dir geklappt hatte. 

und ich hab mich schon gewundert, dass so schnell ein update vorliegt :-)

closed 2023-04-08 by andig

emcepe, 2023-04-08:
Hi @andig,

da das nun gefixt ist, wärst du so lieb und machst ein neues Release mit dem Fix? 😃

Dankeschön.

Morpheus211178, 2023-04-09:
@andig @DerAndereAndi 

Wisst Ihr schon wann ein neuer release kommt in dem dieser Fix aktiv ist ?

VolkerK62, 2023-04-11:
just for fun
Ich habe mal bei Tibber wegen Hyundai-Power-up nachgefragt, mit der Frage, wie das funktioniert, wenn es doch außerhalb von Korea keine veröffentlichte Schnittstelle gibt.
Antwort:
`danke für deine Nachricht! Die Integration der Fahrzeuge ist natürlich mit den Herstellern abgesprochen :) `

panzerdev, 2023-04-11:
> just for fun Ich habe mal bei Tibber wegen Hyundai-Power-up nachgefragt, mit der Frage, wie das funktioniert, wenn es doch außerhalb von Korea keine veröffentlichte Schnittstelle gibt. Antwort: [...]

Ich bin schon ins tägliche Rate Limit gelaufen mit 200 irgendwas requests... Würde die API anfragen nicht aus der Hand geben um ehrlich zu sein...