evcc-io/evcc #7228: Falsche Anzeige der Ladeleistung in "Live-Community"

wedojan, 2023-04-05:
### Describe the bug

Die Ladeleistung wird derzeit mit 4.295.814 kw angegeben. Das scheint mir etwas zu hoch bei 203 Fahrzeugen.

### Steps to reproduce

1. Unter Auswertung Ladeenergie
2. Im Reiter "Live-Community"

[image]

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Docker container

### Version

0.115.0

RTTTC, 2023-04-05:
> Komisch, bei mir sind es aktuell 1.061kW bei 232 von 1220 Nutzern. Das passt doch , oder?

Power seems already fixed, energy charged - not yet.

wedojan, 2023-04-05:
Die Ladeleistung wird jetzt auch wieder korrekt angezeigt.
Aber die RTTTC schreibt ist die Summe der geladenen Energie zu hoch.

29.441,5 MWh / 1224 Benutzer = 24 MWh / Benutzer  (macht nicht wirklich Sinn)

[image]

andig, 2023-04-06:
Ich bin noch dran. Ich möchte eine zweite Änderung im Server umsetzen die aber noch Testfehler produziert, daher gehts nicht so schnell wie gewünscht.

andig, 2023-04-08:
Sollte wieder passen:

[image]

War ein Ausreißer dabei den ich löschen musste. Was mir aufgefallen ist: wir sind jetzt bei Sonne schon in der Größenordnung von 1MW koordinierter grüner Ladeleistung. Nicht schlecht :)

closed 2023-04-08 by andig

wedojan, 2023-04-12:
Vielen Dank für die schnelle Lösung.
Ich bin wirklich begeistert, was ihr hier so auf die Beine gestellt habt.
Macht weiter so!