evcc-io/evcc #7246: UI in some cases shows target charging start start in the past

DerAndereAndi, 2023-04-06:
### Describe the bug

UI shows charging planner start in the past, when target SoC is already reached but target time is not.

See image:
- Time of Screenshot 6:30
- Charging target time 7:00
- Charging target 95% reached
- Text shows: "Zielladen geplant. Ladung startet heute 02:45"

[image]

### Steps to reproduce

1. Set a SoC Target
2. Use either dynamic tarif or CO2 data (I did CO2)
3. Set a target time in the future
4. The planner uses charging slots except at least the last hour slot of the target time
5. Open the UI in such a slot close to the end where SoC is reached but time is not

### Configuration details

```yaml
Nothing special
```

### Log details

```text
Might not be required, can be provided if needed
```

### What type of operating system are you running?

Linux

### Version

0.115.0

andig, 2023-04-06:
@DerAndereAndi debug Log wäre super- das darf egtl (tm) schon im Backend nicht passieren.

DerAndereAndi, 2023-04-06:
Hier das entsprechende Log. Das Laden wurde um 2:46 beendet.

[log.txt](https://github.com/evcc-io/evcc/files/11172967/188_log.txt)

closed 2023-05-02 by github-actions[bot]