evcc-io/evcc #7263: Shelly 3EM Pro: Energy-Skalierung falsch

premultiply, 2023-04-06:
Habe seid letzter Woche zu meiner Pracht Wallbox noch ein Shelly 3 EM Pro hinzugefügt.
Leider passt die Einheit geladen nicht mehr.
Vorher wurde immer in Wh angegeben Jetzt wird es in Kw/h angegeben und somit wird eigentlich zuviel geladen.
Screenshot anbei
[image]

_Originally posted by @eismarkt in https://github.com/evcc-io/evcc/discussions/7080_

andig, 2023-04-07:
/cc @thierolm

thierolm, 2023-04-10:
Ich bräuchte ein trace log von dem Gerät, dass die Ausgabe oben erzeugt ...

thierolm, 2023-04-10:
Desweiteren wäre auch die Ausgabe von `evcc -l trace meter` und ein dazugehäriger Screenshot der Shelly-App, welcher die entsprechenden oder zu korrigierenden Werte zeigt von Vorteil ...

closed 2023-04-10 by andig

eismarkt, 2023-04-11:
@thierolm  brauchst du die Daten jetzt von dem Pro noch?

thierolm, 2023-04-11:
Nein, eine Bestätigung, dass der Fix das Problem löst wäre gut. :-)

eismarkt, 2023-04-11:
Ihr schreibt hier aber vom 3em  NICHT von dem PRO??!!

thierolm, 2023-04-11:
Wir schreiben vom 3EM mit der Gen2 API. Diese wird vom 3EM PRO genutzt. Richtig?

andig, 2023-04-11:
Bitte neues Thema.

PR #7335 (2023-04-10): Shelly 3EM Gen2: TotalEnergy Unit Fix
Fixes #7263

Shelly 3EM Gen2 devices are providing total active energy on all phases in Wh!

Refer to https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/EMData#status