evcc-io/evcc #7271: Exponential Backoff für 1p/3p Phasenswitch

thomas-wolf, 2023-04-07:
**Is your feature request related to a problem? Please describe.**
Für eine Überschussladestrategie an bewölkten Tagen erlebt man recht häufig einen Switch hoch auf 3 Phasen, wenn länger als enable.delay enable.threshold Überschuss vorhanden ist, und entsprechend nach kurzer Zeit, wenn wieder eine Wolke die Sonne verdeckt, einen Switch runter auf 1p nach disable.delay. Der Phasenswitch in jede Richtung unterbricht den Ladevorgang und macht den Ladevorgang an bewölkten Tagen ineffizienter als notwendig. 

**Describe the solution you'd like**
Man kann jeden Switch von 1p von 3p als Versuch interpretieren, der als erfolgreich angesehen wird, falls nicht innerhalb von x (konfigurierbar) Minuten nach dem Switch hoch wieder ein Switch runter erfolgte. Erfolgte innerhalb dieser x Minuten wieder ein switch auf 1p, war der Versuch erfolglos.

Ferner kann man für jeden "Retry", also jeden weiteren Versuch, auf 3p zu schalten, ein Exponential Backoff implementieren, so dass der Delay für jeden weiteren Switch auf 3p zunimmt - nur wenn für diese Zeit genug Überschuss da ist, wird ein weiterer Versuch unternommen. Erfolgreiche Versuche sowie Datumswechsel setzen die Retry Strategie zurück.

Weiterhin kann man die Anzahl der Retries je Tag auf einen Maximalwert n setzen. Nach Überschreiten von n wird kein Versuch eines Phasenswitches mehr unternommen und es bleibt beim einphasigen Laden.

Wünschenswert wäre ggf. auch eine Entkopplung des Delays für den Start des Ladevorgangs vom Delay für den Versuch eines Switches auf 3p.

andig, 2023-04-07:
Was spricht denn gegen grössere enable/disable delays? Unabhängig von 1p3p hätte man da ja das identische Problem?

thomas-wolf, 2023-04-09:
Erstmal vorweg: Ich finde evcc ganz prima! Habe zwei Installationen eingerichtet und in beiden Fällen war es sehr, sehr einfach, zudem ist das Maß an Konfigurierbarkeit prima. 

Ich denke, die private Zielgruppe, die sich für eine Überschussladestrategie mit 1p/3p Switch interessiert, hat zumeist eine PV Anlage <= 10 kWhp auf dem Dach. Häufig wird man nicht mehr als 16A pro Phase nehmen, weil entweder das Auto oder die Verkabelung limitierend sind oder man auch die extrem ungleiche Lastverteilung auf die drei Phasen vermeiden möchte.

16A einphasig bedeutend ca. 3,6 kW Ladeleistung, wohingegen das Minimum für 3p 3*6A  ~= 4,2 kW ist. Die üblichen Größen von PV-Anlagen auf dem Hausdach dürften bei Bewölkung oder höherem Eigenverbrauch einen Überschuss produzieren, der doch häufiger um die 4,2 kW oszilliert, also den Schwellenwert für 3p recht häufig nach oben und unten durchbrechen.

Der Sprung der Ladeleistung durch den Switch auf 3p jedoch stellt eine Motivation dar, relativ aggressiv auf 3p zu wechseln, sobald genug Überschuss vorhanden ist. An einem heiteren Tag ist damit auch alles erzählt.

An einem bewölkten Tag oder auch an einem Tag mit wechselndem Eigenbedarf ergibt dann aber folgendes Kompromiss-Problem: Wie beschrieben, habe ich bei hohem Überschuss der PV-Anlage eine hohe Motivation, aggressiv auf 3p zu switchen. Also setze ich den delay für enable auf einen niedrigen Wert.

Wenn dann aber nicht mehr ausreichend Überschuss vorhanden ist, sei es durch Bewölkung oder durch Eigenverbrauch, wird das Delta zu 4,2 kw zugekauft. Für kurze Zeit ist das sicher in sinnvoll, aber meine unermessliche Gier, möglichst nur Solarstrom in den Wagen zu pumpen, verlangt von mir, dass ich den delay für disable dann auch auf einen vergleichsweise geringen Wert setze, um das nicht zu lange zu tun.

Damit habe ich dann eine relativ aggressive Strategie, die rasch auf 3p wechselt, aber auch bald wieder auf 1p, falls der Überschuss nicht ausreicht. Entsprechend viele Phasenswitches kann ich beobachten. Nun bedeutet mWn ein Phasenswitch aber jedes Mal einen Abbruch und Neustart des Ladevorgangs mit einer Unterbrechung von ca. einer Minute, was die Strategie ineffzienter macht, wenn der zur Verfügung stehende Überschuss stark schwankt.

An diesen Tagen mit starken Schwankungen wäre es also schön, wenn der delay Wert insbesondere für enable / Switch von 1p auf 3p nach "erfolglosen" Switches auf 3p, d.h. wenn innerhalb von einer Karenzzeit von x Minuten ein Switch zurück auf 1p erfolgen musste, sich graduell vergrößern würde, bis nach n erfolglosen Switches letztlich aufgegeben und nur noch einphasig geladen wird.

Vielleicht weniger komplex ließe sich das Thema auch lösen, indem man die delay-Werte für den Start des einphasigen Ladevorgangs und den des dreiphasigen Ladevorgangs voneinander entkoppelte. Dann kann ich den Wert für den 3p Switch relativ konservativ setzen, ohne lange auf einen Start des initialen 1p-Ladevorgangs warten zu müssen.

Frage auch zur Doku and dieser Stelle: ist der enable threshold für den Switch auf dreiphasiges Laden mit Faktor drei zu bewerten? Wenn ich den z.B. auf 300W setze, müssen dann für delay Zeit 4,2 kWh + 300W = 4,5 kWh Überschuss vorhanden sein oder 4,2 kWh + 3*300W = 5,1 kWh?

andig, 2023-04-09:
Verstehe- bin aber noch nicht überzeugt dass das eine gute Idee ist denn Dein Backoff ist ja in beiden Richtungen falsch. Due verlängerst die Einspeisung oder eben den Bezug. OT: mit Speicher als Zwischenpuffer gibts das Problem gar nicht.

bzgll. Thresholds: die schauen immer auf den Netzzähler, sind also nicht phasenabhängig.

thomas-wolf, 2023-04-09:
Gut, man sollte sich folgende Fragen stellen: wie misst man den Erfolg einer Überschussladestrategie und mit wie viel „Strafe“ möchte man dabei zugekaufte Energie versehen. 

Der erste Punkt ist vermutlich noch einfach: die beste Strategie ist diejenige, die unter gleichen Umständen am meisten Solarüberschuss in die Autobatterie bringt. 

Wie stark man dann eingekauften Strom bestrafen möchte, ist vermutlich eine individuelle Entscheidung. Rational wäre zumindest ein Faktor Preis je zugekaufter kWh / Einspeiseverguetung je kWh.

Damit vertrete ich die Meinung, dass Du mit der von mir beschriebenen Backoff Strategie im Mittel besser fährst, aber der Unterschied mag marginal sein. Hängt vielleicht auch maßgeblich davon ab, wie lange der Ladevorgang bei einem Switch unterbrochen ist. 

 Den Backoff darfst Du nicht beim Switch runter von 3p auf 1p anwenden, sondern nur beim hochswitchen, so dass Du anfangs aggressiv hochswitcht, aber wenn es sich herausstellt, dass der Überschuss doch nicht stabil genug ist, zunehmend weniger aggressiv switcht. 
 
Man müsste das mal genau durchrechnen. Aber ja, mit einem Speicher löst sich das Problem ganz elegant und man muss auch nicht zukaufen.

andig, 2023-04-10:
Es kommt halt drauf an was Du willst- nix einspeisen oder nix beziehen. Daher hängt ist die Logik spezifisch für Deinen Fall. Aus unserer Sicht erstmal keine Prio und damit wontfix.

closed 2023-04-10 by andig