evcc-io/evcc #7273: guardGracePeriodElapsed does not work as expected

MarkusGH, 2023-04-07:
### Describe the bug

guardGracePeriodElapsed checks the guard timer, but when pases are switched this timer is set to elapsed

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
NO
```

### Log details

```text
Apr  7 13:02:26 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:02:26 send openWB/set/isss/Lp2Current: '6'
Apr  7 13:02:38 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:02:38 send openWB/set/isss/Lp2Current: '13'
Apr  7 13:02:44 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:02:44 send openWB/set/isss/Lp2Current: '13'
Apr  7 13:02:50 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:02:50 send openWB/set/isss/Lp2Current: '14'
Apr  7 13:02:56 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:02:56 send openWB/set/isss/Lp2Current: '15'
Apr  7 13:03:02 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:03:02 send openWB/set/isss/Lp2Current: '16'
Apr  7 13:03:26 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:03:26 send openWB/set/isss/Lp2Current: '14'
Apr  7 13:03:32 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:03:32 send openWB/set/isss/Lp2Current: '16'
Apr  7 13:04:38 homebridge evcc[2071342]: [lp-1  ] DEBUG 2023/04/07 13:04:38 charger: guard elapse
Apr  7 13:04:38 homebridge evcc[2071342]: [lp-1  ] DEBUG 2023/04/07 13:04:38 guard timer inactive
Apr  7 13:04:38 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:04:38 send openWB/set/isss/Lp2Current: '0'
Apr  7 13:04:38 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:04:38 send openWB/set/isss/U1p3pLp2: '3'
Apr  7 13:04:38 homebridge evcc[2071342]: [lp-1  ] DEBUG 2023/04/07 13:04:38 charger: guard elapse
Apr  7 13:04:38 homebridge evcc[2071342]: [lp-1  ] DEBUG 2023/04/07 13:04:38 guard timer inactive
Apr  7 13:04:44 homebridge evcc[2071342]: [lp-1  ] WARN 2023/04/07 13:04:44 charger logic error: disabled but charging
Apr  7 13:04:44 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:04:44 send openWB/set/isss/Lp2Current: '0'
Apr  7 13:04:44 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:04:44 send openWB/set/isss/Lp2Current: '6'
Apr  7 13:04:44 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:04:44 send openWB/set/isss/Lp2Current: '6'
Apr  7 13:04:50 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:04:50 send openWB/set/isss/Lp2Current: '16'
Apr  7 13:04:57 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:04:57 send openWB/set/isss/Lp2Current: '15'
Apr  7 13:05:02 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:05:02 send openWB/set/isss/Lp2Current: '8'
Apr  7 13:05:08 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:05:08 send openWB/set/isss/Lp2Current: '6'
Apr  7 13:05:14 homebridge evcc[2071342]: [openwb] TRACE 2023/04/07 13:05:14 send openWB/set/isss/Lp2Current: '16'
```

### What type of operating system are you running?

Linux

### Version

NO

andig, 2023-04-07:
I have already voted for removing the timer. What is your proposal?

MarkusGH, 2023-04-07:
I would think it is not a good thing to let the charger and evcc get out of sync (for a longer period of time) without letting the user know. I suppose we need a separate timer which starts to run when things get out of sync.

andig, 2023-04-07:
Der Timeout hat nichts mit out of sync zu tun?!

andig, 2023-04-07:
Ich mach hier mal zu. Es wird nicht klar, was das Problem und was das erwartete Verhalten ist.

closed 2023-04-07 by andig

MarkusGH, 2023-04-07:
Ich hatte gehofft es wäre offensichtlich:

Die Funktion guardGracePeriodElapsed prüft den guardUpdated Timer um kurz nach Änderungen des enabled State oder der Phasen für 30 Sekunden Warnungen zu unterdrücken:

```
func (lp *Loadpoint) guardGracePeriodElapsed() bool {
	return time.Since(lp.guardUpdated) > guardGracePeriod
}
```

Das setzt voraus, dass der guardUpdated Timer bei solchen Änderungen auf die aktuelle Zeit gesetzt wird.
Zumindest beim Umschalten der Phasen ist das nicht der Fall, da der Timer - wie man im Log sieht - auf elapsed gesetzt wird.