evcc-io/evcc #7277: Daheimladen Touch Verhalten beim Anstecken des Fahrzeugs

StevieC121176, 2023-04-07:
### Describe the bug

In der config ist `mode: off` eingestellt. Ist in der Wallbox Plug & Charge = Off eingestellt, beginnt das Auto nach Anstecken und entsperren mit der RFID Karte sofort zu laden. Ist Plug&Charge = ON beginnt es sofort nach anstecken zu laden. Der Modus von EVCC `mode: off` wird nicht beachtet. 
Es ist also nicht möglich das Laden über evcc zu starten.

Ich habe eine Daheimladen Touch mit der Firmware 1.12

### Steps to reproduce

1.
2.
3.
...
.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.117  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: pv 
  modbus: tcpip 
  name: pv2
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: battery  
  modbus: tcpip 
  capacity: 12.8 
  name: battery2
- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.178.116  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid3

chargers:
- type: template
  template: daheimladen-mb 
  host: 192.168.178.74
  port: 502
  name: wallbox9
- type: template
  template: homematic 
  host: 192.168.178.68  
  port: 2010  
  device: '0******'
  standbypower: 15  
  user: *****
  password: *****
  meterchannel: 6  
  switchchannel: 3 
  integrateddevice: false  
# icon: bike  
  name: wallbox10
- type: template
  template: shelly 
  host: 192.168.178.101  
  user: *****
  password: *****
  channel: 0  
  standbypower: 15  
  integrateddevice: false  
#  icon: bike  
  name: wallbox1

vehicles:
- type: template
  template: enyaq
  title: Skoda Enyaq 80iV
  user: *****
  password: *****
  vin: *****
  capacity: 82
  phases: 3
  cache: 15
  mode: off
  minSoC: 30
  targetSoC: 99
  minCurrent: 6
  maxCurrent: 16
  identifiers:
  - *****
  name: ev6
  icon: car
- type: template
  template: skoda 
  title: Skoda Citigo iV  
  user: *****
  password: *****
  vin: *****
  capacity: 38  
  phases: 2  
  cache: 15  
  mode: off  
  minSoC: 30
  targetSoC: 99  
  minCurrent: 6  
  maxCurrent: 16  
  identifiers:
  - ***** 
  name: ev7
  icon: car
- type: template
  template: offline 
  title: E-Bike
  capacity: 0.75
  phases: 1  
  mode: off  
  minCurrent: 1
  maxCurrent: 6
  name: ev8
  icon: bike
- type: template
  template: id 
  title: Volkswagen ID5  
  user: *****
  password: *****
  vin: *****
  capacity: 82  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: Off  
  minSoc: 30  
  targetSoc: 99  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev9

loadpoints:
- title: Garage
  charger: wallbox9
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
#  Priority: 1
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
- title: Fahrrad
  charger: wallbox10
  vehicle: ev8
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
 # Priority: 2
- title: Individual
  charger: wallbox1
  vehicle: ev8
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
 # Priority: 2

site:
  title: Mein Zuhause
  meters:
    grid: grid3
    pvs:
    - pv1
    - pv2
    batteries:
    - battery2
  residualPower: 100
#  bufferSoC: 90
#  prioritySoC: 80
  bufferSoC: 100
  prioritySoC: 90

influx:
  url: http://192.168.178.80:8086
  database: evcc
  user: *****
  password: *****

modbusproxy:
  - port: 5200
    uri: 192.168.178.74:502 # IP und Port des Gerätes, das abgefragt werden soll

mqtt:
  broker: 192.168.178.80:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: *****
  password: *****

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.234 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh

  planner:
    type: grünstromindex
    zip: "*****"
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.115.0 (ceb3e4d9)

VolkerK62, 2023-04-07:
Du hast bei den Einstellungen in der Box das Nachladen aktiviert? RSDA = ON

Wird wirklich geladen, oder zeigt die Box nur "charging" an, aber der Ladestrom = 0A.

StevieC121176, 2023-04-07:
Das RSDA = On ist bei mir immer an. Es gibt noch einen Menüpunkt wo man das Authentifizieren per RFID einschalten kann (Plug&Charge ON/Off)
Normal steht das auf off (so hat es noch vor der letzten Änderung am Daheimladen Template funktioniert.
Nun lädt er sofort nach dem Anstecken (bei Plug&CHarge On) los bzw nach dem Authentifizieren per RFID bei Plug&Charge Off.
Und ja er lädt, Stromstärke geht auf 16A und am Auto zeigt er auch an das geladen wird.

VolkerK62, 2023-04-07:
Das Verhalten ist bei mir identisch. Mit der Ausnahme, dass bei mir `charge current = 0` eingestellt ist, wenn das Laden beendet wurde. Dadurch fängt das Laden nach dem einstecken erst an, wenn evcc es erlaubt und `charge current >= 6` schickt.
Was macht evcc, wenn "unerlaubt" geladen wird?
Hast du mit dem neuen Template schon mal mit evcc geladen und darüber auch das Laden beendet?

StevieC121176, 2023-04-07:
Nach dem "unerlaubten Starten" kann ich die Wallbox problemlos mit evcc bedienen. Soll heissen wechseln der Modi (Schnell, min+PV und PV sowie das einstellen der Stromstärke (min/Max) klappt dann tadellos. Nur eben das Verhalten beim Anstecken ist nicht korrekt.
Beim stoppen über evcc ist das Verhalten noch seltsamer. Er schaltet von 16A auf 6A. Zeigt nur noch an das das Auto verbunden ist, obwohl es noch lädt. Schalte ich dann wieder auf "Schnell", wird wieder mit 16A 11KW geladen und so auch wieder angezeigt.
Ich habe heute zum ersten mal mit dem neuen Template geladen. Er ist gleich voll mal schauen was passiert wenn er fertig ist mit Laden.

StevieC121176, 2023-04-07:
Ok Fahrzeug ist fertig geladen. Ladungsende wird auch angezeigt. Nur der Modus Schalter bleibt auf Schnell stehen, müsste eigentlich auf Aus gehen.

[image]