evcc-io/evcc #7314: Tibber Pulse Grid meter timeout

naltatis, 2023-04-09:
Hallo an alle,

eine Frage an die Tibber Pulse Nutzer:
Habt ihr auch immer wieder das Problem, dass ein Grid Meter Timeout kommt?

Bei mir geht der Fehler dann auch nicht mehr weg und evcc kann nicht mehr bedient werden. Nur ein Neustart hilft. 
Auch Zielladen läuft dann nicht mehr, wie mit heute Nacht aufgefallen ist.
Ich suche später mal die Stelle im Debug-Log, aber ich hatte gestern Abend eine Zielzeit geplant. Eigentlich hätte dann nachts geladen werden sollen, das ist aber nicht passiert. Ist das Verhalten bei nicht vorhandenem Grid-meter so gewollt?

Viele Grüße und schöne Ostern

_Originally posted by @blenni in https://github.com/evcc-io/evcc/discussions/7307_

naltatis, 2023-04-09:
Problem tritt bei Verbindungsabbrüchen > 1 Minute auf.

foto-andreas, 2023-04-09:
Ich hab das Problem auch. Bin aber nicht vor Ort und kann daher nicht so recht Logs liefern. Ich kann aber mal im Bestand der letzten Tage schauen. @naltatis brauchst du was spezielles im Log. Tritt ja offenbar ohne Auto vor Ort auf. Danbn kann ich ja mal schalten: cache, site und tibber auf trace?

foto-andreas, 2023-04-09:
Hab da was...
```
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [site  ] DEBUG 2023/04/09 00:13:45 grid currents: [0 0 0]A
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [site  ] DEBUG 2023/04/09 00:13:45 pv power: 0W
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [site  ] DEBUG 2023/04/09 00:13:45 site power: 102W
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 gridPower: 102
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 gridCurrents: [0 0 0]
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 pvPower: 0
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 lp-1/chargePower: 0
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 lp-1/mode: pv
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 lp-1/chargeCurrents: [0 0 0]
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 lp-1/chargeCurrent: 0
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 lp-1/chargedEnergy: 0
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 lp-1/chargeDuration: 0s
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [lp-1  ] DEBUG 2023/04/09 00:13:45 charge meter total import: 1335.500kWh
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 lp-1/chargeTotalImport: 1335.4996597222232
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 lp-1/connected: false
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 lp-1/charging: false
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 homePower: 102
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 greenShare: 0
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 tariffEffectivePrice: 0.3315
Apr  9 00:13:45 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:13:45 lp-1/enabled: false
Apr  9 00:13:57 v220211162457168365 evcc[312913]: [pulse ] TRACE 2023/04/09 00:13:57 {"id":"44bd0b1e-***","type":"complete"} client
Apr  9 00:13:57 v220211162457168365 evcc[312913]: [pulse ] TRACE 2023/04/09 00:13:57 {"type":"connection_init","payload":{"token":"***"}} client
Apr  9 00:13:57 v220211162457168365 evcc[312913]: [pulse ] TRACE 2023/04/09 00:13:57 {"type":"connection_ack"} server
Apr  9 00:13:57 v220211162457168365 evcc[312913]: [pulse ] TRACE 2023/04/09 00:13:57 {"id":"78cb0d9e-***","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
Apr  9 00:13:57 v220211162457168365 evcc[312913]: [pulse ] TRACE 2023/04/09 00:13:57 {"id":"78cb0d9e-***","type":"complete"} server
Apr  9 00:13:57 v220211162457168365 evcc[312913]: [pulse ] TRACE 2023/04/09 00:13:57 no running subscription. exiting... client
Apr  9 00:13:57 v220211162457168365 evcc[312913]: [pulse ] TRACE 2023/04/09 00:13:57 {"id":"78cb0d9e-****","type":"next","payload":{"errors":[{"message":"unable to start stream api:e90e4c627d25417383ba*** for device e90e4c62-7d25-4173-83ba-***","locations":[{"line":1,"column":28}],"path":["liveMeasurement"]}]}} server
Apr  9 00:13:57 v220211162457168365 evcc[312913]: [pulse ] ERROR 2023/04/09 00:13:57 unexpected end of JSON input
Apr  9 00:13:57 v220211162457168365 evcc[312913]: [pulse ] TRACE 2023/04/09 00:13:57 no running subscription. exiting... client
Apr  9 00:14:00 v220211162457168365 evcc[312913]: [site  ] DEBUG 2023/04/09 00:14:00 ----
Apr  9 00:14:00 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:14:00 lp-1/chargePower: 0
Apr  9 00:14:01 v220211162457168365 evcc[312913]: [site  ] ERROR 2023/04/09 00:14:01 grid meter: timeout
Apr  9 00:14:01 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:14:01 greenShare: 0
Apr  9 00:14:01 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:14:01 tariffEffectivePrice: 0.3315
Apr  9 00:14:15 v220211162457168365 evcc[312913]: [site  ] DEBUG 2023/04/09 00:14:15 ----
Apr  9 00:14:15 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:14:15 lp-1/chargePower: 0
Apr  9 00:14:16 v220211162457168365 evcc[312913]: [site  ] ERROR 2023/04/09 00:14:16 grid meter: timeout
Apr  9 00:14:16 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:14:16 greenShare: 0
Apr  9 00:14:16 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:14:16 tariffEffectivePrice: 0.3315
Apr  9 00:14:30 v220211162457168365 evcc[312913]: [site  ] DEBUG 2023/04/09 00:14:30 ----
Apr  9 00:14:30 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:14:30 lp-1/chargePower: 0
Apr  9 00:14:31 v220211162457168365 evcc[312913]: [site  ] ERROR 2023/04/09 00:14:31 grid meter: timeout
Apr  9 00:14:31 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:14:31 greenShare: 0
Apr  9 00:14:31 v220211162457168365 evcc[312913]: [cache ] TRACE 2023/04/09 00:14:31 tariffEffectivePrice: 0.3315

```

foto-andreas, 2023-04-09:
PS: Das handshake-Trace aus dem Diskussions-Thread sehe ich nicht im Log.

naltatis, 2023-04-11:
Kannst du im log sehen ab wann die timeout Einträge gekommen sind bzw. was vorher im log ausgegeben wurde? Am besten 'Pulse' auf 'trace' setzen.

blenni, 2023-04-11:
Nein, auf debug nicht.
Im vorigen Intervall war noch alles i.O.

```
12:45:02 raspberrypi evcc[3410]: [lp-1  ] DEBUG 2023/04/11 12:45:02 charge power: 0W
12:45:02 raspberrypi evcc[3410]: [lp-2  ] DEBUG 2023/04/11 12:45:02 charge power: 0W
12:45:02 raspberrypi evcc[3410]: [site  ] DEBUG 2023/04/11 12:45:02 pv power: 1980W
12:45:02 raspberrypi evcc[3410]: [site  ] DEBUG 2023/04/11 12:45:02 grid power: -1604W
12:45:02 raspberrypi evcc[3410]: [site  ] DEBUG 2023/04/11 12:45:02 grid currents: [0 0 0]A
12:45:02 raspberrypi evcc[3410]: [site  ] DEBUG 2023/04/11 12:45:02 site power: -1604W
12:45:02 raspberrypi evcc[3410]: [lp-2  ] DEBUG 2023/04/11 12:45:02 charge currents: [0 0 0]A
12:45:02 raspberrypi evcc[3410]: [lp-2  ] DEBUG 2023/04/11 12:45:02 charge meter total import: 3838.238kWh
12:45:02 raspberrypi evcc[3410]: [lp-2  ] DEBUG 2023/04/11 12:45:02 charger status: A
12:45:12 raspberrypi evcc[3410]: [site  ] DEBUG 2023/04/11 12:45:12 ----
12:45:12 raspberrypi evcc[3410]: [lp-1  ] DEBUG 2023/04/11 12:45:12 charge power: 0W
12:45:12 raspberrypi evcc[3410]: [lp-2  ] DEBUG 2023/04/11 12:45:12 charge power: 0W
12:45:12 raspberrypi evcc[3410]: [site  ] DEBUG 2023/04/11 12:45:12 pv power: 1977W
12:45:13 raspberrypi evcc[3410]: [site  ] ERROR 2023/04/11 12:45:13 grid meter: timeout
12:45:22 raspberrypi evcc[3410]: [site  ] DEBUG 2023/04/11 12:45:22 ----
```

Ich stelle um auf Trace-Log.

Version: `evcc 0.115.0 (1a9d7caa)`

foto-andreas, 2023-04-12:
Bei mir läuft gerade nochmal ein Test mit Trace in Version 0.116.0. Ich bin mir eigentlich recht sicher, dass ich heute wieder mit der Nightly von gestern timeouts hatte. Im Log war nach normaler pulse-TRACE-Meldung im nächsten Block dann plötzlich das grid-Timeout drin. Ich habe jetzt nochmal mehr TRACE eingeschaltet und melde mich, sobald ich weitere Infos habe.

blenni, 2023-04-12:
Hab ich gemacht, aber ich befürchte, das hilft nicht viel. Wie @foto-andreas schreibt, kommen die normalen Trace-Meldugnen und dann der Timeout - mehr nicht.

```
Apr 12 10:23:15 raspberrypi evcc[9307]: [pulse ] TRACE 2023/04/12 10:23:15 {"id":"xxx","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":141.3,"lastMeterConsumption":19558.6296,"lastMeterProduction":9978.3523,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Apr 12 10:23:24 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:24 ----
Apr 12 10:23:24 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:24 charge power: 1400W
Apr 12 10:23:24 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:23:24 charge power: 0W
Apr 12 10:23:24 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:24 pv power: 1926W
Apr 12 10:23:24 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:24 grid power: -141W
Apr 12 10:23:24 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:24 grid currents: [0 0 0]A
Apr 12 10:23:24 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:24 site power: -141W
Apr 12 10:23:25 raspberrypi evcc[9307]: [lp-1  ] ERROR 2023/04/12 10:23:25 charge meter: recv timeout
Apr 12 10:23:25 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:25 charge meter total import: 2454.881kWh
Apr 12 10:23:25 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:25 charger status: C
Apr 12 10:23:26 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:26 vehicle status: C
Apr 12 10:23:26 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:26 vehicle soc: 30%
Apr 12 10:23:26 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:26 vehicle range: 13km
Apr 12 10:23:26 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:26 pv charge current: 6.61A = 6A + 0.614A (-141W @ 1p)
Apr 12 10:23:26 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:26 max charge current: 6.61A
Apr 12 10:23:34 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:34 ----
Apr 12 10:23:34 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:34 charge power: 1450W
Apr 12 10:23:34 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:23:34 charge power: 0W
Apr 12 10:23:34 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:34 pv power: 1926W
Apr 12 10:23:34 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:34 grid power: -141W
Apr 12 10:23:34 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:34 grid currents: [0 0 0]A
Apr 12 10:23:34 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:34 site power: -141W
Apr 12 10:23:34 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:23:34 charge currents: [0 0 0]A
Apr 12 10:23:34 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:23:34 charge meter total import: 3845.253kWh
Apr 12 10:23:34 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:23:34 charger status: A
Apr 12 10:23:44 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:44 ----
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:44 charge power: 1498W
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:23:44 charge power: 0W
Apr 12 10:23:44 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:44 pv power: 1931W
Apr 12 10:23:44 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:44 grid power: -141W
Apr 12 10:23:44 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:44 grid currents: [0 0 0]A
Apr 12 10:23:44 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:44 site power: -141W
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:44 charge currents: [6.71 0.001 0]A
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:44 detected active phases: 1p
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:44 charge meter total import: 2454.889kWh
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:44 charger status: C
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:44 soc estimated: 30.06% (vehicle: 30.00%)
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:44 vehicle soc: 30%
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:44 vehicle range: 13km
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:44 pv charge current: 7.23A = 6.61A + 0.614A (-141W @ 1p)
Apr 12 10:23:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:44 max charge current: 7.23A
Apr 12 10:23:54 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:54 ----
Apr 12 10:23:54 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:23:54 charge power: 1596W
Apr 12 10:23:54 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:23:54 charge power: 0W
Apr 12 10:23:54 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:54 pv power: 1932W
Apr 12 10:23:54 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:54 grid power: -141W
Apr 12 10:23:54 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:54 grid currents: [0 0 0]A
Apr 12 10:23:54 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:23:54 site power: -141W
Apr 12 10:23:54 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:23:54 charge currents: [0 0 0]A
Apr 12 10:23:54 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:23:54 charge meter total import: 3845.253kWh
Apr 12 10:23:54 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:23:54 charger status: A
Apr 12 10:24:04 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:24:04 ----
[... cut ...]
Apr 12 10:24:34 raspberrypi evcc[9307]: [site  ] ERROR 2023/04/12 10:24:34 grid meter: timeout
Apr 12 10:24:44 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:24:44 ----
Apr 12 10:24:44 raspberrypi evcc[9307]: [lp-1  ] DEBUG 2023/04/12 10:24:44 charge power: 1726W
Apr 12 10:24:44 raspberrypi evcc[9307]: [lp-2  ] DEBUG 2023/04/12 10:24:44 charge power: 0W
Apr 12 10:24:44 raspberrypi evcc[9307]: [site  ] DEBUG 2023/04/12 10:24:44 pv power: 1935W
Apr 12 10:24:44 raspberrypi evcc[9307]: [site  ] ERROR 2023/04/12 10:24:44 grid meter: timeout
```

Soll ich noch mehr auf Trace stellen?

naltatis, 2023-04-13:
Nach `2023/04/12 10:23:15` kamen dann gar keine `pulse` Logeinträge mehr?

blenni, 2023-04-13:
Nein, nichts mehr

naltatis, 2023-04-13:
Ok, das ist doof. Hat du ne Idee was um die Uhrzeit passiert ist. Hast du die Möglichkeit bei deiner Installation in diesem Zustand (grid timeout) mal kurzzeitig (> 1min) das Netzwerk (Internetverbindung) zu trennen? Wäre interessant ob der Reconnect dann funktioniert. Sieht ja aus als ob die Kommunikation abgerissen ist, die graphql Library das aber nicht mitbekommen hat.

blenni, 2023-04-13:
Ja, das ist kein Problem, das kann ich machen.

Nur ums nochmal zu verstehen, was soll ich vom Internet nehmen? Den Pi, auf dem evcc läuft, den Pulse oder beides?
Und zu dem Zeitpunkt, zu dem der Fehler aktiv ist, richtig?

naltatis, 2023-04-13:
@blenni die Verbindung von evcc zum Internet. An welcher Stelle (Router, Kabel, OS, ...) du das trennst sollte dann hoffentlich egal sein :D

foto-andreas, 2023-04-13:
Zwischenstand von mir: Seit gestern Abend kein Timeout mehr gehabt. Zum Verständnis: Die timeout-Meldung kommt aus einem anderen Thread, der feststellt, dass die verfügbaren Daten im Cache einen zu alten Timestamp haben?

foto-andreas, 2023-04-13:
Müsste man nicht dennoch zusätzlich einen erneuten Verbindungsaufbau machen für den Fall, dass die Tibber-Seite einfach aus irgendeinem Grunde nichts mehr schickt, obwohl die Verbindung da ist? Es könnte sich in deren Server doch durch einen Fehler der Update-Thread verabschieden, dem ja nicht unbedingt selbst die Verbindung gehören muss. Ich würde das Ankommen von Werten aus der Subscription nicht mit einer bestehenden Verbindung gleichsetzen.

foto-andreas, 2023-04-14:
Weiterhin keine Tibber-Timeouts mit 0.116.0 bei mir.

andig, 2023-04-14:
> Aber wenn bei Tibber der Sende-Thread abraucht kommt halt ggf. einfach nix mehr.

Genau das vermute ich auch. Die Fehlerbehebung sollte allerdings bei Tibber passieren- deshalb würde ich vorschlagen mal ein Ticket beim Support aufzumachen.

foto-andreas, 2023-04-18:
Offenbar kommt von Tibber eben nicht wieder irgendwas. Ich hatte heute Mittag wieder das gleicher Problem und hab nun zusammen mit dem Logauszug eine Anfrage an Tibber geschickt. Ob da was brauchbares kommt, sehen wir dann. 

Dennoch bin ich der Ansicht, dass evcc für den Fall, dass sagen wir mal 5 Minuten nix vom Pulse gekommen ist, der Client die Verbindung kappen und neu aufbauen sollte. Ja, Tibber ist Schuld, aber wenn morgens mein Auto nicht geladen ist, finde ich es doof, dass evcc dem gegenüber nicht fehlertolerant ist. 

Ich hatte ab 11:38 dauerhaft den timeout, vorher mehrmals kurz vorübergehend. Um 15:00 hat dann der geplante Ladevorgang nicht begonnen. 

```
Apr 18 11:38:26 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:26 grid currents: [0 0 0]A
Apr 18 11:38:26 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:26 pv power: 0W
Apr 18 11:38:26 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:26 site power: 197W
Apr 18 11:38:26 v220211162457168365 evcc[1000795]: [pulse ] TRACE 2023/04/18 11:38:26 {"id":"63546df1-2694-49eb-983b-...","type":"next","payload":{"data":{"liveMeasurement":{"power":193,"powerProduction":0,"lastMeterConsumption":12965.5461,"lastMeterProduction":-1,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Apr 18 11:38:29 v220211162457168365 evcc[1000795]: [pulse ] TRACE 2023/04/18 11:38:29 {"id":"63546df1-2694-49eb-983b-...","type":"next","payload":{"data":{"liveMeasurement":{"power":216,"powerProduction":0,"lastMeterConsumption":12965.5462,"lastMeterProduction":-1,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Apr 18 11:38:32 v220211162457168365 evcc[1000795]: [pulse ] TRACE 2023/04/18 11:38:32 {"id":"63546df1-2694-49eb-983b-...","type":"next","payload":{"data":{"liveMeasurement":{"power":210,"powerProduction":0,"lastMeterConsumption":12965.5464,"lastMeterProduction":-1,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Apr 18 11:38:35 v220211162457168365 evcc[1000795]: [pulse ] TRACE 2023/04/18 11:38:35 {"id":"63546df1-2694-49eb-983b-...","type":"next","payload":{"data":{"liveMeasurement":{"power":209,"powerProduction":0,"lastMeterConsumption":12965.5466,"lastMeterProduction":-1,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Apr 18 11:38:39 v220211162457168365 evcc[1000795]: [pulse ] TRACE 2023/04/18 11:38:39 {"id":"63546df1-2694-49eb-983b-...","type":"next","payload":{"data":{"liveMeasurement":{"power":207,"powerProduction":0,"lastMeterConsumption":12965.5468,"lastMeterProduction":-1,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Apr 18 11:38:41 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:41 ----
Apr 18 11:38:41 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:41 grid power: 207W
Apr 18 11:38:41 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:41 grid currents: [0 0 0]A
Apr 18 11:38:41 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:41 pv power: 0W
Apr 18 11:38:41 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:41 site power: 207W
Apr 18 11:38:41 v220211162457168365 evcc[1000795]: [pulse ] TRACE 2023/04/18 11:38:41 {"id":"63546df1-2694-49eb-983b-...","type":"next","payload":{"data":{"liveMeasurement":{"power":222,"powerProduction":0,"lastMeterConsumption":12965.5469,"lastMeterProduction":-1,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Apr 18 11:38:44 v220211162457168365 evcc[1000795]: [pulse ] TRACE 2023/04/18 11:38:44 {"id":"63546df1-2694-49eb-983b-...","type":"next","payload":{"data":{"liveMeasurement":{"power":213,"powerProduction":0,"lastMeterConsumption":12965.5471,"lastMeterProduction":-1,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Apr 18 11:38:56 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:56 ----
Apr 18 11:38:56 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:56 grid power: 213W
Apr 18 11:38:56 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:56 grid currents: [0 0 0]A
Apr 18 11:38:56 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:56 pv power: 0W
Apr 18 11:38:56 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:38:56 site power: 213W
Apr 18 11:39:11 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:11 ----
Apr 18 11:39:11 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:11 grid power: 213W
Apr 18 11:39:11 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:11 grid currents: [0 0 0]A
Apr 18 11:39:11 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:11 pv power: 0W
Apr 18 11:39:11 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:11 site power: 213W
Apr 18 11:39:26 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:26 ----
Apr 18 11:39:26 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:26 grid power: 213W
Apr 18 11:39:26 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:26 grid currents: [0 0 0]A
Apr 18 11:39:26 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:26 pv power: 0W
Apr 18 11:39:26 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:26 site power: 213W
Apr 18 11:39:41 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:41 ----
Apr 18 11:39:41 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:39:41 grid power: 213W
[... cut ...]
Apr 18 11:40:56 v220211162457168365 evcc[1000795]: [site  ] ERROR 2023/04/18 11:40:56 grid meter: timeout
Apr 18 11:41:11 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:41:11 ----
Apr 18 11:41:11 v220211162457168365 evcc[1000795]: [site  ] ERROR 2023/04/18 11:41:11 grid meter: timeout
Apr 18 11:41:26 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:41:26 ----
Apr 18 11:41:26 v220211162457168365 evcc[1000795]: [site  ] ERROR 2023/04/18 11:41:26 grid meter: timeout
Apr 18 11:41:41 v220211162457168365 evcc[1000795]: [site  ] DEBUG 2023/04/18 11:41:41 ----
```

Andreas

naltatis, 2023-04-19:
Ja, das Problem scheint definitiv auf der Tibber-Seite zu liegen. Ich sehe bei mir im Log genau das gleich wie bei dir - auch zur selben Zeit. Ab `2023/04/18 11:38:37` gabs bei mir keine Daten mehr von Tibber ohne dass die Netzwerkverbindung abgebrochen ist.

Heißt wir müssen sollten uns vermutlich doch um irgend eine Art Reconnect kümmern wenn dieser Fall auftritt.

andig, 2023-04-29:
Was sagt denn der Tibbersupport?

andig, 2023-05-08:
Ping @foto-andreas gabs eine Antwort vom Support? Hast Du evtl. eine Ticket# für uns?

sebnaf, 2023-05-08:
Ich lese mal mit, da ich eine TicketID bräuchte. Danke.

stna1981, 2023-05-10:
Ich habe das Problem seit heute auch mit 1.116.7. Wäre wirklich gut, wenn von evcc-Seite da ein Workaround bereitgestellt werden könnte.

foto-andreas, 2023-05-10:
PS: Wie man bei Tibber ein richtiges Support-Ticket erstellt? Keine Ahnung.... Ich kenne nur den Chat.

stna1981, 2023-05-10:
Aber wurde da nicht schon seitens evcc ein Fix eingebaut? Das müsste der hier sein: https://github.com/evcc-io/evcc/pull/7575
Der ist aber seit einiger Zeit schon offen, wie ist denn da der aktuelle Status?

naltatis, 2023-05-11:
Ja, habe ich gerade auch in der App gepusht bekommen. Konnte gleich die "evcc via UI Neustarten" Funktion ausprobieren 😬

foto-andreas, 2023-05-11:
Führt diese Situation dann auch dazu, dass beim Einstecken des Fahrzeugs sofort begonnen wird mit Vollast zu laden obwohl eine Zielladung eingestellt ist? Oder ist das eine andere Baustelle?

stna1981, 2023-05-11:
Die Meldung in der App hat ja mit dem hier beschriebenen Problem nicht wirklich was zu tun.

stna1981, 2023-05-16:
@naltatis Gibt es hier ein Update bzgl. dieses Bugs bzw. des angefangenen Fixes #7575?

andig, 2023-05-18:
Der Ball liegt bei Tibber.

sebnaf, 2023-05-18:
Ich sehe das leider ähnlich. Viele Hersteller haben diverse Probleme mit deren APIs. Die wenigsten werden wir ändern können.

mseele, 2023-06-10:
Aufgrund von https://github.com/evcc-io/evcc/discussions/8378 habe ich auf den Tibber Pulse umgestellt und habe nun auch die Timeouts (bisher jeden Tag):

```
Jun 10 08:13:31 pi evcc[7753]: [pulse ] ERROR 2023/06/10 08:13:31 unexpected end of JSON input
Jun 10 08:13:36 pi evcc[7753]: [site  ] ERROR 2023/06/10 08:13:36 grid meter: timeout
Jun 10 08:13:46 pi evcc[7753]: [site  ] ERROR 2023/06/10 08:13:46 grid meter: timeout
Jun 10 08:13:56 pi evcc[7753]: [site  ] ERROR 2023/06/10 08:13:56 grid meter: timeout
Jun 10 08:14:06 pi evcc[7753]: [site  ] ERROR 2023/06/10 08:14:06 grid meter: timeout
...bis zum Neustart
```

Was ist der aktuelle Stand zu dem Problem? Der PR dazu (https://github.com/evcc-io/evcc/pull/7575) wurde ja leider geschlossen!?!

So ist der Tibber Pulse leider keine Alternative für mich und leider noch schlechter als der Solarlog S0-Zähler. Wenn es irgendwas gibt bei dem ich hier helfen kann (Tibber anfragen oder PR testen) gerne melden.

andig, 2023-06-10:
Aus meiner Sicht https://github.com/evcc-io/evcc/issues/7314#issuecomment-1553234818. Einen Fix haben wir aktuell nicht. Hardwarelösung: Modbuszähler einbauen.

foto-andreas, 2023-06-11:
Einen zusätzlichen Zähler einzubauen kanns doch nicht sein. Ich will Tibber mit Kostenoptimierung usw. und außerdem nicht noch einen Raspi hinstellen. Aktuell läuft mein evcc für den Standort in der Cloud. Wenn der Tibber-Pulse dranhängt, ist ein IR-Leser auch keine Alternative. @naltatis gemeinsam nochmal Code/Fix ansehen. Ist ohne Kenne von  go für mich zwar nicht gerade fix an der Ampel, aber wenn Hilfe benötigt wird, ist es mir das auch wert.

PR #7330 (2023-04-10): Improve Tibber Pulse Reconnect
Now we don't stop trying to reconnect to Tibber if connection is lost. The default of go-graphql-client (and our configured value) was to abort trying after 1min.

fixes #7314
closed 2023-06-17 by andig