evcc-io/evcc #7315: EEBus Config Setup does not work on macOS

gramss, 2023-04-09:
### Describe the bug

On plain macOS 13.3 with an M1 processor the EEBus Setup does not work.

Probably no-fix as EEBus is only experimental?

### Steps to reproduce

1. Follow installation setup for macOS
2. Setup Porsche Mobile Charge Connect
3. Have questions about what to do now exactly (go on website and click on the theoretically new appearing HomeManager)
4. HomeManager does not appear and also local SKI is not shown
...

Test if Wallbox works
1. Setup everything on Debian/Raspberry
2. evcc config -> Wallbox -> Porsche Mobile Charge Connect -> SKI-ID is shown of Home Manager
3. HomeManager is shown now in Wallbox
4. Pairing works
5. evcc config just crashes afterwards when no evcc.yaml was generated before and the menu was configured wrongly.. :)
6. (Reset Wallbox as you know have 2 Homemanagers and the Wallbox does not feature a possibility to delete a HomeManager after pairing)

### Configuration details

```yaml
not existing -> initial Setup
```

### Log details

```text
does this also exist on evcc config??
```

### What type of operating system are you running?

Linux

### Version

latest

DerAndereAndi, 2023-04-09:
I am developing the EEBUS stack and support on latest macOS and testing it on macOS with M1. It works and is supported.

Sadly I don't really understand what the issue is:
- with "HomeManager" you mean a running evcc instance?
- what do you mean with "evcc config just crashes" ? What is the output?
- You can remove previously paired Energy Managers in the Porsche Mobile Charger Connect. In the web interface and the devices display

Please provide the output when "crashing". Also the version of evcc you are using.

gramss, 2023-04-09:
Thank you for responding so quickly!

With 'HomeManager' I meant a EEBus (server?) instance that is spun up either during `evcc config` or normal evcc operation with corresponding EEBus settings.

I take it back that the `evcc config` on my Mac did not create a custom Local SKI ID. It just did display the information only once during the normal `evcc config` process while creating a new `evcc.yaml` file.
Unfortunaltley, at any stage using my M1 Mac, it was not possible to find the "HomeManager" spun up by my M1-Mac.

Regarding the crash: I was just suprised that no `evcc.yaml` was created when I choose the menu option: "add a device" .
The last output of the so-called "crash" was indeed a bare configuration that could be copied into my own `evcc.yaml` but misses the Pub-Priv-Key pair:
```
evcc configure
[main  ] INFO 2023/04/09 16:00:05 evcc 0.115.0
[main  ] INFO 2023/04/09 16:00:05 evcc 0.115.0

The next steps will guide you through the creation of a configuration file for evcc.
Please note that this process cannot cover all possible scenarios.
You can cancel the process by pressing CTRL-C anytime.

NOTE: This functionality is in an experimental state!
  This means that the created configuration file may not work
  in a future update of evcc and would need to be recreated.
  We are looking forward to your feedback at https://github.com/evcc-io/evcc/discussions/

Let's go:

 In which mode should this process guide you through the configuration? Advanced mode (Ask more details, requires technical know-how)

 What do you want to do? Configure a single device (**has to be added manually to a configuration file**!)

- Setup a device

Choose one of the following device categories

 Choose one of the following device categories wallbox

- Configuration wallbox

 Choose a wallbox: Porsche Mobile Charger Connect

-- EEBUS -----------------------------------

[eebus ] INFO 2023/04/09 16:00:28 Local SKI:  3c2e622c1a30f47aa0d8641569dd0a1c696095fc

--------------------------------------------

You selected a wallbox, which will be accessed via the EEBUS protocol.
For that the wallbox needs to be connected to evcc. This can usually be done in the web interface of the wallbox.
Please press the enter key once you started the process.

Please provide the following settings:

 Subject Key Identifier (SKI) 4....cb31
 IP address 192.168.42.117

Testing the configuration of Porsche Mobile Charger Connect ...

Porsche Mobile Charger Connect Porsche Mobile Charger Connect was successfully added.

The configuration:

  type: template
  template: pmcc 
  ski: 4...cb31  
  ip: 192.168.42.117  
  name: wallbox1

```

Definitely my error, but I do think that this is missing the pub-pri keypair that is now present when creating a `evcc.yaml` from scratch through the config.

Regarding removing previously paired Energy Managers (wrongly called HomeManagers from me): I did not find this option through the "Heimanwender" user. Through the reset of the device I also gained "Kundendienst" user. Is this setting maybe only available there?

On MacOS and on Raspbian
```
$evcc --version
evcc version 0.115.0
```

DerAndereAndi, 2023-04-09:
Thanks for the additional details.

evcc does not crash, it just exists as you only added one device and the details for that is printed. I agree that in this scenario the EEBUS details are missing, but the requires larger changes as it would also require to display a more complete config output than it currently does.

As a workaround I suggest to create a full setup instead of only adding a single EEBUS device.