evcc-io/evcc #7318: Add BatterX PV+Grid+Battery Solution

gramss, 2023-04-09:
I would like to add this system from batterX: https://www.batterx.io/wp-content/uploads/2019/05/batterX-Home-Series-DE.pdf

It also includes our Frosenius Symo 7.0.3-M data (represented by "External Solar" power)
(OT: I was confused by the current documentation and old issues with stale links how to add this directly. For a first setup I would still try to integrate it before I can include the complete BatterX system, but later this would not be needed as the data would be already there in the other system)

There exists also a good JSON-based API with documentation: https://batterx.gitbook.io/batterx-livex-api/

Is there any good templates I could look into to build this myself? Or is this quite easy to setup for experienced EVCC developers with the API at hand?

Thanks and happy easter!

StefanSchoof, 2023-04-09:
The Frosenius Symo is Sunspec compatible and you can use the generic Sunspec inverter. https://docs.evcc.io/docs/devices/meters/#wechselrichter-2

andig, 2023-04-12:
@gramss Cloud APIs für kritische Messwerte sind meistens... Mist. Kann man machen, ist aber nicht schön. Gibts alternativ auch eine Lokale API? Falls nein schau Dir gerne mal https://github.com/evcc-io/evcc/blob/3ca35660c4b67280b39e02767d85881a2609657c/templates/definition/meter/sonnenbatterie.yaml an- würdest Du auf der Basis einen PR hinbekommen?

gramss, 2023-04-12:
Hi @andig , The batterX does feature a nice Standalone App with the Fronius included, but the liveX API (second link) I send is completely reachable through a local IP address.

And yes. I think I can work with the .yaml you provided.
Although some explaining might be good.
What variables can be changed and have no meaning to another underlaying system packing me in the right "meter" category? Is it just the place of the file or are there any "magic variables" I need to be aware of?

If you might be able to point me out to the evcc render engine (I suppose there is some as the sub-section is called `render`)I might have an easier time setting this thing up.. :)

And yes, I am feeling comfortable to write a PR myself. I thought I would learn some go, but fiddling around with this template style is totally fine for me 👍

VolkerK62, 2023-04-12:
>/cc @VolkerK62 vllt. auch was für die etwas anvancedtere Doku- wie baue+teste ich ein Template?

Sorry, da kann ich nicht weiterhelfen.
Im Gegenteil, so eine Erklärung hätte ich auch gerne. Hab mal versucht einen PR zu testen und bin kläglich gescheitert.
Es wäre schön, wenn es dazu eine ausführliche Beschreibung gäbe, die auch interessierte "No-Nerds" verstehen. ;-) 
Ein issue dazu gibt es https://github.com/evcc-io/docs/issues/98

gramss, 2023-04-12:
Just some notes while I look into your code / started hacking: (next post can be in German 😄 )

- ReadME:
  - VSCode Plugin is called Ventur and not Venture (link is although correct)
  - Command to install dependencies before initial `make ui` is missing (they are in package.json but I do not know how to automatically install them in go / node / vite ? (Seems to be `npm install`)
  
Regarding the `render` question I had:
There seems to be a `devicetest.go` . Maybe it would be a good idea to expose this via i.e. `evcc (device)test -d [meter|charger|vehicle] -c <config.yaml>`
  Looks like `go run meter -c <config.yaml>` does this already @andig ? (this seems to be also `evcc meter -c <config.yaml>` - 
  
  @StefanSchoof Thank you for your suggestions! This is a very good starting point. However the minimal setup to test my custom meter is quite excessive (as I found out while following those steps). I followed the documentation and found the `demo.yaml`. It took me 3-4 trail and errors until I got my first bits of custom meter to "work" (aka throw an error from my custom part of the yaml) because I only wanted the most minimal setup possible from the otherwise full-blown `demo.yaml`.
--> This is not needed with the `evcc meter -c <config.yaml>`

Okay, now I got a reading through the custom config and I also adapted another template to the batterX system.
@andig the template documentation did help to understand it better what is happening. For the custom config I recognized that I needed to write `jq: .["123”].["1"]` and not `jq: .data.tuples["123"]["1"]`. This was a bit confusing in the [HTTP plugins references](https://docs.evcc.io/docs/reference/plugins#http-lesenschreiben) for me.

Raw Data:
<details>
```
{
    "273":
    {
        "1": 23260
    },
    "305":
    {
        "1": 33
    },
    "337":
    {
        "1": 79
    }
}
```
</details>

---
But now I have a bigger problem of understanding what data evcc really needs in order to unlock "Überschussladen".

The other `allinone=true` templated solutions I saw seem to do some scripting in order to display one source of power exclusively. This is somewhat anti-intuitive for me as my system is constantly floating with a mix of energy sources (grid + solar1 + solar2 + battery). For me I would expect to give evcc all the current Energy consumed in my home at first (see the red virtual "energy meters" in the diagram below). Then separately I would add additional information about the energy that generated by the sum of `pv` + `battery` and take the delta to best-case charge my car with it.

This picture here from BatterX pretty much sums it up how my setting looks like and I would love if somebody can make some pretty circles around what Watt readings from where I should add together in which kind of "grid/power" meter for evcc :) 

[image]

And of course thank you already for the great support of pointing out the different puzzle pieces for achieving the things so far! 🥳 
( I need to find a way to include the yaml for the template I did here.. Probably best to just start a PR as a draft right now.)

premultiply, 2023-04-13:
You need to query:
- grid meter
- each solar meter behind the grid meter
- each battery meter behind the grid meter

So the trick is to filter out exactly these pure values from each device.
Never care about any calculated value like "house consumption" or mixture values like "ac power" on hybrid devices.

evcc will automatically sum up all solar and battery meters defined at the `site`.

Multiple HTTP queries to the same url will be cached so don't care about that. Its fine to "call" them on each value instance.

gramss, 2023-04-13:
@premultiply This definitely solved many many knots in my head! Thank you!! 👏 

I now got this reading with `./evcc meter -c batterX-template.yaml`:
```
my_grid
-------
Power: 29W

my_pv
-----
Power: 3578W

my_battery
----------
Power:    -980W
Soc:      61%
Capacity: 10.0kWh
```
And I finally understand the `if eq .usage "pv` part of the template approach. This is not something that the system says what is used but rather `evcc` now looks at my device from different angles. This was confusing for me but cleared away through your comment!

Does it make sense to add more vales like different `currents` on phases per meter for `evcc` ?

Also, I hope I (inverted) the sign for the battery correctly? In the reading above the battery is currently charging.
The import into the battery is by the system positive. I manually added the singed part but by your statement @premultiply I should have in this scenario a positive W number for the battery while the battery is receiving energy (by solar).

gramss, 2023-04-15:
> Also, I hope I inverted (on purpose) the sign for the battery correctly? In the reading above the battery is currently charging. The import into the battery is by the system defined as positive. I manually added the inverting of the sing but by your statement @premultiply I should have in this scenario a positive W number for the battery while the battery is receiving energy (by solar)?

Based on the implementation of the [Sonnenbatterie.yaml](https://github.com/evcc-io/evcc/blob/3ca35660c4b67280b39e02767d85881a2609657c/templates/definition/meter/sonnenbatterie.yaml) cross checking with the [API](https://jlunz.github.io/homeassistant/#/api/getApiV2Status) of the Sonnenbatterie-10, it seems that charging a battery needs to be counted as MINUS/negative Watt.

> 
> Pac_total_W | -5 | W, watt | AC Power greater than ZERO is discharging Inverter AC Power less than ZERO is charging | Signed Integer
> -- | -- | -- | -- | --

With this, my template seems to be good to go and ready for a PR.

andig, 2023-04-15:
Then lets go :)

gramss, 2023-04-15:
Just missing the last bit..
I have implemented that external solar (behind grid meter) and solar directly connected to batterX are always summed.

I wanted to add something like that:
```
   {{- if eq true .externalpv }}
    jq: .["2913"].["3"] + .["1634"].["0"] # External Solar + BatterX Solar (Power Total in W)
   {{- end }}
   {{- if .externalpv }}
   {{- else }}
    jq: .["1634"].["0"] # BatterX Solar (Power Total in W)
   {{- end }}
```

With an additional parameter:
```
-name: externalpv
 default: false
```

But somehow `.externalpv` was not of the same type as "false" or "true":
```
[main  ] FATAL 2023/04/15 13:29:26 cannot create meter 'my_pv': cannot create meter 'template': template: eebus.tpl:11:7: executing "eebus.tpl" at <eq true .externalpv>: error calling eq: incompatible types for comparison
```

Any hints what am I missing? I am probably outside of the spec of your render?

andig, 2023-04-15:
„true“

needs be string due to the internal structure.

PR #22476 (2025-07-28): Add batterX Home
Fix #7318

Dieser PR hier sollte nun endlich die initial erstellten Features aus dem issue https://github.com/evcc-io/evcc/issues/7318 bringen.

Battery-Control habe ich auf Basis des aktuellen top sonnenbatterie templates auch erstellen können.

Das System wurde auf der neusten v205 evcc in Kombination mit realen Strompreisen und einem dummy Loadpoint getestet.

Umschaltung von `Normal` auf `Charge` konnte ich durchführen. 
`Hold` hab ich ohne ladendes Fahrzeug leider nicht testen können.

cc:
@Engineer-Andre

Background warum der PR so lange gedauert hat:
Während des Testpha [... cut]