evcc-io/evcc #7322: The experimental feature to set the minimum current in the UI does not support lower values than 6A

DerAndereAndi, 2023-04-09:
### Describe the bug

If a wallbox and EV support lower values than 6A as minimum current, the UI should allow to support these values. Currently it does not. Also the minimum power is calculated with 1 phase, even when the loadpoint has 3 phases defined.

[image]

### Steps to reproduce

1. Install a nightly build
2. Enable experimental features in the UI setting
3. Use a wallbox and EV combination with minimum current support <6A (ISO15118-2)

### Configuration details

```yaml
- Wallbox with ISO15118-2 support like Porsche Mobile Charger Connect
- EV with ISO15118-2 (AC) support like Porsche Taycan
```

### Log details

```text
None
```

### What type of operating system are you running?

Linux

### Version

0.115.0

cschlipf, 2023-04-14:
Would be great to set lower values. For example a Tesla can charge at much lower currents, thus eliminating the need for phase switching: https://www.youtube.com/watch?v=XY9k1-uDhLw

DerAndereAndi, 2023-04-14:
@cschlipf As evcc sets the maximum currents in the wallbox and the Tesla uses IEC61851 to communicate with the wallbox, no values lower than 6A can be set. To do this with a Tesla, you would need to set the maximum current in the EV and not in the Wallbox. This issue is not about this.

TobiasSt78, 2023-04-15:
Anyway, between 1st value and 2nd value there is a calculation mistake in power calculation. 
First value (max) calculates with 3 phases, min value calculates with 1phase. It is not reasonable why a 1 phase charger should have 6 amps but gives two different results.
[image]

TobiasSt78, 2023-04-15:
Same happens with Shelly plugs who definitly have only 1 phase
[image]

closed 2023-05-12 by github-actions[bot]