evcc-io/evcc #7323: Hyundai Bluelink Connection error

PVWolf, 2023-04-09:
### Describe the bug

My Vehicle is not displayed in the webpage for choosing.

[image]

If I try the test for vehicle configuration I get this:

sudo evcc -l debug vehicle
[main  ] INFO 2023/04/09 21:09:53 evcc 0.115.0
[main  ] INFO 2023/04/09 21:09:53 using config file: /etc/evcc.yaml
[db    ] INFO 2023/04/09 21:09:53 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/04/09 21:09:53 connecting 114 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/04/09 21:09:53 tcp://localhost:1883 connected
[main  ] ERROR 2023/04/09 21:09:55 creating vehicle IONIQ failed: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
Capacity: 0.0kWh
Features: [Offline]

The konfiguration for the vehicel worked befor (middel of March) and doesn't after I come back from my onside commisioning. In this time I've changed nothing. While going trough the blogs I read something about an App update of Bluelink and other Users getting problem's with this too. I also made the update from v 0.114.0 to v 0.115.0 after I discovered that something is wrong with the vehicle communication, but doesn't help.

This is my config of my Hyundai Ioniq:

vehicles:
- type: template
  template: hyundai
  title: IONIQ
  user: ****@gmail.com
  password: *******
  vin: *****
  capacity: 39
  phases: 1
  language: de
  mode: pv
  minSoC: 15
  targetSoC: 95
  minCurrent: 6
  maxCurrent: 30
  name: IONIQ

### Steps to reproduce

1. 
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: ***

interval: 1s # control cycle interval

sponsortoken: ***

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.***.***
  name: SMA_Homemanager_2_0
- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.***.***
  password: ***
  name: PV_SMA_Tripower_6

chargers:
- type: template
  template: smaevcharger
  host: 192.168.***.***
  user: ***
  password: ***
  name: EVCharger

vehicles:
- type: template
  template: hyundai
  title: IONIQ
  user: ***@gmail.com
  password: ***
  vin: ***
  capacity: 39
  phases: 1
  language: de
  mode: pv
mode: pv
  minSoC: 15
  targetSoC: 95
  minCurrent: 6
  maxCurrent: 30
  name: IONIQ

loadpoints:
- title: EVCharger
  charger: EVCharger
  vehicles: IONIQ
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 31
  resetOnDisconnect: true

site:
  title: Wolfshöhle
  meters:
    grid: SMA_Homemanager_2_0
    pvs:
    - PV_SMA_Tripower_6

hems:
  type: sma
  AllowControl: false

mqtt:
  broker: localhost:1883
  topic: evcc
  user: ****
  password: ****
  clientid: 114

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.43 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.07 # [currency]/kWhtariffs:

telemetry: true # default: false
```

### Log details

```text
[main  ] INFO 2023/04/09 21:22:11 evcc 0.115.0
[main  ] INFO 2023/04/09 21:22:11 using config file: /etc/evcc.yaml
[main  ] INFO 2023/04/09 21:22:11 starting ui and api at :7070
[db    ] INFO 2023/04/09 21:22:12 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/04/09 21:22:12 connecting *** at tcp://localhost:1883
[mqtt  ] DEBUG 2023/04/09 21:22:12 tcp://localhost:1883 connected
[main  ] ERROR 2023/04/09 21:22:15 creating vehicle IONIQ failed: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
[lp-1  ] WARN 2023/04/09 21:22:15 vehicles option is deprecated
[semp  ] WARN 2023/04/09 21:22:15 SEMP_BASE_URL unspecified, using http://192.168.***.***:7070 instead
[site  ] INFO 2023/04/09 21:22:15 site config:
[site  ] INFO 2023/04/09 21:22:15   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/04/09 21:22:15     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/04/09 21:22:15     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/04/09 21:22:15   vehicles:
[site  ] INFO 2023/04/09 21:22:15     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/04/09 21:22:15 loadpoint 1:
[lp-1  ] INFO 2023/04/09 21:22:15   mode:        pv
[lp-1  ] INFO 2023/04/09 21:22:15   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/04/09 21:22:15   meters:      charge ✓
[lp-1  ] INFO 2023/04/09 21:22:15     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/04/09 21:22:15 phase timer inactive
[lp-1  ] DEBUG 2023/04/09 21:22:15 pv timer inactive
[lp-1  ] DEBUG 2023/04/09 21:22:15 guard timer inactive
[main  ] FATAL 2023/04/09 21:22:16 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.115.0

StefanSchoof, 2023-04-09:
I think this is  https://github.com/evcc-io/evcc/issues/7227

PVWolf, 2023-04-09:
What 7227 means?

StefanSchoof, 2023-04-09:
Click the link. It is a issue with the same topic

andig, 2023-04-09:
Problem ist im nightly gelöst.

closed 2023-04-09 by andig

zickebua, 2023-04-11:
gibt es nightly für homeassistant??