evcc-io/evcc #7328: Support: Sungrow connection: "missing loadpoints"

sterlp, 2023-04-10:
### Describe the bug

I can for some reason not connect to the Sungrow SH10RT using the default configuration.

Connection with the modbus CLI works - the network is as I found connected to the SH10RT and not to the WIFI stick.

```
❯ modbus 192.168.178.165:502 -s 1 4999
Parsed 0 registers definitions from 1 files
4999: 2023 0x7e7
```

Running on mac using brew.

### Steps to reproduce

1. Install evcc
2. Run configuration (simple)
3. start evcc

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 36ecf06b24816610dc033fcc2863407ad8c2a72f9e3f5830853346839a1edffb

interval: 10s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.165
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.165  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.165  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3

site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
❯ ./evcc --log trace
[main  ] INFO 2023/04/09 11:41:26 evcc 0.115.0
[main  ] INFO 2023/04/09 11:41:26 using config file: /Users/sterlp/evcc.yaml
[main  ] INFO 2023/04/09 11:41:26 starting ui and api at :7070
[db    ] INFO 2023/04/09 11:41:26 using sqlite database: /Users/sterlp/.evcc/evcc.db
[db    ] TRACE 2023/04/09 11:41:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/04/09 11:41:26 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/04/09 11:41:26 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/04/09 11:41:26 SELECT * FROM `settings` 6 <nil>
[main  ] FATAL 2023/04/09 11:41:26 missing loadpoints
[main  ] FATAL 2023/04/09 11:41:26 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.115.0

StefanSchoof, 2023-04-10:
Your Config is missing the `loadpoints` https://docs.evcc.io/docs/reference/configuration/loadpoints

StefanSchoof, 2023-04-10:
@andig Wäre es einfach möglich die Fehlermeldung von "missing loadpoints" auf was in der Art "Error in your configuration: missing loadpoints" zu ändern?

andig, 2023-04-10:
Gute Idee, schaue ich mir an 👍🏻

closed 2023-04-10 by andig

bbday, 2023-04-10:
I have same issue and i have same config.
I dont have any wallbox or vehicle why do i need to setup it to start app, should be not possible ignore it???

premultiply, 2023-04-10:
A charge controller without a loadpoint would be useless.

andig, 2023-04-10:
Evcc doesn‘t work without loadpoints.