evcc-io/evcc #7393: iOS 12 Support wiederherstellen

naltatis, 2023-04-13:
> Ich vermute, das hängt am vite update auf 4 (in https://github.com/evcc-io/evcc/pull/7225 von @naltatis). Das erhöht die Browser Baseline von Safari 13 auf 14 (https://vitejs.dev/guide/migration.html#modern-browser-baseline-change)

_Originally posted by @StefanSchoof in https://github.com/evcc-io/evcc/discussions/7388#discussioncomment-5601534_

closed 2023-04-13 by naltatis

HeikoH78, 2023-04-14:
@naltatis Hallo Michael, hat funktioniert, mit iOS 12 funktioniert die Anzeige nun wieder. Vielen Dank für den schnellen Fix!

naltatis, 2023-04-14:
@HeikoH78 perfekt!

@Sami0178 schade, ich schau mir das noch mal an. Ging es denn mit einer vorherigen Version da schon mal?

Sami0178, 2023-04-14:
Ja, bisher lief jede Version tadellos.
Bin nur gerade zu doof die die Nightly Version zu testen. ->Vielleicht gehts ja doch mit der Änderung.
Mach bestimmt grundsätzlich was falsch:
Puttycmd:
curl -1sLf \
  'https://dl.cloudsmith.io/public/evcc/unstable/setup.deb.sh' \
> sudo -E bash
curl: error while loading shared libraries: /lib/arm-linux-gnueabihf/libldap_r-2.4.so.2: invalid ELF header

naltatis, 2023-04-14:
@Sami0178 Ich hab den fix mal hier auf der Demo Instanz ausgerollt. Schau mal bitte ob die Seite mit deinem Browser läuft
https://demo-next.evcc.io/

PR #7413 (2023-04-13): Restore iOS 12 support
Added vite-legacy plugin to generate a second legacy js bundle which is only requested by browsers that need it.
Modern browsers are not negatively affected. They use the smaller modern js bundle (as before).

```
vite v4.2.1 building for production...
✓ 340 modules transformed.
../dist/assets/polyfills-legacy-3a11d40f.js   34.03 kB │ gzip:  13.59 kB  << for legacy browsers
../dist/assets/index-legacy-4a9c963f.js      662.02 kB │ gzip: 174.60 kB  << for legacy browsers
...
../dist/assets/index-cc4a17cb.js             452.53 kB │ gzip: 147.75 kB  << for modern browsers
✓  [... cut]