evcc-io/evcc #7394: Fahrzeugerkennung Mercedes über Ladeklappe möglich?

Uwe1984, 2023-04-13:
Ich habe 2 Fahrzeuge (Smart EQ und Mercedes EQB). 
Die Fahrzeugerkennung des Mercedes funktioniert aber leider nicht, weil Mercedes wohl den "Status" nicht liefert. 
Wäre ein Erkennung über die Ladeklappe möglich? Ähnlich Ioniq 5. 
Der Status der Ladeklappe wird in der App angezeigt. Sollte also auch in der API vorhanden sein. 

Hier mal die Antwort-JSON auf "evcc vehicle -l trace" 

`docker exec -it evcc /bin/sh /app # evcc vehicle -l trace [main  ] INFO 2023/04/13 12:57:43 evcc 0.114.1 [main  ] INFO 2023/04/13 12:57:43 using config file: /etc/evcc.yaml [smart ] TRACE 2023/04/13 12:57:43 GET https://id.mercedes-benz.com/as/authorization.oauth2?access_type=offline&client_id=70d89501-938c-4bec-82d0-6abb550b0825&code_challenge=aInPowRgSs9ximSgsLBfxignYd_VN_tNcHIb4b0uPKk&code_challenge_method=S256&redirect_uri=https%3A%2F%2Foneapp.microservice.smart.mercedes-benz.com&response_type=code&scope=openid+profile+email+phone+ciam-uid+offline_access&state=vCfP-OTuBiYi
[smart ] TRACE 2023/04/13 12:57:43 GET https://id.mercedes-benz.com/ciam/auth/login?resume=%2Fas%2FE7t6qChS8S%2Fresume%2Fas%2Fauthorization.ping&request_info=eyJlbmMiOiJBMjU2Q0JDLUhTNTEyIiwia2lkIjoiRklSU1RfRU5DUllQVElPTl9JRCIsImFsZyI6IlBCRVMyLUhTNTEyK0EyNTZLVyIsImN0eSI6IkpXVCIsInAyYyI6ODE5MiwicDJzIjoiM21qNjBrdHNGR3lFTFNqQiJ9.LMtdxh8XDI0kSybtPjFprE8wxJCL5laTzoZZ6036-wBkrUp7iKQK5Zt3w7ghfZ-jbLcaFGVLI-zriO27b2m29bpcSb3fWbui.VLVZHKTKv5_8Y0DL8sHBjg.KF1703l6SviRT2wm_ijELyYL9VvY2bdHt94arXsa_QrVst54SIBpP-oCCpcM-AghFD8yglDkrnT6KdURwUKtEGELtH_ST5hcTMeLbos4exj3_P2hI4mkPNX5A0Iz5ZOAH638gBzLx0bAGfnh0VgQXQZEJFq5F6ZNf5RZhwEsA6pmqKyue3lz8E-myFkwvYKhphzQFt5lwzJ_hCQ6559Np0s2kaVbRH-nXn0c2Bmv904kY6Fv_BdVwqwyF_N8elHBr3bhAiMBQmXjQ07g3pZ-htl5kD-w5ddmo8Ur86c1LL_xTJzLrA7AZkMF3MSAoXcbq4Ias1gPbiUZ62zpRD_5wQhP8FECiyfL7eq7mn5XXJFDZnk_ePPBUwRYSul23nTdt5NA5J7sKo4or8n1GJo_k4G8iijhUBofviMKP8-DaxlqdXs87fWBuj6-wvxec3-mVHCfSem7hibd7z36APMtAwmxn3fNTPZ3314w5xfNcm9V2j5uC3jFhS6M33BlJdMIE72utdWqvN2POmKOVa7eF8vUvm7SQETtLBSdUXl0I_TizP_DChbNyK4Qm3DoZQ6vgPV1yX0XV-26W1O5hHYVHPWRNqbvlY1syEbbtxGwAF-dMD3dTxkjpGeEdQErRgF_AJV29ZKJXlC75LkGeuoAQbzfq1rSEJRN5dn4X5fM4VjammzqKb6q1URNsC1Ftp2yqJeSaqkmPhZaca99qLyplnwnaoec1--iXvRc2JEqKuKxfhZu-W1YGnS7GGRQMLTdUot8GTKZ0TrqqQKHZ8x4VgxxpOx-3eKnu_dnkEvh7HauEPA22rzV4TxKj7pZ_XoW8Sl2unDb3Qv4Zxr-HuF5YGeZ7G57RS9jgTLPVn7IUzBz2QToI72593SvwyDfCd7cIvmFYi5vz0cNv4xawP4Az6-AjPE1_2h5DYBGKJpirYk8MJEIq2d6uxIf5O5drs7RQ0PaERgU54xD1MfpLyIg-7yesVOiX1mtkASST1N2LtQ.fgxSJOZF6ZOzLpkm0_RBFJ25zz2pj8gsbRYdA3UfsO0
[smart ] TRACE 2023/04/13 12:57:43 <!DOCTYPE html><html lang=en xmlns=http://www.w3.org/1999/xhtml><head><meta charset=utf-8><meta http-equiv=X-UA-Compatible content="IE=edge"><meta name=viewport content="width=device-width,initial-scale=1"><link rel=icon href=/ciam/auth/static/favicon.ico><title>Mercedes me</title><link rel=stylesheet href=https://cdn.jsdelivr.net/npm/semantic-ui@2.4.2/dist/semantic.min.css integrity="sha256-UXesixbeLkB/UYxVTzuj/gg3+LMzgwAmg3zD+C4ZASQ=" crossorigin=anonymous><script src=https://cdn.jsdelivr.net/npm/jquery@3.5.1/dist/jquery.min.js integrity="sha256-9/aliU8dGd2tb6OSsuzixeV4y/faTqgFtohetphbbj0=" crossorigin=anonymous></script><script src=https://cdn.jsdelivr.net/npm/semantic-ui@2.4.2/dist/semantic.min.js integrity="sha256-CgSoWC9w5wNmI1aN8dIMK+6DPelUEtvDr+Bc2m/0Nx8=" crossorigin=anonymous></script><script>/*<![CDATA[*/
      const initialState = "{\"resume\":\"\/as\/E7t6qChS8S\/resume\/as\/authorization.ping\",\"country\":\"GB\",\"recaptchaSiteKey\":\"6LcmthsbAAAAAJABgXzZXLHJ3qVv2-Khab0aQM6F\",\"urlPF\":\"https:\/\/id.mercedes-benz.com\",\"isBotCheckActive\":true,\"appName\":\"seqcprod\",\"contextPath\":\"\/ciam\/auth\",\"isRememberMeEnabledByDefault\":false,\"touchpoint\":null,\"isRecaptchaActive\":true,\"isPasswordStrengthMeterEnabled\":false,\"acr\":[\"OTP\",\"PASSWORD\",\"REMEMBER_ME\"],\"deployedStage\":\"PROD\",\"requestInfo\":\"eyJlbmMiOiJBMjU2Q0JDLUhTNTEyIiwia2lkIjoiRklSU1RfRU5DUllQVElPTl9JRCIsImFsZyI6IlBCRVMyLUhTNTEyK0EyNTZLVyIsImN0eSI6IkpXVCIsInAyYyI6ODE5MiwicDJzIjoiM21qNjBrdHNGR3lFTFNqQiJ9.LMtdxh8XDI0kSybtPjFprE8wxJCL5laTzoZZ6036-wBkrUp7iKQK5Zt3w7ghfZ-jbLcaFGVLI-zriO27b2m29bpcSb3fWbui.VLVZHKTKv5_8Y0DL8sHBjg.KF1703l6SviRT2wm_ijELyYL9VvY2bdHt94arXsa_QrVst54SIBpP-oCCpcM-AghFD8yglDkrnT6KdURwUKtEGELtH_ST5hcTMeLbos4exj3_P2hI4mkPNX5A0Iz5ZOAH638gBzLx0bAGfnh0VgQXQZEJFq5F6ZNf5RZhwEsA6pmqKyue3lz8E-myFkwvYKhphzQFt5lwzJ_hCQ6559Np0s2kaVbRH-nXn0c2Bmv904kY6Fv_BdVwqwyF_N8elHBr3bhAiMBQmXjQ07g3pZ-htl5kD-w5ddmo8Ur86c1LL_xTJzLrA7AZkMF3MSAoXcbq4Ias1gPbiUZ62zpRD_5wQhP8FECiyfL7eq7mn5XXJFDZnk_ePPBUwRYSul23nTdt5NA5J7sKo4or8n1GJo_k4G8iijhUBofviMKP8-DaxlqdXs87fWBuj6-wvxec3-mVHCfSem7hibd7z36APMtAwmxn3fNTPZ3314w5xfNcm9V2j5uC3jFhS6M33BlJdMIE72utdWqvN2POmKOVa7eF8vUvm7SQETtLBSdUXl0I_TizP_DChbNyK4Qm3DoZQ6vgPV1yX0XV-26W1O5hHYVHPWRNqbvlY1syEbbtxGwAF-dMD3dTxkjpGeEdQErRgF_AJV29ZKJXlC75LkGeuoAQbzfq1rSEJRN5dn4X5fM4VjammzqKb6q1URNsC1Ftp2yqJeSaqkmPhZaca99qLyplnwnaoec1--iXvRc2JEqKuKxfhZu-W1YGnS7GGRQMLTdUot8GTKZ0TrqqQKHZ8x4VgxxpOx-3eKnu_dnkEvh7HauEPA22rzV4TxKj7pZ_XoW8Sl2unDb3Qv4Zxr-HuF5YGeZ7G57RS9jgTLPVn7IUzBz2QToI72593SvwyDfCd7cIvmFYi5vz0cNv4xawP4Az6-AjPE1_2h5DYBGKJpirYk8MJEIq2d6uxIf5O5drs7RQ0PaERgU54xD1MfpLyIg-7yesVOiX1mtkASST1N2LtQ.fgxSJOZF6ZOzLpkm0_RBFJ25zz2pj8gsbRYdA3UfsO0\",\"lang\":\"en\",\"brand\":\"smart\",\"hideRegistration\":false}"// SPA initialization data
      const landingPage = null// page to load on init
      const landingPageParams = null
      const landingErrors = null// errors to display on init
      const autoResume = null// return to PF on init if true
      const contextPath = "\/ciam\/auth"
      const brand = "smart"
      const pageTitle = null
      const showError = null
      const errorLang = null
      const errorSummary = null
      const errorStatus = null
      const errorRequestId = null
      const errorShowBack = null
      /*]]>*/</script><link href=/ciam/auth/static/js/app.48b09f5c.js rel=preload as=script><link href=/ciam/auth/static/js/chunk-vendors.dd6674c8.js rel=preload as=script></head><body><noscript><strong>We're sorry but this page doesn't work properly without JavaScript enabled. Please enable it to continue.</strong></noscript><div id=app></div><script src=/ciam/auth/static/js/chunk-vendors.dd6674c8.js></script><script src=/ciam/auth/static/js/app.48b09f5c.js></script></body></html>
[smart ] TRACE 2023/04/13 12:57:43 POST https://id.mercedes-benz.com/ciam/auth/login/user
[smart ] TRACE 2023/04/13 12:57:44 {"username":"***"}
--
{"country":"DE","hasPass":true,"lang":"de"}
[smart ] TRACE 2023/04/13 12:57:44 POST https://id.mercedes-benz.com/ciam/auth/login/pass
[smart ] TRACE 2023/04/13 12:57:44 {"username":"***","password":"***","rememberMe":true}
--
{"result":"RESUME2OIDCP","token":"eyJlbmMiOiJBMjU2Q0JDLUhTNTEyIiwia2lkIjoiMTMyOTg4OSIsImFsZyI6IlJTQS1PQUVQLTI1NiIsImN0eSI6IkpXVCJ9.jRJgSpDobkng3Zw4ffL6SjIBfkwu-K8QudBbMbBkHdQ1JGW3ME5BeR9CmhXhBKG7MbKDHU0NKTaH_RX_whdyAS_A_Q7QNO7-JeBtk9POZD7c3A2W6un8UP9UiKm9A1nzQgQvhLEIettMfvgBvm6vBKAf92z7tCBC1gtV8qBZqY6G1LrOZJ80cXS0-qI0Yy8KVR2D9pDDhzW-RMfRyKaSNWW8XBtSDie9fkWMvaucLCXZ8uqaLCVBASWL7mcb2rFMaUAf9XkPJCCz-SxLOLMgBzl4xGBm2Cy28UJWJaqRrI9fg2VMaJqDzEwga3Ns-kHkhO9nrSMhgZ8_hmNoHL-ulg.hpyK5UXS0MO2D5v8XRbVtQ.9simvmuqI0aBjP6nthzO8e8d6VEBlvbaNW8UMSXqdGu4ZF8uShgzCx1WM-Q6MNVvqgt0akngXnlDllrtORyxYb5ktraqxl-hyTc6j_s90llNkECcVcze8sxffF-dsqRkULpzcfuoHJcfKzOim5MfS5oQ5vVjUeEwefSq46AYkcsHuZ5ko6_yhXuCfc_2n1mJY3WklifadTWI7929zTpya08x2ZRMTTDF_T8cvSVoZQAGjKkvyALXddqz5cXmBFF8RGnto9t7LndJJZ_BDpI-5K2xouhm51aTl-ZLHE60AgK7Z8UsfSH1obq-3zxF0CJP5gjjh053uBlZL7H5EoWzuIrFsc-_2wHqmG0Wjh2pFRIL7GjPaMeemVVDSOlnt7XJxRKcw00_-vlkt1iMNiPH27jZwLZhyO5fg-C5SaUQeKGXBU1R3qNZuQsI2jWGe0UffCXWGbQKlfbTKC7v4Mx1Y7CH755lNb1bJ4ICUrcuAk4osPzwnzjl_b9CDjkv4-qt4KrvYhBNOYkXEONzCmL_lPknOHChxuRjoZ68UdUQks5T1KVi3HXc1XD_UR6JiGXYQKZ7dFJOPvCChnl40bo_z9ZWSqcviXLxNSYOF8j6vYyBBmGFuo775X2eYAmzqvuj5kCvekP0ATJ5mFwvXGjacJ-5B1Hmg2OXAqWvCmArR5CUCJYGuq7HaEbI2Z8eqKJBcPNQhLttj7V3tBwPEPxOIdYj9mPekZ43Cxp1rMnPvQSCYUpkue2e6nuRTi9jqUmoIvWJfX_GPpg0N9fJSaKV0GperTuWq9EZJiGkldx5S9CoUc8RKAHOwubGQKoDd6ocKqA99aepESmhyC55qFyzmA.yg3Y26GxXnKkfvw8EGeS-AThRyYHigAekaIhlp4kHWE"}
[smart ] TRACE 2023/04/13 12:57:44 POST https://id.mercedes-benz.com/as/E7t6qChS8S/resume/as/authorization.ping
[smart ] TRACE 2023/04/13 12:57:44 token=eyJlbmMiOiJBMjU2Q0JDLUhTNTEyIiwia2lkIjoiMTMyOTg4OSIsImFsZyI6IlJTQS1PQUVQLTI1NiIsImN0eSI6IkpXVCJ9.jRJgSpDobkng3Zw4ffL6SjIBfkwu-K8QudBbMbBkHdQ1JGW3ME5BeR9CmhXhBKG7MbKDHU0NKTaH_RX_whdyAS_A_Q7QNO7-JeBtk9POZD7c3A2W6un8UP9UiKm9A1nzQgQvhLEIettMfvgBvm6vBKAf92z7tCBC1gtV8qBZqY6G1LrOZJ80cXS0-qI0Yy8KVR2D9pDDhzW-RMf [... cut ...] scription":"Warndreieck"},{"code":"H12","description":"Verbandstasche"},{"code":"H15","description":"Ladekabeltasche"},{"code":"I01","description":"Klimatisierungsautomatik"},{"code":"I13","description":"Sidebags"},{"code":"I29","description":"Keine Klimavorruestung für heisse Laender"},{"code":"I34","description":"12-Volt-Steckdose"},{"code":"I40","description":"Fußmatten Ve
Soc:           100%
Capacity:      17.6kWh
Charge status: B
Range:         136km
Odometer:      1507km
Features:      []`

premultiply, 2023-04-13:
Kein JSON zu sehen.

Uwe1984, 2023-04-13:
Ok, wie meinst du das? Ich kenne mich leider nicht so gut aus. Muss ich das Auto anstecken und das Fahrzeug in evcc anwählen und dann nochmal den Befehl ausführen " evcc vehicle -l trace " ?

VolkerK62, 2023-04-13:
es sind nur "Smart" Daten vorhanden.
>.Soc:     Get "https://api.mercedes-benz.com/vehicledata/v2/vehicles/W1N9M0CB2PN035072/containers/electricvehicle": oauth2: token expired and refresh token is not set

mit deinem Mercedes Login stimmt etwas nicht.

Uwe1984, 2023-04-13:
@andig ,

wäre es denn denkbar, dass ich die Fahrzeugeekennung über z.B. Iobroker und ein Skript umsetze? 

In Iobroker habe ich alle Attribute von meinem EQB und es gibt auch einen evcc Adapter. Allerdings kann ich kein Attribut im evcc Iobroker Adapter finden, womit ich das Fahrzeug wählen kann. 

Gibt es denn eine Schnittstelle in evcc um das von extern zu steuern?  Eventuell muss das nur im Iobroker Adapter nachgepflegt werden.

andig, 2023-04-13:
Ja, das sind die schon vielfach diskutierten Plugins für ein `custom` vehicle.