evcc-io/evcc #7395: Wrong timer function resulting in regular charging stop

tafx, 2023-04-13:
### Describe the bug

Charging process in interrupted with period of timer "enable: delay: 3m" in evcc.yaml. Ot starts again and the cycle repeats until the car is disconnected or refuses to start over again.
Tha charging mode does not affect the behavior. Tested in "Solar" and in "Fast". Also the current PV power generation does not influence the charging interruption. 
The timer  counting down during charging in "Solar" mode is on enclosed screenshot as well as time plot of charging power with visible interuptions.
The observed behavior is after the update to v0.116.0. Older versions not affected.
[image]
[image]

### Steps to reproduce

1. Connect the car.
2. Set the charging mode.
3. Car is then charging with interuptions.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
# sponsortoken:

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - type: custom
    name: grid
    power:
      source: mqtt
      topic: evcc/tometers/grid
      timeout: 30s
      scale: -1

  - type: custom
    name: pv
    power:
      source: mqtt
      topic: evcc/tometers/pv
      timeout: 30s
      scale: 1

  - type: custom
    name: battery
    power:
      source: mqtt
      topic: evcc/tometers/battery
      timeout: 30s
      scale: 1
    soc:
      source: mqtt
      topic: evcc/tometers/batterysoc
      timeout: 30s
      scale: 1

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
  - name: Pravy_WB
    type: template
    template: alfen
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.2.16 # Hostname
    port: 502 # Port

  - name: Levy_WB
    type: template
    template: alfen
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.2.17 # Hostname
    port: 502 # Port

vehicles:
  - name: E-C4
    type: template
    template: citroen
    title: E-C4 # Wird in der Benutzeroberfläche angezeigt # Optional
    user: xxxx@gmail.com # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: xxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 50 # Akku-Kapazität in kWh # Optional

# site describes the EVU connection, PV and home battery
site:
  title: Domov # display name for UI
  meters:
    grid: grid # grid meter
    pvs:
      - pv # list of pv inverters/ meters
    battery: battery # battery meter
  prioritySoC: 65 # give home battery priority up to this soc (empty to disable)
  bufferSoC: 95 # ignore home battery discharge above soc (empty to disable)
  residualPower: 100

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  
  - title: Levé stání # display name for UI
    charger: Levy_WB # charger
    mode: "pv" # set default charge mode, use "off" to disable by default if charger is publicly available
    vehicle: # set default vehicle (disables vehicle detection)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: connected
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 90m
      min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
      target: 90 # always charge to 100%
      estimate: true # set false to disable interpolating between api updates (not recommended)
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    enable: # pv mode enable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: -0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 5m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 10m # switch charger contactor not more often than this (default 5m)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 14 # maximum charge current (default 16A)

  - title: Pravé stání # display name for UI
    charger: Pravy_WB # charger
    mode: "pv" # set default charge mode, use "off" to disable by default if charger is publicly available
    vehicle: E-C4 # set default vehicle (disables vehicle detection)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: connected
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 90m
      min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
      target: 90 # always charge to 100%
      estimate: true # set false to disable interpolating between api updates (not recommended)
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    enable: # pv mode enable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 5m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 10m # switch charger contactor not more often than this (default 5m)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 14 # maximum charge current (default 16A)
```

### Log details

```text
How to acces the log when evcc is runing under HAOS as a addon?
```

### What type of operating system are you running?

Linux

closed 2023-04-13 by andig

Hofyyy, 2023-04-13:
@tafx you can lower your kw capacity for your car a little bit (5%) because it is used for calculqtion and see if it helps as a workaround.

tafx, 2023-04-13:
@Hofyyy @andig 
So I observed the log and found out that evcc is restarting after 3 mins that is causing the interuption.
Here is the part of the log but is not perfect as I am pasting it from Home Assistant addon window.
[EVCC log.txt](https://github.com/evcc-io/evcc/files/11224164/EVCC.log.txt)

andig, 2023-04-13:
Your short log has 3 restarts of evcc for unknown reason