evcc-io/evcc #7399: Fehlermeldung vehicle status: PRIVACY_ACTIVATED wenn Fahrzeug im Privatmodus (Potsche)

MarkusGH, 2023-04-13:
### Describe the bug

Wenn ein in der Konfiguration hinterlegtes Fahrzeug im Privatmodus ist bekommt man nach Neustart permanent die Fehlermeldung `vehicle status: PRIVACY_ACTIVATED `

### Steps to reproduce

1) Porsche in Privatmodus versetzen
2) evcc neustarten

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: localhost
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 3s # control cycle interval

log: info
levels: 
  porsche: trace 
  lp-1: debug
  lp-2: debug
  openwb: trace

mqtt:
  broker: XXX:1883
  topic: evcc

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.4063 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.0816 # [currency]/kWh

meters:
- name: grid1
  type: movingaverage
  decay: 1 # 0.5
  meter:
    type: custom
    power:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Power 
      scale: -1  # Die Leistung wird in Watt erwartet. Falls z.b. kW geliefert wird, kann dies hier mit dem Wert 1000 angepasst werden
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    energy: 
      source: mqtt
      topic: mbmd/solaredge1-1-1/Import 
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    currents:
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L1 
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L2 
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L3 
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
- name: pv1
  type: custom
  power:
    source: mqtt
    topic: mbmd/solaredge1-1/Power 
    timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
  energy: 
    source: mqtt
    topic: mbmd/solaredge1-1/Export 
    timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
  currents:
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L1 
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L2 
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L3 
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist

chargers:
- name: openwb1
  type: openwb
  phases1p3p: true
  broker: XXX # openWB IP
  id: 1                      # loadpoint ID
- name: openwb2
  type: openwb
  phases1p3p: true
  broker: XXX # openWB IP
  id: 2                      # loadpoint ID

vehicles:
- name: ev1
  type: template
  template: porsche
  title: XXX  # Wird in der Benutzeroberfläche angezeigt
  user: XXX  # Benutzername um auf das Gerät zuzugreifen
  password: XXX  # Passwort des Benutzerkontos
  vin: XXX# Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind
  capacity: 17.9  # Akku-Kapazität in kWh
  phases: 2
  minCurrent: 6
  maxCurrent: 16
  cache: 20m

loadpoints:
- title: Carport
  charger: openwb2
  resetOnDisconnect: true
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  enable:
    threshold: -600
    delay: 1m
  disable:
    threshold: 800
    delay: 3m
  guardduration: 5m
- title: Straße
  charger: openwb1
  resetOnDisconnect: true
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  enable:
    threshold: -600
    delay: 1m
  disable:
    threshold: 800
    delay: 3m

site:
  title: H73
  meters:
    grid: grid1
    pvs:
    - pv1
  residualPower: -150 
#  maxCurrent: 63

influx:
  url: http://localhost:8086
  database: evcc
```

### Log details

```text
Apr 13 14:04:11 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:04:11 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:04:17 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:04:17 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:04:24 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:04:24 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:04:29 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:04:29 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:04:35 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:04:35 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:04:41 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:04:41 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:04:47 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:04:47 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:04:53 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:04:53 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:04:59 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:04:59 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:05:05 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:05:05 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:05:11 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:05:11 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:05:17 homebridge evcc[4171245]: [main  ] ERROR 2023/04/13 14:05:17 vehicle status: PRIVACY_ACTIVATED
```

### What type of operating system are you running?

Linux

### Version

 evcc version 0.116.0

andig, 2023-04-13:
Ja.... und?

andig, 2023-04-13:
Ja, und? Es kommt nunmal kein Status- was sollen wir da tun?

MarkusGH, 2023-04-13:
Keinen Fehler ausgeben?

andig, 2023-04-13:
Es ist halt einer wenn wir einen Status brauchen- Fahrzeugerkennung ist damit nicht mehr möglich. Lösung: Fahrzeug entfernen oder Privacy Modus aus. Verkürzt gesagt: Privacy mode ist nicht unterstützt.

closed 2023-04-13 by andig

MarkusGH, 2023-04-13:
Das Fahrzeug ist doch nicht einmal angestöpselt!
Das steht in der Werkstatt und an der Wallbox hängt der Ersatzwagen als "Gastfahrzeug".

andig, 2023-04-13:
Ist doch ganz einfach: dann musst du es aus Deiner Konfiguration entfernen. Du hast ein ladendes Fahrzeug und deshalb läuft die Fahrzeugerkennung und -Abfrage. Wenn Du "Gastfahrzeug" auswählst sollte das Problem gelöst sein.

MarkusGH, 2023-04-13:
So einfach finde ich das nicht.

1) Auswahl von "Gastfahrzueg bringt die Meldungen nicht zum Verschwinden. 

2) Das bedeutet, der Endanwender soll jedes Mal wenn sich jemand in das Auto setzt und bei der Abfrage im PCM den Privatmodus angeschaltet lässt die Konfiguration editieren? Ernsthaft?

Zusatzinfo: Nach 30 Minuten hörte der Spuk auf, aber in der Zeit wurde die Fahrzeug API jede Minute gepollt. Ich denke nicht, dass das das richtige Verhalten ist:

Apr 13 14:16:25 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:25 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:31 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:31 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:37 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:37 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:43 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:43 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:49 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:49 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:55 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:55 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:01 homebridge evcc[609570]: [porsche] TRACE 2023/04/13 14:17:01 GET https://api.ppa.porsche.com/app/connect/v1/vehicles/***?mf=BATTERY_LEVEL&mf=BATTERY_CHARGING_STATE&mf=CLIMATIZER_STATE&mf=E_RANGE&mf=HEATING_STATE&mf=MILEAGE
Apr 13 14:17:02 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:02 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:07 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:07 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:13 homebridge evcc[609570]: [lp-1  ] INFO 2023/04/13 14:17:13 start charging ->
Apr 13 14:17:13 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:13 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:19 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:19 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:25 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:25 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:31 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:31 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:37 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:37 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:43 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:43 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:49 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:49 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:55 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:55 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:01 homebridge evcc[609570]: [porsche] TRACE 2023/04/13 14:18:01 GET https://api.ppa.porsche.com/app/connect/v1/vehicles/***?mf=BATTERY_LEVEL&mf=BATTERY_CHARGING_STATE&mf=CLIMATIZER_STATE&mf=E_RANGE&mf=HEATING_STATE&mf=MILEAGE
Apr 13 14:18:02 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:02 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:07 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:07 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:13 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:13 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:19 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:19 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:25 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:25 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:31 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:31 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:37 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:37 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:43 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:43 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:49 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:49 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:55 homebridge evcc[609570]: [porsche] TRACE 2023/04/13 14:18:55 GET https://api.ppa.porsche.com/app/connect/v1/vehicles/***?mf=BATTERY_LEVEL&mf=BATTERY_CHARGING_STATE&mf=CLIMATIZER_STATE&mf=E_RANGE&mf=HEATING_STATE&mf=MILEAGE
Apr 13 14:18:56 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:56 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:01 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:01 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:07 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:07 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:13 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:13 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:19 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:19 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:25 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:25 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:31 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:31 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:37 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:37 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:43 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:43 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:49 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:49 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:55 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:55 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:20:01 homebridge evcc[609570]: [porsche] TRACE 2023/04/13 14:20:01 GET https://api.ppa.porsche.com/app/connect/v1/vehicles/***?mf=BATTERY_LEVEL&mf=BATTERY_CHARGING_STATE&mf=CLIMATIZER_STATE&mf=E_RANGE&mf=HEATING_STATE&mf=MILEAGE
Apr 13 14:20:02 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:20:02 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:20:07 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:20:07 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:20:13 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:20:13 vehicle status: PRIVACY_ACTIVATED

andig, 2023-04-13:
> Das bedeutet, der Endanwender soll jedes Mal wenn sich jemand in das Auto setzt und bei der Abfrage im PCM den Privatmodus angeschaltet lässt die Konfiguration editieren? Ernsthaft?

Wir können jeden sinnvollen Vorschlag diskutieren. Bis heute kannte ich den Modus nichtmal. Dass eine erwartete (=charging) Statusabfrage da einen Fehler liefert ist völlig ok. Hier gibts zwei Fragen:

1. warum wird bei Fehler minütlich gepollt (@tobiasbayer hat das was mit https://github.com/evcc-io/evcc/pull/6252 zu tun? Hier steht `cache: 20m`, das *darf* also nicht passieren.
2. warum hört der Spuk nach Auswahl des Gastfahrzeugs (=wenn kein Soc mehr vom Porsche benötigt wird) nicht auf?

Sind das die Fragen?

MarkusGH, 2023-04-13:
> Wir können jeden sinnvollen Vorschlag diskutieren. Bis heute kannte ich den Modus nichtmal. Dass eine erwartete (=charging) Statusabfrage da einen Fehler liefert ist völlig ok.

Jetzt sind wir wieder beieinander. Mir geht es ja auch nicht ums Meckern sondern darum mitzuhelfen, die Software zu verbessern.

Die API antwortet:

`"measurements":[{"key":"HEATING_STATE","status":{"isEnabled":false,"cause":"NOT_SUPPORTED"}},{"key":"MILEAGE","status":{"isEnabled":false,"cause":"PRIVACY_ACTIVATED"}},{"key":"CLIMATIZER_STATE","status":{"isEnabled":false,"cause":"PRIVACY_ACTIVATED"}},{"key":"BATTERY_LEVEL","status":{"isEnabled":false,"cause":"PRIVACY_ACTIVATED"}},{"key":"BATTERY_CHARGING_STATE","status":{"isEnabled":false,"cause":"PRIVACY_ACTIVATED"}},{"key":"E_RANGE","status":{"isEnabled":false,"cause":"PRIVACY_ACTIVATED"}}]`

Macht aber auf jeden Fall keinen Sinn, da einen "Error" auszugeben, das ist ein erwartbarer Betriebszustand.
Maximal ist das eine "Info".

> Hier gibts m.E. zwei andere Fragen:
> 
> 1. warum wird bei Fehler minütlich _das Fahrzeug_ gepollt (@tobiasbayer hat das was mit [Add exponential backoff retries for failed cache refresh #6252](https://github.com/evcc-io/evcc/pull/6252) zu tun? Hier steht `cache: 20m`, das _darf_ also nicht passieren.
> 2. warum hört der Spuk nach Auswahl des Gastfahrzeugs (=wenn kein Soc mehr vom Porsche benötigt wird) nicht auf?
> 
> Sind das die Fragen?

Das sind die neuen - noch interessanteren - Fragen.

MarkusGH, 2023-04-13:
Das sind alle Meldungen ab Neustart:

Apr 13 14:15:52 homebridge evcc[609570]: [porsche] TRACE 2023/04/13 14:15:52 GET https://api.ppa.porsche.com/app/connect/v1/vehicles
Apr 13 14:15:55 homebridge evcc[609570]: [lp-1  ] INFO 2023/04/13 14:15:55 start charging ->
Apr 13 14:15:55 homebridge evcc[609570]: [porsche] TRACE 2023/04/13 14:15:55 GET https://api.ppa.porsche.com/app/connect/v1/vehicles/***?mf=BATTERY_LEVEL&mf=BATTERY_CHARGING_STATE&mf=CLIMATIZER_STATE&mf=E_RANGE&mf=HEATING_STATE&mf=MILEAGE
Apr 13 14:15:56 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:15:56 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:15:56 homebridge evcc[609570]: [lp-1  ] WARN 2023/04/13 14:15:56 charger logic error: disabled but charging
Apr 13 14:16:01 homebridge evcc[609570]: [lp-1  ] INFO 2023/04/13 14:16:01 stop charging <-
Apr 13 14:16:01 homebridge evcc[609570]: [porsche] TRACE 2023/04/13 14:16:01 GET https://api.ppa.porsche.com/app/connect/v1/vehicles/***?mf=BATTERY_LEVEL&mf=BATTERY_CHARGING_STATE&mf=CLIMATIZER_STATE&mf=E_RANGE&mf=HEATING_STATE&mf=MILEAGE
Apr 13 14:16:02 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:02 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:07 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:07 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:13 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:13 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:19 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:19 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:25 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:25 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:31 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:31 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:37 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:37 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:43 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:43 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:49 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:49 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:16:55 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:16:55 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:01 homebridge evcc[609570]: [porsche] TRACE 2023/04/13 14:17:01 GET https://api.ppa.porsche.com/app/connect/v1/vehicles/***?mf=BATTERY_LEVEL&mf=BATTERY_CHARGING_STATE&mf=CLIMATIZER_STATE&mf=E_RANGE&mf=HEATING_STATE&mf=MILEAGE
Apr 13 14:17:02 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:02 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:07 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:07 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:13 homebridge evcc[609570]: [lp-1  ] INFO 2023/04/13 14:17:13 start charging ->
Apr 13 14:17:13 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:13 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:19 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:19 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:25 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:25 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:31 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:31 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:37 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:37 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:43 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:43 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:49 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:49 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:17:55 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:17:55 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:01 homebridge evcc[609570]: [porsche] TRACE 2023/04/13 14:18:01 GET https://api.ppa.porsche.com/app/connect/v1/vehicles/***?mf=BATTERY_LEVEL&mf=BATTERY_CHARGING_STATE&mf=CLIMATIZER_STATE&mf=E_RANGE&mf=HEATING_STATE&mf=MILEAGE
Apr 13 14:18:02 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:02 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:07 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:07 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:13 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:13 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:19 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:19 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:25 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:25 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:31 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:31 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:37 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:37 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:43 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:43 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:49 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:49 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:18:55 homebridge evcc[609570]: [porsche] TRACE 2023/04/13 14:18:55 GET https://api.ppa.porsche.com/app/connect/v1/vehicles/***?mf=BATTERY_LEVEL&mf=BATTERY_CHARGING_STATE&mf=CLIMATIZER_STATE&mf=E_RANGE&mf=HEATING_STATE&mf=MILEAGE
Apr 13 14:18:56 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:18:56 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:01 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:01 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:19:07 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:19:07 vehicle status: PRIVACY_ACTIVATED
[... cut ...]
Apr 13 14:25:31 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:25:31 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:25:37 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:25:37 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:25:43 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:25:43 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:25:49 homebridge evcc[609570]: [main  ] ERROR 2023/04/13 14:25:49 vehicle status: PRIVACY_ACTIVATED
Apr 13 14:29:13 homebridge evcc[609570]: [lp-1  ] INFO 2023/04/13 14:29:13 stop charging <-

andig, 2023-04-13:
2. ist hoffentlich gelöst: https://github.com/evcc-io/evcc/commit/6098cc54a41b5aa7392dfc511387991e6c88a0c1

tobiasbayer, 2023-04-13:
Durch scharf Hingucken ist mir jetzt nichts aufgefallen, was zu dem Retry-Verhalten führen könnte. Ich werde die Tage mal versuchen, das im laufenden Betrieb nachzustellen.

andig, 2023-04-13:
Done: https://github.com/evcc-io/evcc/commit/aedd958d001d31ec1a1ba1cfe899270017376d25

tobiasbayer, 2023-04-13:
> Durch scharf Hingucken ist mir jetzt nichts aufgefallen, was zu dem Retry-Verhalten führen könnte. Ich werde die Tage mal versuchen, das im laufenden Betrieb nachzustellen.

Ich kanns mit meinem Ioniq 5 nicht reproduzieren. Die Calls an die Fahrzeug-API werden im (von mir absichtlich provozierten Fehlerfall) wie erwartet immer seltener.
Evtl. irgendeine Spezialität in der Porsche Providerimplementierung?