evcc-io/evcc #7427: Sunny Portal smaapis.de integration

yipikaye, 2023-04-13:
Sunny Home manager direct API calls from EVCC are limited to Sunny home manager 2 and do not support sunny home manager 1.x and off grid systems.

Integration with smaapis.de will simply setup and support more systems.

### Discussed in https://github.com/evcc-io/evcc/discussions/7403

<div type='discussions-op-text'>

<sup>Originally posted by **yipikaye** April 13, 2023</sup>
Hi All, 

Great application! I have a proposal to simplify SMA integration and support more systems.

It appears SMA have been quitely working on and improving their SMA monitoring and APIs. Sunny portal provides centralised management of SMA systems and now has central API (https://sandbox.smaapis.de/live/index.html)

There is now a centralised near real time API for SMA systems which has battery, PV generation, consumption, grid feed in, diesel generation etc in one call. SMA recommendations is not to call more frequently than every 10 sec but this seems ok. SMA say this API should not be used for real time monitoring but now I have found many paid services which use this API.

My interest in this is I have Sunny Home Manager 1, which does not work with EVCC however Sunny Home Manager 1 and 2 works with Sunny Portal and smaapis.de. I'm also off grid so the diesel generation in the API is important, where there is Diesel used then EV charging should stop

For a device: 
`GET /v1/devices/{deviceId}/measurements/sets/{setType}` - Get live measurement set for the given device.

Returns
```
{
  "device": {
    "deviceId": "61ae6556-42fa-45ec-8f67-0c92b056856e",
    "name": "SB5000TL",
    "timezone": "Europe/Berlin"
  },
  "setType": "EnergyMix",
  "set": {
    "time": "2020-03-23T12:40:00",
    "pvConsumptionRate": 0.59727045903977,
    "batteryConsumptionRate": 0.09727045903977,
    "gridConsumptionRate": 0.29727045903977,
    "autarkyRate": 0.89727045903977,
    "batteryStateOfCharge": 65.24914285714286,
    "totalConsumption": 3914.541
  },
  "status": "Ok"
}
```

For a whole plant
`GET /v1/plants/{plantId}/measurements/sets/{setType}`

```
{
  "plant": {
    "plantId": "25057",
    "name": "Test Plant",
    "description": "Test Plant Description",
    "timezone": "Europe/Berlin"
  },
  "status": "Ok",
  "error": "DeviceCommunication",
  "setType": "EnergyBalance",
  "set": {
    "time": "2020-03-23T12:40:00",
    "pvGeneration": 3110.263,
    "directConsumption": 3512.402,
    "totalConsumption": 3914.541,
    "batteryCharging": 0,
    "batteryDischarging": 0,
    "batteryStateOfCharge": 65.24914285714286,
    "dieselGeneration": 402.139,
    "combinedHeatAndPowerGeneration": 396.455,
    "hydroGeneration": 0,
    "gridFeedIn": 396.455,
    "gridConsumption": 402.139,
    "totalGeneration": 3908.857,
    "selfConsumption": 3512.402,
    "autarkyRate": 0.89727045903977,
    "selfConsumptionRate": 0.8985752100933854,
    "selfSupply": 3512.402
  }
}
```

 

</div>

andig, 2023-04-14:
Sounds like you could do this with templates, similar to https://github.com/evcc-io/evcc/issues/7318. Want to try a PR?

/cc @premultiply

yipikaye, 2023-04-14:
Templates sounds good, open to anything. Whats a PR?

gramss, 2023-04-15:
@yipikaye a PR is a "Pull Request" which means that you clone this repository, make your changes and offer them back to the community in form of a PR. Then it can be seen on the code level what your changes were and they are automatically checked to not break things. If all goes well, this is the way of integrating your device in the official release of `evcc` 🙂 

You can read more about them here: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

yipikaye, 2023-04-15:
Oh right, I can get that done. I'll be on holidays in a few months so I'll give it a go

andig, 2023-05-09:
friendly ping @yipikaye, a working config should suffice

yipikaye, 2023-05-12:
Hi @andig, 
It looked a little more complex than I had first thought. So I moved my attention to beers and sunshine :)

andig, 2023-05-12:
I can take a look but I would need access credentials.

yipikaye, 2023-05-12:
I am happy to give you access to my system and API keys. Would you prefer to desicss via email?

andig, 2023-05-12:
Yes, please, info@evcc.io!

andig, 2023-05-12:
Per SMA feedback it seems that b2c usage of their apis is not desired. B2B usage would require us to provide a cloud service which is out of scope for now. Looks like wontfix.

closed 2023-05-12 by andig