evcc-io/evcc #7434: Ladeplanung nicht mehr übersetzt

Balujuhu, 2023-04-14:
### Describe the bug

Wenn ein Fahrzeug angesteckt ist, steht bei der Ladeplannung jetzt "None" statt der dt. Übersetzung. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 192.168.1.223  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 192.168.1.223  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.1.232  
  name: wallbox5

vehicles:
- type: template
  template: hyundai 
  title: Ioniq  
  user: *****
  password: *****
  vin: *****
  capacity: 39  
  phases: 1  
  icon: car  
  cache: 15m  
  language: de  
  mode: pv  
  minSoc: 40  
  targetSoc: 80  
  minCurrent: 6  
  maxCurrent: 10  
  name: ev3
- type: template
  template: hyundai 
  title: Kona  
  user: *****
  password: *****
  vin: *****
  capacity: 39  
  phases: 1  
  icon: car  
  cache: 15m  
  language: de  
  mode: pv  
  minSoc: 45  
  targetSoc: 80  
  minCurrent: 6  
  maxCurrent: 10  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 10
  resetOnDisconnect: true

tariffs:
  grid:
    type: tibber
    token: *****

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
  smartCostLimit: 0.18

Version: `0.116.0`
```

### Log details

```text
[main  ] INFO 2023/04/14 09:44:38 evcc 0.116.0
[main  ] INFO 2023/04/14 09:44:38 using config file: /etc/evcc.yaml
[main  ] INFO 2023/04/14 09:44:38 starting ui and api at :7070
[db    ] INFO 2023/04/14 09:44:38 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] FATAL 2023/04/14 09:44:38 cannot create meter 'grid1': cannot create meter 'template': cannot create meter 'custom': power: dial tcp 192.168.1.223:1502: connect: connection refused
[main  ] FATAL 2023/04/14 09:44:38 will attempt restart in: 5m0s
[server] FATAL 2023/04/14 09:44:38 unlinkat /tmp/evcc-7070: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.116.0

StevieC121176, 2023-04-14:
Ja, gleiches auch bei der Einstellung der CO2 Grenze

andig, 2023-04-14:
@naltatis das ist ein CI build- ist da im toml was abhanden gekommen?

naltatis, 2023-04-14:
Viel banaler. Bei den Grid-Settings hatte ich einfach den Translation Key vergessen. Beim Ladeplan war der da, in der deutschen Datei stand aber noch der englische "none" Text drin. Ist nun gefixt.

💚 Danke fürs Finden.

PR #7461 (2023-04-14): Translation: kein Plan, keine Grenze
- added missing translation key in grid modal
- adjusted German translation for no plan

fixes #7434
closed 2023-04-14 by naltatis