evcc-io/evcc #7440: Problem with Easee Home under 0.116.0

cschlipf, 2023-04-14:
### Describe the bug

After upgrading to 0.116.0 I get this error:
cannot create charger 'easee-garage'
cannot create charger 'template'
cannot create charger 'easee'
Get "https://api.easee.cloud/api/chargers/XXXXXX/site"
context deadline exceeded (Client.Timeout exceeded while awaiting headers)

### Steps to reproduce

1. Upgrade to 0.116.0
2. Try to launch EVCC -> Error shown
3. Downgrade to 0.115.0 -> Working again
...

### Configuration details

```yaml
loadpoints:
  - title: Garage
    charger: easee-garage
    vehicle: tesla-m3-lr
    mode: pv
    resetOnDisconnect: true
    phases: 0
    minCurrent: 6
    maxCurrent: 16

chargers:
  - name: easee-garage
    type: template
    template: easee
    user: XXXX # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: XXXX # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    charger: XXXX
```

### Log details

```text
cschlipf@home:~ $ evcc --log debug
[main  ] INFO 2023/04/14 11:05:35 evcc 0.116.0
[main  ] INFO 2023/04/14 11:05:35 using config file: /home/cschlipf/evcc.yaml
[main  ] INFO 2023/04/14 11:05:35 starting ui and api at :7070
[db    ] INFO 2023/04/14 11:05:35 using sqlite database: /home/cschlipf/.evcc/evcc.db
[main  ] FATAL 2023/04/14 11:05:46 cannot create charger 'easee-garage': cannot create charger 'template': cannot create charger 'easee': i/o timeout
[main  ] FATAL 2023/04/14 11:05:46 will attempt restart in: 5m0s
[server] FATAL 2023/04/14 11:05:46 unlinkat /tmp/evcc-7070: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.116.0

Snited, 2023-04-14:
same problem here

foto-andreas, 2023-04-14:
Hatte ich vorhin auch, aktuell scheint es aber wieder zu funktionieren. Ich vermute ein temporäres Problem bei Easee - wie leider öfter mal.

lanzahn, 2023-04-14:
Same problem with SMA since the update:

[image]

Works with a restart for some time until it seems a timeout is detected. Downgrading seems to have resolved the issue for now. I will also check the network for issues.

cschlipf, 2023-04-14:
> Hatte ich vorhin auch, aktuell scheint es aber wieder zu funktionieren. Ich vermute ein temporäres Problem bei Easee - wie leider öfter mal.

Unter 0.115 ging es aber. Zweimal hin und hergeswitched. Beide Mal ging es unter 0.116 nicht, unter 0.115 sofort wieder.

Snited, 2023-04-14:
> > Hatte ich vorhin auch, aktuell scheint es aber wieder zu funktionieren. Ich vermute ein temporäres Problem bei Easee - wie leider öfter mal.
> 
> Unter 0.115 geht es aber. Zweimal hin und hergeswitched. Beide Mal ging es unter 0.116 nicht, unter 0.115 sofort wieder.

kann ich so bestätigen

souhl, 2023-04-14:
Ich habe das auch immer wieder seid gestern mit der Easee. Im Easee Cloud Status wird nichts angezeigt.

sash2kk, 2023-04-14:
Ich habe das Problem auch - allerdings mit 0.115. Ich vermute tatsächlich ein Problem bei Easee (obwohl bei Status alles auf grün). Hoffentlich nicht schon der Untergang nach dem Verkaufsverbot und Massenentlassungen.

cschlipf, 2023-04-14:
Es scheint derzeit aber auch in der Tat Probleme mit Easee Home zu geben. Auch unter Version 115:

[image]

Eventuell ist 0.115 nur etwas fehlertoleranter als 0.116? 0.115 startet wenigstens noch.

Snited, 2023-04-14:
für die HACS Integration der Easee gab es auch ein update kürzlich, wegen veränderungen an der Easee API... Vielleicht kommts auch aus der Richtung.
Vor zwei tagen war das Update: "Recent changes in the API from Easee has required modifications in the integration to keep everyting working as before."

cschlipf, 2023-04-14:
> Ich habe das Problem auch - allerdings mit 0.115. Ich vermute tatsächlich ein Problem bei Easee (obwohl bei Status alles auf grün). Hoffentlich nicht schon der Untergang nach dem Verkaufsverbot und Massenentlassungen.

Verkaufsverbot aktuell doch noch nur in Schweden oder wurde das etwa schon ausgeweitet?

Snited, 2023-04-14:
> Verkaufsverbot aktuell doch noch nur in Schweden oder wurde das etwa schon ausgeweitet?

die Duskussion ist quatsch. Selbst in Schweden dürfen die bereits installierten Wallboxen ganz normal weiter betrieben werden. Das hat ja mit der API nichts zu tun.

sash2kk, 2023-04-14:
> Verkaufsverbot aktuell doch noch nur in Schweden oder wurde das etwa schon ausgeweitet?

Nein, bislang nur in Schweden. Das Verkaufsverbot selbst macht mir erstmal wenig Sorgen - die aktuellen Entlassungen schon mehr. Gibt es eher selten, dass die (Service-)Qualität bei Diensten dadurch besser wird.

cschlipf, 2023-04-14:
> > > Ich habe das Problem auch - allerdings mit 0.115. Ich vermute tatsächlich ein Problem bei Easee (obwohl bei Status alles auf grün). Hoffentlich nicht schon der Untergang nach dem Verkaufsverbot und Massenentlassungen.
> > 
> > 
> > Verkaufsverbot aktuell doch noch nur in Schweden oder wurde das etwa schon ausgeweitet?
> 
> Nein, bislang nur in Schweden. Das Verkaufsverbot selbst macht mir erstmal wenig Sorgen - die aktuellen Entlassungen schon mehr. Gibt es eher selten, dass die (Service-)Qualität bei Diensten dadurch besser wird. Aber vielleicht wieder Back to topic...
> 
> Ich habe mal ein Ticket bei Easee aufgemacht. Aber so wie es klingt, gab es ja scheinbar vor ein paar Tagen eine Anpassung der Cloud an die evcc noch nicht angepasst ist?!

Dann wäre es aber komisch, dass die alte Version besser läuft als die neue.

cschlipf, 2023-04-14:
Update! Die Probleme in der Easee Cloud scheinen gerade weg zu sein.

Habe auf 0.116 upgedatet und siehe da - es läuft nun auch unter 0.116:
[image]

Also kein prinzipieller Fehler von 0.116.0, aber eventuell müsste an der Fehlertoleranz beim Startup was gemacht werden. Ein temporärer Fehler in der Easee Cloud sollte nicht verhindern, dass EVCC starten kann.

closed 2023-04-14 by andig