evcc-io/evcc #7446: Fahrzeugerkennung stört Ladevorgang

TecTim18, 2023-04-14:
### Discussed in https://github.com/evcc-io/evcc/discussions/7444

<div type='discussions-op-text'>

<sup>Originally posted by **TecTim18** April 14, 2023</sup>
Hallo zusammen,

leider bin ich mit dem Update 0.116.0 nicht ganz so zufrieden.
Bei uns ist das Laden aktuell nicht mehr möglich, da die Fahzeugerkennung immer wieder das ausgewählte Fahrzeug kickt und auf Gast-oder den 60km entfernten Zweitwagen auswählt, da dieser dort gerade geladen wird, was EVCC über die App Anbindung sieht und als zuhause ladendes Fahrzeug interpretiert.

- Ich nutze EVCC via HomeAssistant Add-on
- Vor dem Update gab es mit der Fahrzeugauswahl und automatischen Erkennung (die nichts didektieren kann) keine Probleme
- Einmal händisch eingestellt blieb das Fahrzeug gesetzt bis es abgesteckt wurde
- Im Screenshot ist ein abgebrochener Ladevorgang zu sehen bei welchem unser Kona zuvor 6s geladen hat und dann gestoppt wurde und zugleich die Fahrzeugerkennung los lief

[image]

Ich kann in den ChangeLogs nichst erkennen, dass auf eine Anpassung im Bereich Fahrzeugerkennung hinweist. Gibt es dennoch etwas, das hierfür verantwortlich sein kann?
Wie erwähnt, hat die 0.115.0 tadellos funktioniert 👌 

EVCC yaml:
```
network:
  schema: http
  host: evcc.local
  port: 7070
interval: 3s # control cycle interval

sponsortoken: XXX
telemetry: true
plant: XXX
log: trace

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    token: "XXX"

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

meters:
  - type: template
    template: senec-home
    usage: grid
    host: 192.168.20.12
    name: grid_meter
  - type: template
    template: senec-home
    usage: pv
    host: 192.168.20.12
    name: pv_meter
  - type: template
    template: senec-home
    usage: battery
    host: 192.168.20.12
    name: battery

chargers:
  - name: go-e_V3_WB
    type: template
    template: go-e-v3
    host: 192.168.20.13
  - name: go-e_WB
    type: template
    template: go-e
    host: 192.168.20.14

loadpoints:
  - title: Garage
    charger: go-e_V3_WB
    mode: pv # Standard Modus des Ladepunktes
    phases: 0 # Phasenumschaltung mit 0 aktiviert
    mincurrent: 6
    maxcurrent: 16
    resetOnDisconnect: true # Rueksetzung auf Standard-Modus des Ladepunktes nach beenden des Ladevorgangs
    enable:
      delay: 2m # Dauer ueber die ausreichend Ueberschuss vorhanden sein muss um Ladung zu starten
    disable:
      delay: 5m # Dauer uber die auch ohne genuegend Ueberschuss weiter geladen wird
    priority: 1

  - title: Carport
    charger: go-e_WB
    mode: pv # Standard Modus des Ladepunktes
    phases: 3
    mincurrent: 6
    maxcurrent: 16
    resetOnDisconnect: true # Rueksetzung auf Standard-Modus des Ladepunktes nach beenden des Ladevorgangs
    enable:
      delay: 3m # Dauer ueber die ausreichend Ueberschuss vorhanden sein muss um Ladung zu starten
    disable:
      delay: 5m # Dauer uber die auch ohne genuegend Ueberschuss weiter geladen wird
    priority: 0

vehicles:
  - name: Twingo
    type: template
    template: renault
    title: Renault Twingo
    user: XXX
    password: XXX
    capacity: 21 # Akku-Kapazität in kWh
    mincurrent: 8
    targetSoc: 80

  - name: Hyundai Kona
    type: template
    template: offline
    title: Hyundai Kona # Wird in der Benutzeroberfläche angezeigt # Optional
    capacity: 64 # Akkukapazität in kWh # Optional

site:
  title: Energie Management Fremuth
  meters:
    grid: grid_meter
    pvs: pv_meter
    batteries: battery

influx:
  url: http://192.168.20.10:8086
  database: XXX
  user: XXX
  password: XXX

mqtt:
  broker: 192.168.20.10:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: XXX
  password: XXX
```

</div>

Born91, 2023-04-14:
Ich hab ähnliche Probleme. Der Server start im Minutentakt neu, meldet nach dem Neustart Probleme mit dem Fahrzeug. Aktuell ist ein Überschussladen aufgrund des Fehlers nicht mehr vernünftig möglich. bis 0.115.0 hatte ich auch wirklich keinerlei Probleme.

atddo, 2023-04-14:
> Ich hab ähnliche Probleme. Der Server start im Minutentakt neu, meldet nach dem Neustart Probleme mit dem Fahrzeug. Aktuell ist ein Überschussladen aufgrund des Fehlers nicht mehr vernünftig möglich. bis 0.115.0 hatte ich auch wirklich keinerlei Probleme.

Hallo, da schließe ich mich an.
Ich hab HomeAssistant im Einsatz mit go-e Charger und Sungrow-Hybrid-Wechselrichter, aktuell noch ohne installierten Akku.

Born91, 2023-04-14:
> Hallo, da schließe ich mich an. Ich hab HomeAssistant im Einsatz mit go-e Charger und Sungrow-Hybrid-Wechselrichter, aktuell noch ohne installierten Akku.

Bin ja schon mal froh nicht der einzige zu sein. Hab gerade dazu ein extra Issue aufgemacht, da es nur im erweiterten Sinne mit diesem Problem was zu tun hat.
Nutze auch eine Go-E aber mit einem Huawei WR.

tobiasbayer, 2023-04-14:
Ich habe das Problem aus. Es scheint nicht direkt an evcc zu liegen. Sobald man den Watchdog in Home Assistant ausschaltet, läuft der evcc-Server durch. Der Watchdog scheint zu denken, evcc wäre abgestürzt und startet in deswegen andauernd neu.

andig, 2023-04-14:
Generell: ohne Logfile werden solche Issues nicht bearbeitet mangels ausreichender Information. Euer Restartproblem wäre auch im Logfile sichtbar geworden.

closed 2023-04-14 by andig

Born91, 2023-04-14:
> Generell: ohne Logfile werden solche Issues nicht bearbeitet mangels ausreichender Information. Euer Restartproblem wäre auch im Logfile sichtbar geworden.

Problem ist mit dem Restart ohne Watchdog nur halb gelöst. Eigentlich besteht es immer noch. Ich bin selbst nicht so tief in der Materie. Leider weis ich nicht wo ich die Logfiles in HA finde. Und in der Doku finde ich auch nichts in Verbindung mit HA. Von daher sollte da vielleicht schon nach einer richtigen Lösung gesucht werden.

Born91, 2023-04-14:
Nur schlecht, dass das Problem scheinbar nur in HA auftritt. Und das exakt in der Nacht des updates auf 0.116.0. Vorher lief alles ohne jegliche Probleme. Und wie ich in den Kommentaren lese, ist es ja kein Einzelfall. Auch in der Facebook Gruppe ist es von dem Problem geplagte User.

andig, 2023-04-14:
@Born91 das HA Image wird aus der Community bereit gestellt. Wir supporten das nicht- da hilft auch kein Daumen runter. Und selbst wenn: ohne weitere Informationen ist da einfach nichts zu machen.

StefanSchoof, 2023-04-14:
Ich denke das Issue für das HA Thema ist https://github.com/evcc-io/hassio-addon/issues/61