evcc-io/evcc #7447: Messaging bei Fahrzeugwechsel

harrykausl, 2023-04-14:
### Discussed in https://github.com/evcc-io/evcc/discussions/7409

<div type='discussions-op-text'>

<sup>Originally posted by **harrykausl** April 13, 2023</sup>
Ich benutze das Messaging mit events zur Integration in meine Domoticz-Automation. Ich habe jetzt zusätzlich zu meinem Fiat 500e einen Mercedes B250e angelegt. Ich habe keinen Standard (vehicle) mehr in loadpoints, d.h. bei Anstecken kommt zuerst Gastfahrzeug, der Fiat wird dann erkannt, der Mercedes muss im web-Interface geschaltet werden. Ich habe die events connect, disconnet, start und stop konfiguriert. Es scheint so, als ob beim Wechsel des Fahrzeugs in evcc, egal ob automatisch per Erkennung oder von Hand kein connect bzw start connect mehr getriggert wird. Beim Beenden klappt es (da war ja auch kein Wechsel).  </div>

harrykausl, 2023-05-14:
Ich denke das funktioniert immer noch nicht richtig. Ich habe derzeit meine B-Klasse als Standard definiert.  Wenn ich jetzt den Fiat anstecke und genügend Strom vorhanden ist, bzw die Einstellung zum Laden passt, wird sofort geladen. Erwartungsgemäß wird die B-Klasse mit wird geladen in evcc angezeigt und auch über das Messaging getriggert. Wenn ich dann in evcc das Fahrzeug auf den Fiat umstelle, bekomme ich keinen event getriggert. Erklärt sich m.E. daraus, das sich der Status (angeschlossen und wird geladen) nicht geändert hat.

andig, 2023-05-14:
Du vermutest. Die Wahrheit steht in Logfile.

harrykausl, 2023-05-16:
Da ich in meinem Messaging script alles protokolliere, sehe ich dass in diesem Fall nichts kommt. Bei  nächsten Mal werde ich es loggen.

harrykausl, 2023-05-17:
Hier der Logfile (debug) sowie mein Tracing des Messaging
in meinem eigenen Trace sehe ich, dass nach dem Umschalten des Fahrzeuges auf den Fiat das Messaging wieder mit dem Start für den Mercedes gestartet wird.

Wed 17 May 10:44:41 CEST 2023
Script: /home/pi/scripts_shell/wallbox_status.sh
Alle Parameter: connect B-250e
Anzahl: 2
Alle *: connect B-250e
1: connect
2: B-250e
Wed 17 May 10:44:51 CEST 2023
Script: /home/pi/scripts_shell/wallbox_status.sh
Alle Parameter: start B-250e
Anzahl: 2
Alle *: start B-250e
1: start
2: B-250e
Wed 17 May 10:48:26 CEST 2023
Script: /home/pi/scripts_shell/wallbox_status.sh
Alle Parameter: start B-250e
Anzahl: 2
Alle *: start B-250e
1: start
2: B-250e

May 17 00:14:39 raspberrypi evcc[29278]: [lp-1  ] ERROR 2023/05/17 00:14:39 charger: Get "http://192.168.2.53/status": read tcp 192.168.2.17:48134->192.168.2.53:80: read: connection reset by peer
May 17 10:43:47 raspberrypi systemd[1]: Stopping evcc...
May 17 10:43:47 raspberrypi systemd[1]: evcc.service: Succeeded.
May 17 10:43:47 raspberrypi systemd[1]: Stopped evcc.
May 17 10:43:47 raspberrypi systemd[1]: Started evcc.
May 17 10:43:48 raspberrypi evcc[17885]: [main  ] INFO 2023/05/17 10:43:48 evcc 0.117.3
May 17 10:43:48 raspberrypi evcc[17885]: [main  ] INFO 2023/05/17 10:43:48 using config file: /etc/evcc.yaml
May 17 10:43:48 raspberrypi evcc[17885]: [main  ] INFO 2023/05/17 10:43:48 starting ui and api at :7070
May 17 10:43:48 raspberrypi evcc[17885]: [db    ] INFO 2023/05/17 10:43:48 using sqlite database: /var/lib/evcc/evcc.db
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] WARN 2023/05/17 10:43:50 vehicle 'B-250e' does not support automatic detection
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] INFO 2023/05/17 10:43:50 site config:
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] INFO 2023/05/17 10:43:50   meters:      grid ✓ pv ✓ battery ✓
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] INFO 2023/05/17 10:43:50     grid:      power ✓ energy ✓ currents ✓
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] INFO 2023/05/17 10:43:50     pv 1:      power ✓ energy ✓ currents ✗
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] INFO 2023/05/17 10:43:50     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] INFO 2023/05/17 10:43:50   vehicles:
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] INFO 2023/05/17 10:43:50     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] INFO 2023/05/17 10:43:50     vehicle 2: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] INFO 2023/05/17 10:43:50 loadpoint 1:
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] INFO 2023/05/17 10:43:50   mode:        pv
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] INFO 2023/05/17 10:43:50   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] INFO 2023/05/17 10:43:50   meters:      charge ✓
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] INFO 2023/05/17 10:43:50     charge:    power ✓ energy ✓ currents ✓
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:50 phase timer inactive
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:50 pv timer inactive
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:50 guard timer inactive
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] INFO 2023/05/17 10:43:50 vehicle updated: unknown -> B-250e
May 17 10:43:50 raspberrypi evcc[17885]: [main  ] INFO 2023/05/17 10:43:50 ensure the oauth client redirect/callback is configured for B-250e: http://192.168.2.17:7070/oauth/vehicles/1/callback
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] DEBUG 2023/05/17 10:43:50 ----
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:50 charge power: 0W
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] DEBUG 2023/05/17 10:43:50 pv power: 6240W
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] DEBUG 2023/05/17 10:43:50 battery soc: 99%
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] DEBUG 2023/05/17 10:43:50 battery power: 1W
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] DEBUG 2023/05/17 10:43:50 grid power: -5942W
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] DEBUG 2023/05/17 10:43:50 grid currents: [-7.79 -8.38 -8.38]A
May 17 10:43:50 raspberrypi evcc[17885]: [site  ] DEBUG 2023/05/17 10:43:50 site power: -5941W
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:50 charge voltages: [247 247 246]V
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:50 detected connected phases: 3p
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:50 charge currents: [0 0 0]A
May 17 10:43:50 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:50 charge meter total import: 1641.600kWh
May 17 10:43:51 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:51 charger status: A
May 17 10:43:51 raspberrypi evcc[17885]: [lp-1  ] INFO 2023/05/17 10:43:51 car disconnected
May 17 10:43:51 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:51 set charge mode: pv
May 17 10:43:51 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:51 set min current: 6
May 17 10:43:51 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:43:51 set max current: 16
[... 271 lines cut ...]
[... cut ...]
May 17 10:46:31 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:46:31 charger status: C
May 17 10:46:31 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:46:31 soc estimated: 73.26% (vehicle: 73.00%)
May 17 10:46:31 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:46:31 vehicle soc: 73%
May 17 10:46:31 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:46:31 vehicle range: 167km
May 17 10:46:31 raspberrypi evcc[17885]: [lp-1  ] DEBUG 2023/05/17 10:46:31 pv charge current: 7A = 7A + 0.0013A (-1W @ 3p)

andig, 2023-05-18:
Ich hab mir das nochmal angeschaut. "connect" wird tatsächlich nur bei Verbindung ausgelöst (Änderung Wallboxstatus), aber nicht bei Fahrzeugänderung.

harrykausl, 2023-05-18:
Könntet Ihr das dann mal reinmachen?