evcc-io/evcc #7452: Server stürzt sein 0.116.0 beim Laden ständig ab

Born91, 2023-04-14:
### Describe the bug

Server stützt ständig ab. Phasenwechsel nicht zuverlässig. Anschließend immer Fehlermeldung zu „telemetry“. 
Eine Nutzung von EVCC ist somit gerade nicht möglich.
In der Nach von dem Update traten schon die ersten nicht verständlichen Probleme auf, das in meiner Grafana Ansicht ständig Ausschläge vom Loadpoint (grün) waren und ich die Ursache erst mal nicht verstanden habe.

[image]
[image]
[image]

### Steps to reproduce

Ladevorgang starten.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant:

interval: 10s # control cycle interval

sponsortoken:

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.123  
  port: 502  
  usage: grid  
  storageunit: 1  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.123  
  port: 502  
  usage: pv  
  storageunit: 1  
  modbus: tcpip  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.123  
  port: 502  
  usage: battery  
  storageunit: 1  
  capacity: 5
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.89  
  name: wallbox5

vehicles:
- type: template
  template: fiat 
  title: Luigi  
  user: 
  
  capacity: 37.3  
  phases: 3  
  icon: car  
  cache: 20m  
  pin: '0852'  
  mode: PV  
  minSoc: 5  
  targetSoc: 85  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  guardduration: 4m

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  prioritySoc: 95

tariffs:
  currency: EUR
  grid:
    type: tibber
    token:
  feedin:
    type: fixed
    price: 0.082

influx:
  url: http://localhost:8086
  database: evcc
  user: evcc
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/04/14 15:29:47 evcc 0.116.0
[main  ] INFO 2023/04/14 15:29:47 using config file: /config/evcc.yaml
[main  ] INFO 2023/04/14 15:29:47 starting ui and api at :7070
[db    ] INFO 2023/04/14 15:29:47 using sqlite database: /data/evcc.db
[site  ] INFO 2023/04/14 15:29:50 site config:
[site  ] INFO 2023/04/14 15:29:50   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/04/14 15:29:50     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/04/14 15:29:50     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/04/14 15:29:50     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/04/14 15:29:50   vehicles:
[site  ] INFO 2023/04/14 15:29:50     vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/04/14 15:29:50 loadpoint 1:
[lp-1  ] INFO 2023/04/14 15:29:50   mode:        pv
[lp-1  ] INFO 2023/04/14 15:29:50   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/04/14 15:29:50   meters:      charge ✓
[lp-1  ] INFO 2023/04/14 15:29:50     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2023/04/14 15:29:50 vehicle updated: unknown -> Luigi
```

### What type of operating system are you running?

Linux

### Version

0.116.0

Gulaschkanone88, 2023-04-14:
Hab das gleiche Problem, es setzt jedes mal innerhalb weniger Sekunde die Abfahrtszeit und den Ladestand zurück. So ist es aktuell nicht wirklich nutzbar.

atddo, 2023-04-14:
Ich hab HomeAssistant im Einsatz mit go-e Charger und Sungrow-Hybrid-Wechselrichter, aktuell noch ohne installierten Akku, wie in #7446 geschrieben.

Container startet sich ca jede Minute neu.

tobiasbayer, 2023-04-14:
https://github.com/evcc-io/evcc/issues/7446#issuecomment-1508560734

Born91, 2023-04-14:
> comment)

Scheint zu helfen! Läuft seit her stabil. Auch der automatische Phasenwechsel lief Reibungslos ab.
Danke!

closed 2023-04-14 by Born91

tobiasbayer, 2023-04-14:
Schließen würde ich das noch nicht. Den Watchdog auszuschalten ist ja nur ein Workaround.