evcc-io/evcc #7458: Charging log: Start and end meter not set

Maddogtet, 2023-04-14:
### Describe the bug

Following up of the discussion in https://github.com/evcc-io/evcc/pull/7424

The start and end meter values are not shown in the charging log.
It shows 0 as start and end meter values. 
The wallbox is connected by the ocpp charger template.

### Steps to reproduce

Use a OCPP connected Wallbox
1. Start fast charging session
2. Charge for about 0,1 KWh
3. Stop the charging session
4. Open the charging log -> meter value start and meter value end = 0

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error
  site: trace

interval: 10s # control cycle interval

influx:
  url: http://localhost:8086
  database: evcc
  token: 
  org: Home

vehicles:
- name: ev1
  type: template
  template: tesla 
  title: Model3 (dienstlich)  
  accessToken:
  vin:   
  capacity: 75  
  identifiers:
    - 36426DF1 #Dienstlich Token

- name: ev2
  type: template
  template: tesla 
  title: Model3 (privat)
  accessToken:
  vin: 
  capacity: 75
  identifiers:
    - D66D75F1 #Privat Token

site:
  title: Mein Zuhause
  meters: 
    grid: my_grid
    pv: my_pv
    battery: my_battery
  bufferSoc: 70
  prioritySoc: 65
  residualPower: 250
  maxGridSupplyWhileBatteryCharging: 650

meters:
  - name: my_grid
    type: custom
    power:
      source: calc
      add:
      - source: http
        uri: http://srvdeb:8080/rest/items/TibberAPI_Powerconsumptionproduction
        method: GET
        jq: .state
      - source: http
        uri: http://srvdeb:8080/rest/items/TibberAPI_Powerproduction
        method: GET
        jq: .state
        scale: -1
    currents:
      - source: http
        uri: http://srvdeb:8080/rest/items/TibberAPI_Current1
        method: GET
        jq: .state | rtrimstr(" A")
      - source: http
        uri: http://srvdeb:8080/rest/items/TibberAPI_Current2
        method: GET
        jq: .state | rtrimstr(" A")
      - source: http
        uri: http://srvdeb:8080/rest/items/TibberAPI_Current3
        method: GET
        jq: .state | rtrimstr(" A")

  - name: my_pv
    type: custom
    power:
      source: http
      uri: http://srvdeb:8080/rest/items/PVGeneratorPower
      method: GET
      jq: .state

  - name: my_battery
    type: custom
    power: 
      source: http
      uri: http://srvdeb:8080/rest/items/BatteryPowerCaculated
      method: GET
      jq: .state
      scale: -1 # here it is the other way around
    soc: 
      source: http
      uri: http://srvdeb:8080/rest/items/HTTP_URL_Thing_InverterResponse_BatterySoc
      method: GET
      jq: .state
    capacity: 9.2

chargers:
  - name: wallbox
    type: template
    template: ocpp 

    idtag: D66D75F1
    stationid: ABL_10022482
    connector: 1

loadpoints: 
- title: Garage
  charger: wallbox # Wallbox Referenz
  vehicle: ev1 # Referenz auf Standardfahrzeug
  mode: pv # c
  mincurrent: 6
  maxcurrent: 16
  phases: 0 # automatische Phasenumschaltung
  enable:
    threshold: 0
    delay: 1m #1m
  disable:
    threshold: 0
    delay: 30s #low value because we have the battery

tariffs:
  grid:
    # either
    type: tibber
    cheap: 0.2 # EUR/kWh
    token: "" # access token
    homeid: "" # optional if multiple homes associated to account
  
  feedin:
   type: fixed
   price: 0.08 # 0,12 EUR/kWh
```

### Log details

```text
Will attach the file. Content is to long
```

### What type of operating system are you running?

Linux

### Version

0.116.0 (4f96f736)

Maddogtet, 2023-04-14:
Log attached
[evcc_2023-04-14.log](https://github.com/evcc-io/evcc/files/11235815/evcc_2023-04-14.log)

andig, 2023-04-14:
That's interesting:

```
[ocpp  ] TRACE 2023/04/14 19:37:57 enqueued CALL [3538556300, RemoteStartTransaction] for ABL_10022482
[ocpp  ] TRACE 2023/04/14 19:37:57 dispatched request 3538556300 for ABL_10022482
[ocpp  ] TRACE 2023/04/14 19:37:57 sent JSON message to ABL_10022482: [2,"3538556300","RemoteStartTransaction",{"connectorId":1,"idTag":"D66D75F1","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16}]}}}]
[ocpp  ] TRACE 2023/04/14 19:38:27 completed request 3538556300 for ABL_10022482
[ocpp  ] TRACE 2023/04/14 19:38:27 request 3538556300 for ABL_10022482 timed out
[lp-1  ] ERROR 2023/04/14 19:38:27 charger enable: ocpp message (3538556300): GenericError - Request timed out
```

It would be nice to ask ABL why they timeout on this request.

andig, 2023-04-14:
The values are there but actually using them is not so easy. As workaround you could try adding

    MeterInterval: 60s

to the config. That should trigger regular meter requests.

Maddogtet, 2023-04-14:
looks like this parameter does not work on the ocpp charger?

```
Apr 14 21:57:15 srvdeb evcc[577939]: [main  ] FATAL 2023/04/14 21:57:15 cannot create charger 'wallbox': cannot create charger 'template': invalid key: meterinterval

```

andig, 2023-04-14:
Habs hinzugefügt, in 15min neues nightly. Hoffe es funktioniert...

Maddogtet, 2023-04-14:
Leider immer noch nichts bei den Zählerständen :(

[evcc_2023-04-14_1.log](https://github.com/evcc-io/evcc/files/11236809/evcc_2023-04-14_1.log)
[image]

andig, 2023-04-15:
Mhhm. Die Werte kommen. Könntest Du bei gestopptem evcc mal bitte ein

    evcc charger

machen? Bin gespannt ob da Zählerstände auftauchen.

Maddogtet, 2023-04-15:
Beim ersten Mal ausführen:
```
[main  ] INFO 2023/04/15 12:30:51 evcc 0.116.1 (3d7cd9f1)
[main  ] INFO 2023/04/15 12:30:51 using config file: /etc/evcc.yaml
[db    ] INFO 2023/04/15 12:30:51 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/04/15 12:30:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/04/15 12:30:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/04/15 12:30:51 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/04/15 12:30:51 SELECT * FROM `settings` 6 <nil>
[ocpp  ] WARN 2023/04/15 12:30:51 chargepoint connected, unknown: ABL_10022482
[ABL_10022482] DEBUG 2023/04/15 12:30:52 waiting for chargepoint: 5m0s
[ocpp  ] TRACE 2023/04/15 12:30:52 enqueued CALL [4024744775, GetConfiguration] for ABL_10022482
[ocpp  ] TRACE 2023/04/15 12:30:52 dispatched request 4024744775 for ABL_10022482
[ocpp  ] TRACE 2023/04/15 12:30:52 sent JSON message to ABL_10022482: [2,"4024744775","GetConfiguration",{}]
[ocpp  ] TRACE 2023/04/15 12:30:52 started timeout timer for ABL_10022482
[ocpp  ] TRACE 2023/04/15 12:30:52 received JSON message from ABL_10022482: [3, "4024744775" ,{"configurationKey":[{"key":"ConnectionTimeOut","readonly":false,"value":"0"},{"key":"HeartBeatInterval","readonly":false,"value":"60"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"MeterValuesAlignedData","readonly":false,"value":""},{"key":"StopTxnAlignedData","readonly":false,"value":""},{"key":"MeterValueSampleInterval","readonly":false,"value":"60"},{"key":"MinimumStatusDuration","readonly":false,"value":"0"},{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register"},{"key":"StopTxnSampledData","readonly":false,"value":"Energy.Active.Import.Register"},{"key":"LocalPreAuthorize","readonly":false,"value":"false"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"false"},{"key":"StopTransactionOnInvalidId","readonly":false,"value":"false"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"true"},{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"false"},{"key":"WebSocketPingInterval","readonly":false,"value":"30"},{"key":"PushMeterConfiguration","readonly":false,"value":"true"},{"key":"LateOccupied","readonly":false,"value":"false"},{"key":"AccessPointName","readonly":false,"value":""},{"key":"AccessPointUser","readonly":false,"value":""},{"key":"AccessPointPassword","readonly":false,"value":""},{"key":"ChargeBoxId","readonly":false,"value":"ABL_10022482"},{"key":"ServiceURL","readonly":false,"value":"ws://192.168.100.68:8887/"},{"key":"OcppVersion","readonly":false,"value":"1.6"},{"key":"Comments","readonly":false,"value":""},{"key":"LogLevel","readonly":false,"value":"DEBUG"},{"key":"LogOcppMessages","readonly":false,"value":"true"},{"key":"LogOcppInvocations","readonly":false,"value":"true"},{"key":"FreeCharging","readonly":false,"value":"false"},{"key":"FreeChargingUid","readonly":false,"value":"00000000000000"},{"key":"FreeChargingOffline","readonly":false,"value":"false"},{"key":"GSMForceReconnect","readonly":false,"value":"true"},{"key":"PowerTimeout","readonly":false,"value":"600"},{"key":"HandleNewTransaction","readonly":false,"value":"Cancellation"},{"key":"HandleOldTransaction","readonly":false,"value":"ReenableOld"},{"key":"HandleExpiredTransaction","readonly":false,"value":"Cancellation"},{"key":"ShortenUIDs","readonly":false,"value":"true"},{"key":"AuthorizationKey","readonly":false,"value":""},{"key":"SecurityProtocols","readonly":false,"value":"TLSv13"},{"key":"TransmitMeterSignature","readonly":false,"value":"false"},{"key":"EndSignatureType","readonly":false,"value":"SIGNATURE_TRANSACTION"},{"key":"DisableDowngrading","readonly":false,"value":"true"},{"key":"RequestOwnerPasswords","readonly":true,"value":"false"}],"unknownKey":[]}]
[ocpp  ] TRACE 2023/04/15 12:30:53 handling incoming CALL RESULT [4024744775] from ABL_10022482
[ocpp  ] TRACE 2023/04/15 12:30:53 completed request 4024744775 for ABL_10022482
[ABL_10022482] TRACE 2023/04/15 12:30:53 AccessPointName (r/w): 
[ABL_10022482] TRACE 2023/04/15 12:30:53 AccessPointPassword (r/w): 
[ABL_10022482] TRACE 2023/04/15 12:30:53 AccessPointUser (r/w): 
[ABL_10022482] TRACE 2023/04/15 12:30:53 AllowOfflineTxForUnknownId (r/w): false
[ABL_10022482] TRACE 2023/04/15 12:30:53 AuthorizationCacheEnabled (r/w): true
[ABL_10022482] TRACE 2023/04/15 12:30:53 AuthorizationKey (r/w): 
[ABL_10022482] TRACE 2023/04/15 12:30:53 ChargeBoxId (r/w): ABL_10022482
[ABL_10022482] TRACE 2023/04/15 12:30:53 ClockAlignedDataInterval (r/w): 0
[ABL_10022482] TRACE 2023/04/15 12:30:53 Comments (r/w): 
[ABL_10022482] TRACE 2023/04/15 12:30:53 ConnectionTimeOut (r/w): 0
[ABL_10022482] TRACE 2023/04/15 12:30:53 DisableDowngrading (r/w): true
[ABL_10022482] TRACE 2023/04/15 12:30:53 EndSignatureType (r/w): SIGNATURE_TRANSACTION
[ABL_10022482] TRACE 2023/04/15 12:30:53 FreeCharging (r/w): false
[ABL_10022482] TRACE 2023/04/15 12:30:53 FreeChargingOffline (r/w): false
[ABL_10022482] TRACE 2023/04/15 12:30:53 FreeChargingUid (r/w): 00000000000000
[ABL_10022482] TRACE 2023/04/15 12:30:53 GSMForceReconnect (r/w): true
[ABL_10022482] TRACE 2023/04/15 12:30:53 HandleExpiredTransaction (r/w): Cancellation
[... cut ...]
[ocpp  ] TRACE 2023/04/15 12:31:39 received JSON message from ABL_10022482: [2, "88016041", "StatusNotification" ,{"connectorId":1,"status":"Preparing","errorCode":"NoError","info":"none","timestamp":"2023-04-15T10:31:39.013+00:00","vendorId":"ABL","vendorErrorCode":"none"}]
[ocpp  ] TRACE 2023/04/15 12:31:39 handling incoming CALL [88016041, StatusNotification] from ABL_10022482
[ocpp  ] TRACE 2023/04/15 12:31:39 sent CALL RESULT [88016041] for ABL_10022482
[ocpp  ] TRACE 2023/04/15 12:31:39 sent JSON message to ABL_10022482: [3,"88016041",{}]
Energy:        2681.7kWh
Charge status: B
Enabled:       false
``` 

Kann es sein, dass da die Bootnotification irgendwie quer schießt?

Maddogtet, 2023-04-15:
[evcc_2023-04-15.log](https://github.com/evcc-io/evcc/files/11239194/evcc_2023-04-15.log)
Noch ein Log vom aktuellen Test von Start EVCC bis Stop EVCC

andig, 2023-04-16:
Nochmal bzgl. https://github.com/evcc-io/evcc/issues/7458#issuecomment-1509119178: da fehlt m.E. vom Charger die Bestätigung des `RemoteStartTransaction`. Es genügt nicht, einfach seinerseits `StartTransaction`zu schicken. Das wäre mal ein Ticket beim Hersteller wert.

andig, 2023-04-16:
Das letzte Log hat schonmal den Startwert:

    [db    ] TRACE 2023/04/15 13:02:44 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`odometer`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`) VALUES ("2023-04-15 13:02:44.581","0000-00-00 00:00:00","Garage","","Model3 (dienstlich)",18312.603871,0.000000,0.000000,0.000000) RETURNING `id` 1 <nil>

andig, 2023-04-16:
Der fehlende Stop Zählerstand wird aus dem Log auch klar:

```
[ocpp  ] TRACE 2023/04/15 13:04:46 received JSON message from ABL_10022482: [2, "27393929", "StatusNotification" ,{"connectorId":1,"status":"Finishing","errorCode":"NoError","info":"none","timestamp":"2023-04-15T11:04:45.659+00:00","vendorId":"ABL","vendorErrorCode":"none"}]
[ocpp  ] TRACE 2023/04/15 13:04:46 handling incoming CALL [27393929, StatusNotification] from ABL_10022482
[ocpp  ] TRACE 2023/04/15 13:04:46 sent CALL RESULT [27393929] for ABL_10022482
[ocpp  ] TRACE 2023/04/15 13:04:46 sent JSON message to ABL_10022482: [3,"27393929",{}]
[site  ] DEBUG 2023/04/15 13:04:46 ----
[lp-1  ] DEBUG 2023/04/15 13:04:46 charge power: 0W
[site  ] DEBUG 2023/04/15 13:04:46 pv power: 548W
[site  ] DEBUG 2023/04/15 13:04:46 battery soc: 53%
[site  ] DEBUG 2023/04/15 13:04:46 battery power: 0W
[site  ] DEBUG 2023/04/15 13:04:46 grid power: 10851W
[site  ] DEBUG 2023/04/15 13:04:46 grid currents: [15.6 15.6 15.7]A
[site  ] DEBUG 2023/04/15 13:04:46 site power: 11101W
[lp-1  ] DEBUG 2023/04/15 13:04:46 charger status: B
[lp-1  ] INFO 2023/04/15 13:04:46 stop charging <-
[db    ] TRACE 2023/04/15 13:04:46 UPDATE `sessions` SET `created`="2023-04-15 13:02:44.581",`finished`="2023-04-15 13:04:46.678",`loadpoint`="Garage",`identifier`="",`vehicle`="Model3 (dienstlich)",`odometer`=18312.603871,`meter_start_kwh`=0.000000,`meter_end_kwh`=0.000000,`charged_kwh`=0.250796 WHERE `id` = 26 1 <nil>
[lp-1  ] DEBUG 2023/04/15 13:04:47 soc estimated: 22.30% (vehicle: 22.00%)
[lp-1  ] DEBUG 2023/04/15 13:04:47 vehicle soc: 22%
[lp-1  ] DEBUG 2023/04/15 13:04:47 vehicle soc limit: 50%
[lp-1  ] DEBUG 2023/04/15 13:04:47 vehicle range: 123km
[ocpp  ] TRACE 2023/04/15 13:04:47 enqueued CALL [1539761635, RemoteStopTransaction] for ABL_10022482
[ocpp  ] TRACE 2023/04/15 13:04:47 dispatched request 1539761635 for ABL_10022482
[ocpp  ] TRACE 2023/04/15 13:04:47 sent JSON message to ABL_10022482: [2,"1539761635","RemoteStopTransaction",{"transactionId":2}]
[ocpp  ] TRACE 2023/04/15 13:04:47 started timeout timer for ABL_10022482
[ocpp  ] TRACE 2023/04/15 13:04:47 received JSON message from ABL_10022482: [3, "1539761635" ,{"status":"Accepted"}]
[ocpp  ] TRACE 2023/04/15 13:04:47 handling incoming CALL RESULT [1539761635] from ABL_10022482
[ocpp  ] TRACE 2023/04/15 13:04:47 completed request 1539761635 for ABL_10022482
[ocpp  ] TRACE 2023/04/15 13:04:47 ABL_10022482 ready to transmit again
[ocpp  ] TRACE 2023/04/15 13:04:47 timeout canceled for ABL_10022482
[ocpp  ] TRACE 2023/04/15 13:04:54 received JSON message from ABL_10022482: [2, "40553753", "StopTransaction" ,{"transactionId":2,"idTag":"D66D75F1","timestamp":"2023-04-15T11:04:53.270+00:00","meterStop":2682402,"reason":"Other","transactionData":[{"timestamp":"2023-04-15T11:02:56.264+00:00","sampledValue":[{"value":"2682.141","context":"Transaction.Begin","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"kWh"}]},{"timestamp":"2023-04-15T11:03:58.994+00:00","sampledValue":[{"value":"2682.254","context":"Sample.Periodic","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"kWh"}]},{"timestamp":"2023-04-15T11:04:53.270+00:00","sampledValue":[{"value":"2682.402","context":"Transaction.End","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"kWh"}]}]}]
[ocpp  ] TRACE 2023/04/15 13:04:54 handling incoming CALL [40553753, StopTransaction] from ABL_10022482
[ocpp  ] TRACE 2023/04/15 13:04:54 sent CALL RESULT [40553753] for ABL_10022482
[ocpp  ] TRACE 2023/04/15 13:04:54 sent JSON message to ABL_10022482: [3,"40553753",{"idTagInfo":{"status":"Accepted"}}]
```

Es kommt erst eine `StatusNotification` die uns den Stop schon mitteilt, aber erst im Anschluss die `StopTransaction` mit den finalen Zählerdaten.

Maddogtet, 2023-04-29:
Update von mir: Logs wurden an ABL übergeben mit der Bitte den Fehler zu beheben, dass RemoteStartTransaction und  RemoteStopTransaction auch bestätigt werden. Ich warte jetzt auf Antwort

andig, 2023-05-13:
Super. Ich mache hier mal zu- der Fix sollte auf Seite der Wallbox liegen.

closed 2023-05-13 by andig