evcc-io/evcc #7465: Mercedes Benz SOC request fail

HiroP77, 2023-04-15:
### Describe the bug

Since 0.116.0, the Request to the MB Service is faulty:

errormessagein the UI:

vehicle soc: Get \"https://api.mercedes-benz.com/vehicledata/v2/vehicles/W1N167117XXXXXX63/containers/electricvehicle\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

(I replaced chars in my Vehicle ident number)
config worked in 0.114.0

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
- type: mercedes
  cache: 5m
  name: XXX
  title: XXX
  capacity: 27
  clientId: 0d54d098-XXXX-XXXX-XXXX-928ce0e479f7
  clientSecret: XXXXXXXXXXXXXXX
  vin: W1N1671XXXXXX63
  phases: 2
  onIdentify:
    minSoc: 10
```

### Log details

```text
[lp-1  ] DEBUG 2023/04/15 13:10:39 charger status: B
[lp-1  ] INFO 2023/04/15 13:10:39 stop charging <-
[lp-1  ] DEBUG 2023/04/15 13:10:39 charge meter total import: 880.000kWh
[lp-1  ] DEBUG 2023/04/15 13:10:46 set charge mode: now
[lp-1  ] DEBUG 2023/04/15 13:10:46 pv timer reset
[lp-1  ] DEBUG 2023/04/15 13:10:46 pv timer inactive
[lp-1  ] DEBUG 2023/04/15 13:10:53 set charge mode: now
[lp-1  ] ERROR 2023/04/15 13:10:59 vehicle soc: Get "https://api.mercedes-benz.com/vehicledata/v2/vehicles/W1XXXXXX7763/containers/electricvehicle": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1  ] DEBUG 2023/04/15 13:10:59 pv charge current: 1.57A = 0A + 1.57A (-721W @ 2p)
[site  ] DEBUG 2023/04/15 13:10:59 ----
[lp-1  ] DEBUG 2023/04/15 13:10:59 charge power: 0W
[site  ] DEBUG 2023/04/15 13:10:59 pv power: 3125W
[site  ] DEBUG 2023/04/15 13:11:01 battery soc: 95%
[site  ] DEBUG 2023/04/15 13:11:01 battery power: -1007W
[site  ] DEBUG 2023/04/15 13:11:01 grid power: 1W
[site  ] DEBUG 2023/04/15 13:11:01 site power: -706W
[lp-1  ] DEBUG 2023/04/15 13:11:01 charge voltages: [224 224 225]V
[lp-1  ] DEBUG 2023/04/15 13:11:01 detected connected phases: 3p
[lp-1  ] DEBUG 2023/04/15 13:11:01 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/04/15 13:11:01 charge meter total import: 880.000kWh
[lp-1  ] DEBUG 2023/04/15 13:11:01 charger status: B
[lp-1  ] DEBUG 2023/04/15 13:11:01 charger: guard elapse
[lp-1  ] DEBUG 2023/04/15 13:11:01 guard timer inactive
[lp-1  ] DEBUG 2023/04/15 13:11:01 charger enable
[lp-1  ] DEBUG 2023/04/15 13:11:01 wake-up timer: start
[site  ] DEBUG 2023/04/15 13:11:08 ----
[lp-1  ] DEBUG 2023/04/15 13:11:08 charge power: 10W
[site  ] DEBUG 2023/04/15 13:11:08 pv power: 3149W
[site  ] DEBUG 2023/04/15 13:11:09 battery soc: 95%
[site  ] DEBUG 2023/04/15 13:11:09 battery power: -934W
[site  ] DEBUG 2023/04/15 13:11:09 grid power: 721W
[site  ] DEBUG 2023/04/15 13:11:09 site power: 87W
[lp-1  ] DEBUG 2023/04/15 13:11:09 charge voltages: [222 223 226]V
[lp-1  ] DEBUG 2023/04/15 13:11:09 detected connected phases: 3p
[lp-1  ] DEBUG 2023/04/15 13:11:09 charge currents: [8.7 8.7 0]A
[lp-1  ] DEBUG 2023/04/15 13:11:09 charge meter total import: 880.000kWh
[lp-1  ] DEBUG 2023/04/15 13:11:09 charger status: C
[lp-1  ] INFO 2023/04/15 13:11:09 start charging ->
[lp-1  ] DEBUG 2023/04/15 13:11:09 wake-up timer: stop
[lp-1  ] ERROR 2023/04/15 13:11:09 vehicle soc: Get "https://api.mercedes-benz.com/vehicledata/v2/vehicles/W1NXXXXXXXX77763/containers/electricvehicle": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
```

### What type of operating system are you running?

Linux

### Version

[main  ] INFO 2023/04/15 13:09:37 evcc 0.116.1

andig, 2023-04-15:
Funktioniert 0.114?

victornet, 2023-04-15:
Bei mir leider mit 114.1 gleiche Fehlermeldung. Möglicherweise ein Problem auf der Gegenseite? Aber dort ist bislang scheinbar nichts bekannt: https://developer.mercedes-benz.com/status

andig, 2023-04-15:
Mercedes schafft es nicht, innerhalb von 10s zu antworten. Bitte dort nachfragen. Alternativ müssten wir einen konfigurierbaren Timeout einbauen.

closed 2023-04-15 by andig