evcc-io/evcc #7466: 1p3p Umschaltung funktioniert bei Gastfahrzeug nicht richtig

MarkusGH, 2023-04-15:
Die Phasenumschaltung orientiert sich an den konfigurierten Maxmalströmen. 
Beispiel:
Wallbox kann Min 6A, Max 32A
Fahrzeug kann 1-Phasig 20A, 2-Phasig 16A
Wenn das Fahrzeug ein Gastfahrzeug ist, sind die Fahrzeuggrenzen unbekannt, es wird von einem 1-Phasigen Maximalstrom von 32A ausgegangen (was wegen Schieflast eher nicht passieren wird).
Somit wird erst auf 3 Phasen umgeschaltet, wenn dauerhaft mehr als ca. 7,2 kW zur Verfügung stehen, geladen wird bis dahin aber nur mit max. 4,6 kW.
Abhilfe wäre möglich entweder wenn gemessen würde wieviel Leistung das Fahrzeug tatsächlich nimmt, oder aber durch Implementation von #4072. Dann würde man maxCurrent1p an der Wallbox auf 16A oder 20A setzen (wegen Schieflast wird kaum eine Wallbox mehr machen) und das Problem wäre zumindest entschärft.

VolkerK62, 2023-04-15:
Du könntest ein passendes Offline-Vehicle konfigurieren. Dort kannst du `phases` und `maxcurrent` definieren.

MarkusGH, 2023-04-15:
Sicherlich richtig. Ich kann das auch. Aber der gemeine Endanwender wird das nicht können (oder das Problem gar nicht erkennen und sich wundern warum PV Strom "verschenkt" wird.

VolkerK62, 2023-04-15:
Die Idee unterschiedlicher currents für 1p/3p ist sicherlich gut, aber ob sich an den damit zusammenhängenden Problemen mittlerweile etwas geändert hat?
Bis dahin wäre das dann etwas für die FAQ.

MarkusGH, 2023-04-15:
Die Probleme kenne ich nur vom Hörensagen. Ich würde meinen, dass der Wert einer solchen Konfigurationsmöglichkeit eventuell unterschätzt wurde. 

Das ist jetzt ein weiterer Fall in dem sich zeigt, dass es ohne eine solche Einstellung wohl nicht gehen wird. Das Problem ist auch nach näherem Überlegen nicht auf das Gastfahrzeug begrenzt. 

Hier habe ich ein Fahrzeug, das man mit maxCurrent 16A oder 20A konfigurieren könnte und man wäre das Problem los. Wenn ein Fahrzeug aber z.B. 1p nur 16 A = 3,6 kW kann (Schieflastbegrenzung!) und 3p 32A = 22 kW dann hat man das Problem, dass man entweder die Ladeleistung auf 11 kW = 16A begrenzen musss um eine funktionierende 1p3p Umschaltung zu bekommen, oder aber man hat den Fall, dass bei  PV Leistung zwischen 3,6 und 7,2 kW nicht umgeschaltet wird, das Fahrzeug aber nur 3,6 kW nimmt. Je nach Größe der PV Anlage kommt der Fall häufig vor.

VolkerK62, 2023-04-15:
>Die Probleme kenne ich nur vom Hörensagen.

https://github.com/evcc-io/evcc/issues/3528#issuecomment-1241211948

andig, 2023-04-15:
Aktuell nur mit dem von Volker geschilderten Workaround möglich. Gastfahrzeuge sind ansonsten genau das und es gelten die Parameter vom Ladepunkt.

/cc @premultiply

closed 2023-04-15 by andig

MarkusGH, 2023-04-16:
> > Die Probleme kenne ich nur vom Hörensagen.
> 
> [#3528 (comment)](https://github.com/evcc-io/evcc/issues/3528#issuecomment-1241211948)

Da steht aber nicht worin diese Edge Cases bestanden. Denn am Ende sollte Software ja die Realität abbilden. Und die hat nun mal unterschiedliche Ladeströme 1p/3p - egal ob Gastfahrzeug oder nicht.

@andig: Gibt es keine Möglichkeit das reinzubekommen? Selbst wenn die Einstellung nur für die 1p/3p Umschaltung verwendet würde wäre das ein Riesenfortschritt, denn das Verhalten der 1p/3p Umschaltung mit dieser riesigen "Regelungslücke" ist momentan wirklich ungünstig.