evcc-io/evcc #7474: Nach Update auf V.166.1 wird angestecktes Auto nicht als angesteckt erkannt.

Marty56, 2023-04-16:
### Describe the bug

Ich habe das Upgrade auf V.166.1 gemacht und hatte dabei mein Auto angesteckt.
Der Mode war auf off, dennoch hat das Auto mit minimalen Ladestrom geladen und es wird nicht verbunden angezeigt.

Ich habe dann kurz auf PVMIN umgeschaltet und dann wieder auf Aus. Keine Änderung des Verhaltens.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2023/04/16 02:33:17 evcc 0.116.1
[main  ] INFO 2023/04/16 02:33:17 using config file: /etc/evcc.yaml
[eebus ] INFO 2023/04/16 02:33:17 Local SKI:  d7138aeb02b47ae2a6f10eebeb83e3d87dca63b1
[lp-1  ] WARN 2023/04/16 02:33:25 poll mode '{always 5m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.

Konfiguration (/etc/evcc.yaml):

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# sponsor token: *****
# sponsortoken: *****

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters

meters:
  - name: wallboxmeter
    type: custom
[... 398 lines cut ...]
[site  ] DEBUG 2023/04/16 02:26:13 pv power: 0W
[site  ] DEBUG 2023/04/16 02:26:13 grid power: 170W
[site  ] DEBUG 2023/04/16 02:26:13 site power: 170W
[lp-1  ] DEBUG 2023/04/16 02:26:13 charger status: A
[lp-1  ] DEBUG 2023/04/16 02:26:13 next soc poll remaining time: 4m50s
[site  ] DEBUG 2023/04/16 02:26:23 ----
[lp-1  ] DEBUG 2023/04/16 02:26:23 charge power: 5W
[site  ] DEBUG 2023/04/16 02:26:23 pv power: 0W
[site  ] DEBUG 2023/04/16 02:26:23 grid power: 181W
[site  ] DEBUG 2023/04/16 02:26:23 site power: 181W
[lp-1  ] DEBUG 2023/04/16 02:26:23 charger status: A
[lp-1  ] DEBUG 2023/04/16 02:26:23 next soc poll remaining time: 4m40s
^C[eebus ] DEBUG 2023/04/16 02:26:27 mdns: stop announcement
[eebus ] DEBUG 2023/04/16 02:26:27 b456c8f752941d4da9333d84438f5b60848dd805 websocket read error:  read tcp 192.168.10.142:4712->192.168.10.94:60318: use of closed network connection
[eebus ] DEBUG 2023/04/16 02:26:27 restarting mdns search
[eebus ] DEBUG 2023/04/16 02:26:27 delaying connection to b456c8f752941d4da9333d84438f5b60848dd805 by 1.346s to minimize double connection probability
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.116.1

StefanSchoof, 2023-04-16:
> Ich habe das Upgrade auf V.166.1 gemacht und hatte dabei mein Auto angesteckt

Das kann die Elli wallbox nicht. Du musst immer vor einem Neustart von evcc das Auto trennen und einen Moment warten bis die eebus Verbindung wieder da ist. Das ist ein Verhalten der Elli wallbox, an dem evcc nichts machen kann. Nur Elli könnte das mit einem Update fixen.

Marty56, 2023-04-16:
Ok. Letztlich hat es die Elli Box ohne dass ich das Auto getrennt habe, doch richtig anzeigt.
Ich muss EVCC restarten und einfach nur warten.

closed 2023-04-16 by Marty56