evcc-io/evcc #7481: Ford wird seit 0.116.1 nicht mehr erkannt

weindler, 2023-04-16:
### Describe the bug

Nach dem Update auf die 0116.1 wird der Ford Kuga nicht mehr erkannt, folgende Fehlermeldung kommt.

`vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'ford': login failed: Post \"https://api.mps.ford.com/api/token/v2/cat-with-ci-access-token\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)`

### Steps to reproduce

Kann jederzeit reproduziert werden

### Configuration details

```yaml
Es wurde nichts geändert
```

### Log details

```text
root@EVCC:~# evcc --log debug
[main  ] INFO 2023/04/16 10:26:58 evcc 0.116.1
[main  ] INFO 2023/04/16 10:26:58 using config file: /etc/evcc.yaml
[main  ] INFO 2023/04/16 10:26:58 starting ui and api at :7070
[db    ] INFO 2023/04/16 10:26:58 using sqlite database: /root/.evcc/evcc.db
[tibber] WARN 2023/04/16 10:27:04 cheap rate configuration has been replaced by target charging and is deprecated
[site  ] INFO 2023/04/16 10:27:04 site config:
[site  ] INFO 2023/04/16 10:27:04   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/04/16 10:27:04     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/04/16 10:27:04     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/04/16 10:27:04     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/04/16 10:27:04   vehicles:
[site  ] INFO 2023/04/16 10:27:04     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/04/16 10:27:04 loadpoint 1:
[lp-1  ] INFO 2023/04/16 10:27:04   mode:        off
[lp-1  ] INFO 2023/04/16 10:27:04   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/04/16 10:27:04   meters:      charge ✓
[lp-1  ] INFO 2023/04/16 10:27:04     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/04/16 10:27:04 phase timer inactive
[lp-1  ] DEBUG 2023/04/16 10:27:04 pv timer inactive
[lp-1  ] DEBUG 2023/04/16 10:27:04 guard timer inactive
[main  ] FATAL 2023/04/16 10:27:04 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
root@EVCC:~#
```

### What type of operating system are you running?

Linux

### Version

evcc 0.116.1

weindler, 2023-04-16:
Mit der Version 0.116.0 ist alles in Ordnung, dies ist nun auch wieder meine aktuellste Version. Diese Version läd übrigens auch im Min PV Modus

VolkerK62, 2023-04-16:
War evtl nur ein temporäres Problem bei Ford?!

andig, 2023-04-16:
Bei Ford hat sich nichts geändert. Das Log zeigt auch nur Timeout bei Ford. Kein Todo.

closed 2023-04-16 by andig